EmuELEC/EmuELEC #988: Odroid Go Ultra handheld compatible with Emulec?

stuart1975, 2022-09-06:
Will Emulec support the new handheld from HardKernel Odroid Go Ultra? Which will be released in October 2022.

https://forum.odroid.com/viewtopic.php?f=29&t=45143

shantigilbert, 2022-09-06:
Yes, but please do not post questions here, this is for support, https://github.com/EmuELEC/EmuELEC/issues/604

closed 2022-09-06 by shantigilbert

cederom, 2024-12-13:
Hello world :-) I had older Odroid Go before where EmuELEC worked out of the box. On OGU (Odroid Go Ultra) EmuELEC 4.7 generic does not seem to work out of the box and there is no DTB file?

I am launching the console with back button pressed, can see the boot logo, then screen goes blank with active backlight, no EmuELEC shows up.

https://github-wiki-see.page/m/EmuELEC/EmuELEC/wiki/How-to-Install