EmuELEC/EmuELEC #1024: about emuelec key map setting?

zack-huangzihan, 2022-09-29:
I am currently porting a new device, and I would like to ask where is the configuration file of key mapping in the system?  
I use EVtest to test the key code. I want to modify the behavior detected by this key code. What do I need to modify?
EMUELEC:/usr/bin # evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      fe6e0030.pwm
/dev/input/event1:      rk805 pwrkey
/dev/input/event2:      goodix-ts
/dev/input/event3:      adc-keys
/dev/input/event4:      rockchip,rk809-codec Headphones
/dev/input/event5:      rockchip,hdmi rockchip,hdmi
Select the device event number [0-5]: 1
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x0 product 0x0 version 0x0
Input device name: "rk805 pwrkey"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 116 (KEY_POWER)
Properties:
Testing ... (interrupt to exit)
Event: time 1654197597.502502, type 1 (EV_KEY), code 116 (KEY_POWER), value 1
Event: time 1654197597.502502, -------------- SYN_REPORT ------------
Event: time 1654197597.673092, type 1 (EV_KEY), code 116 (KEY_POWER), value 0
Event: time 1654197597.673092, -------------- SYN_REPORT ------------

7Ji, 2022-09-29:
>  the configuration file of key mapping in the system

There's no system-wide key mapping, or rather should we say event mapping? (key mapping != event mapping, although neither exists as a system-wide daemon). For some ports (games, not the port for EE as yours), they might use [gptokeyb](https://github.com/EmuELEC/gptokeyb) to map gamepad input events to keyboard input events, but that's neither used system-wide or could map your ``KEY_POWER``. EE is a JeOS after all. (``KEY_POWER`` sent by IR device could potentially be re-interpreted by ``eventlircd`` just like on CE, although that's even before it's read as ``KEY_POWER``, but ``rk805`` is definitely not the case)

> I want to modify the behavior detected by this key code

As said above, there's no system-wide event mapping in EmuELEC, so the ``KEY_POWER`` input event is grabbed by kernel itself and thus shutting down the system. So you'll have to either write a small daemon to grab ``event1`` and interprete that on demand, or patch the kernel to act differently on ``KEY_POWER`` input event, or patch ES/RA/etc to grab on ``event1``, etc

---
And as always, please **don't use github issues for discussion**, it is for tracking of actual issues only, use either github discussion, or our forum, or the dev-tech-talk channel on our discord server.

closed 2022-09-29 by 7Ji

zack-huangzihan, 2022-09-29:
Thank you for your reply, as you said, the system should have a default daemon to detect the power_key event, if there is no this setting system, normal should be I press power_key completely without any response, but in fact it power, I think it has a default system testing, I want to modify this behavior. Do you know where the code is

Closed #1024<https://github.com/EmuELEC/EmuELEC/issues/1024> as not planned.