EmuELEC/EmuELEC #1027: [4.5]gamepad not support "BETOP BETOP XF1 BFM DONGLE" and "   Gamesir-T4 1.25"

hmony, 2022-10-07:
**Describe the bug**
V4.5 gamepad not support "BETOP BETOP XF1 BFM DONGLE" and "   Gamesir-T4 1.25"
V4.6 can support  but Recognized as keyboard or mouse，so in retroarch not support 

`Bus=0003 Vendor=05ac Product=061c Version=0111
N: Name="   Gamesir-T4 1.25"
P: Phys=usb-xhci-hcd.0.auto-1.1/input0
S: Sysfs=/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.1/1-1.1:1.0/0003:05AC:061C.0006/input/input10
U: Uniq=
H: Handlers=mouse0 event3 js0 
B: PROP=0
B: EV=1f
B: KEY=4000 0 7fff000000070000 0 0 0 0
B: REL=3
B: ABS=30627
B: MSC=10

I: Bus=0003 Vendor=20bc Product=503d Version=0111
N: Name="BETOP BETOP XF1 BFM DONGLE"
P: Phys=usb-xhci-hcd.0.auto-1.2/input0
S: Sysfs=/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/0003:20BC:503D.0007/input/input11
U: Uniq=
H: Handlers=kbd mouse1 event4 js1 
B: PROP=0
B: EV=1f
B: KEY=4000 0 7fff000000070000 1800000000 100040000800 c000000000000 0
B: REL=3
B: ABS=30627
B: MSC=10`

**Expected behavior**
can auto detect those gamepad.

**EmuELEC version:*4.5*

**Device (please complete the following information):**
 - SOC: [S095l3a]
 - Brand: [ChinaMobile]
 - Model [CM311-1a_YS]

**Additional context**

[BETOP BETOP XF1 BFM DONGLE.zip](https://github.com/EmuELEC/EmuELEC/files/9735033/BETOP.BETOP.XF1.BFM.DONGLE.zip)
[Gamesir-T4 1.25.zip](https://github.com/EmuELEC/EmuELEC/files/9735035/Gamesir-T4.1.25.zip)

Langerz82, 2022-10-08:
Have you tried manually setting the buttons in EmulationStation, press start and in Bluetooth and Controller settings there should be a mapping option.

Langerz82, 2022-10-08:
Try holding the HOME button for 5 seconds to switch modes and try again. 5 seconds to switch back.
ref: https://cdn.shopify.com/s/files/1/2241/8433/files/T4_Manual.pdf

Also does the Gamesir T4 create a file in /tmp/joypads ? Check via SSH.

hmony, 2022-10-09:
this   “Gamesir T4”  is a custom by “Tencent Start”, The  “home” key is replaced with a “Start” key。 it is a USB wireless gamepad, not a Bluetooth one.
In Emuelec 4.5 can't auto detect in ui， in Emuelec 4.6 can, so I can in the Controller setting to mapping the key, but in retroarch system, the gamepad can't detect.
I have the cfg file for the gamepad is uploading in the head of Additional context 。

Langerz82, 2022-10-09:
Check that your retroarch joypad dir is pointing to the same directory with:
```
 cat /storage/.config/retroarch/retroarch.cfg | grep joypad_autoconfig_dir
```
it should show:
```
joypad_autoconfig_dir = "/tmp/joypads"
```

shantigilbert, 2022-10-09:
This needs some udev rules, there is nothing wrong with the configuration. Or perhaps even a driver for this specific controller that unless someone else is willing to add, this falls into wontfix territory.

hmony, 2022-11-27:
Wrong name recognition of the gamepad name 
[image]
[image]
[image]

Langerz82, 2022-11-28:
@hmony Try and manually enter in the configs the names that are shown in " /proc/bus/input/devices". Also rename the filenames the same then see if retroarch picks up the configs.

Langerz82, 2022-12-24:
Run sdljoytest in SSH, does the controller dongles work ok?

Langerz82, 2024-01-31:
@hmony Try latest build if it's still an issue feel free to reply and I will re-open issue.

closed 2024-01-31 by Langerz82