EmuELEC/EmuELEC #1037: Gamepad Bluetooth V013 dont work on emuelec 4.6 test

Filester10, 2022-10-22:
Hello I have a bluetooth gamepad V013 that works fine in emulec version 4.5, in version 4.6 it doesn't work, it says connected successfully but it is not detected by the system.
I leave attached photo of the gamepad and configuration file in emuelec 4.5. Is it possible to make it work in version 4.6?
thanks

input_device = "V013"
input_driver = "udev"
input_vendor_id = "6473"
input_product_id = "1026"
input_r_y_plus_axis = "+3"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_minus_axis = "-2"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_r_btn = "14"
input_save_state_btn = "14"
input_down_btn = "h0down"
input_volume_down_btn = "h0down"
input_r_y_minus_axis = "-3"
input_l_btn = "13"
input_load_state_btn = "13"
input_r_x_plus_axis = "+2"
input_y_btn = "10"
input_x_btn = "11"
input_menu_toggle_btn = "11"
input_b_btn = "7"
input_reset_btn = "7"
input_a_btn = "8"
input_up_btn = "h0up"
input_volume_up_btn = "h0up"
input_select_btn = "17"
input_l3_btn = "20"
input_start_btn = "18"
input_exit_emulator_btn = "18"
input_l_x_plus_axis = "+0"
input_l_y_minus_axis = "-1"
input_enable_hotkey_btn = "17"
input_l2_btn = "15"
input_rewind_btn = "15"
input_r2_btn = "16"
input_toggle_fast_forward_btn = "16"
input_l_y_plus_axis = "+1"
input_r3_btn = "21"
input_fps_toggle_btn = "21"
input_l_x_minus_axis = "-0"
input_gun_trigger_mbtn = "1"

[image]
[image]

shantigilbert, 2022-10-22:
Use Github for BUGS only, this is not a bug. Use the forum or discord for help on this. But don't expect a solution since this is just a cheap generic controller

shantigilbert, 2022-10-22:
https://github.com/EmuELEC/EmuELEC/issues/604

closed 2022-10-22 by shantigilbert