EmuELEC/EmuELEC #1045: DualShock3（Maybe other controllers） mapping is not correct in v4.4 and v4.5

royfokk, 2022-11-06:
I am using a khadas VIM3, every thing is OK on V4.3. 
Since V4.4 the controller doesn't work properly anymore.

Default DS3 mapping in EmuELEC:

  <inputConfig type="joystick" deviceName="Sony PLAYSTATION(R)3 Controller">
        <input name="a" type="button" id="13" value="1" />
        <input name="b" type="button" id="14" value="1" />
        <input name="down" type="button" id="6" value="1" />
        <input name="hotkeyenable" type="button" id="16" value="1" />
        <input name="left" type="button" id="7" value="1" />
        <input name="leftanalogdown" type="axis" id="1" value="1" />
        <input name="leftanalogleft" type="axis" id="0" value="-1" />
        <input name="leftanalogright" type="axis" id="0" value="1" />
        <input name="leftanalogup" type="axis" id="1" value="-1" />
        <input name="leftshoulder" type="button" id="10" value="1" />
        <input name="leftthumb" type="button" id="1" value="1" />
        <input name="lefttrigger" type="button" id="8" value="1" />
        <input name="right" type="button" id="5" value="1" />
        <input name="rightanalogdown" type="axis" id="3" value="1" />
        <input name="rightanalogleft" type="axis" id="2" value="-1" />
        <input name="rightanalogright" type="axis" id="2" value="1" />
        <input name="rightanalogup" type="axis" id="3" value="-1" />
        <input name="rightshoulder" type="button" id="11" value="1" />
        <input name="rightthumb" type="button" id="2" value="1" />
        <input name="righttrigger" type="button" id="9" value="1" />
        <input name="select" type="button" id="0" value="1" />
        <input name="start" type="button" id="3" value="1" />
        <input name="up" type="button" id="4" value="1" />
        <input name="x" type="button" id="12" value="1" />
        <input name="y" type="button" id="15" value="1" />
</inputConfig>

After deleting the above mapping, I reconfigured it. The correct DS3 mapping in V4.4 and V4.5 should be:

        <inputConfig type="joystick" deviceName="Sony PLAYSTATION(R)3 Controller" deviceGUID="030000004c0500006802000011810000">
                <input name="a" type="button" id="1" value="1" />
                <input name="b" type="button" id="0" value="1" />
                <input name="down" type="button" id="14" value="1" />
                <input name="hotkeyenable" type="button" id="10" value="1" />
                <input name="left" type="button" id="15" value="1" />
                <input name="leftanalogdown" type="axis" id="1" value="1" />
                <input name="leftanalogleft" type="axis" id="0" value="-1" />
                <input name="leftanalogright" type="axis" id="0" value="1" />
                <input name="leftanalogup" type="axis" id="1" value="-1" />
                <input name="leftshoulder" type="button" id="4" value="1" />
                <input name="leftthumb" type="button" id="11" value="1" />
                <input name="lefttrigger" type="button" id="6" value="1" />
                <input name="right" type="button" id="16" value="1" />
                <input name="rightanalogdown" type="axis" id="4" value="1" />
                <input name="rightanalogleft" type="axis" id="3" value="-1" />
                <input name="rightanalogright" type="axis" id="3" value="1" />
                <input name="rightanalogup" type="axis" id="4" value="-1" />
                <input name="rightshoulder" type="button" id="5" value="1" />
                <input name="rightthumb" type="button" id="12" value="1" />
                <input name="righttrigger" type="button" id="7" value="1" />
                <input name="select" type="button" id="8" value="1" />
                <input name="start" type="button" id="9" value="1" />
                <input name="up" type="button" id="13" value="1" />
                <input name="x" type="button" id="2" value="1" />
                <input name="y" type="button" id="3" value="1" />
        </inputConfig>

**EmuELEC version:**
 - Version 4.4 4.5

**Device (please complete the following information):**
 - SOC: A311D
 - Brand: khadas VIM3
 - Model VIM3
 - Version v13

**Additional context**
So far I have only tested the DS3. Let me know if you need me to test other controllers.

royfokk, 2022-11-06:
Default DS3 mapping in EmuELEC:

  <inputConfig type="joystick" deviceName="Sony PLAYSTATION(R)3 Controller">
        <input name="a" type="button" id="13" value="1" />
        <input name="b" type="button" id="14" value="1" />
        <input name="down" type="button" id="6" value="1" />
        <input name="hotkeyenable" type="button" id="16" value="1" />
        <input name="left" type="button" id="7" value="1" />
        <input name="leftanalogdown" type="axis" id="1" value="1" />
        <input name="leftanalogleft" type="axis" id="0" value="-1" />
        <input name="leftanalogright" type="axis" id="0" value="1" />
        <input name="leftanalogup" type="axis" id="1" value="-1" />
        <input name="leftshoulder" type="button" id="10" value="1" />
        <input name="leftthumb" type="button" id="1" value="1" />
        <input name="lefttrigger" type="button" id="8" value="1" />
        <input name="right" type="button" id="5" value="1" />
        <input name="rightanalogdown" type="axis" id="3" value="1" />
        <input name="rightanalogleft" type="axis" id="2" value="-1" />
        <input name="rightanalogright" type="axis" id="2" value="1" />
        <input name="rightanalogup" type="axis" id="3" value="-1" />
        <input name="rightshoulder" type="button" id="11" value="1" />
        <input name="rightthumb" type="button" id="2" value="1" />
        <input name="righttrigger" type="button" id="9" value="1" />
        <input name="select" type="button" id="0" value="1" />
        <input name="start" type="button" id="3" value="1" />
        <input name="up" type="button" id="4" value="1" />
        <input name="x" type="button" id="12" value="1" />
        <input name="y" type="button" id="15" value="1" />
</inputConfig>

royfokk, 2022-11-07:
The SDL_GameControllerDB and retroarch joypads cfg are not correct either.

shantigilbert, 2022-11-16:
You need to setup your controllers in ES and they configuration will be updated. This is not considered a bug.

closed 2022-11-16 by shantigilbert

Langerz82, 2022-12-20:
@shantigilbert Maybe we should insert the record that the user has proposed. There are a couple of users with issues with the default configuration being incorrect. I think this would be beneficial for auto plug n play for ps3 controllers.

Examples:
https://emuelec.discourse.group/t/default-controller-mapping-is-weird-in-v4-6/1525
https://emuelec.discourse.group/t/joystick-for-ps3-doesnt-work-propertly-from-4-4-to-4-6/1553

shantigilbert, 2022-12-20:
@Langerz82 if you think it will fix this issue I have no problem with a PR.