EmuELEC/EmuELEC #1047: About TMNT

clondy999, 2022-11-08:
My device 905x2 and 905x3
Emuelec version: 4.6-TEST-07082022 and EmuELEC-Amlogic ng.aarch64-4.5-Nexus_ devel_ 20221106212519-Generic
Tested game TMNT
Error reason: It can run successfully, but only sound appears, no game image
Log:

`assemblyName: TMNT
appBasePath: /storage/roms/ports/tmntsr/gamedata
appArgs: {}
Redirecting ParisEngine to MONOMODDED_ParisEngine.dll
System.IO.FileNotFoundException: Could not find file "/emuelec/configs/Tribute Games/TMNT/ModSettings.dat"
File name: '/emuelec/configs/Tribute Games/TMNT/ModSettings.dat'
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x001ef] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode) [0x00000] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode)
  at Paris.Game.Menu.Options.LoadModConfig () [0x00006] in <5e8421c2b5ed4c668c20971b92833165>:0 
Params Set:
> None
Removing Program.CrashHandler handler...
[?25lController 0: Twin USB Joystick, GUID: bc200055, Mapping: 03000000bc2000000055000011010000,Twin USB Joystick,dpleft:h0.8,rightx:a2,dpright:h0.2,rightshoulder:b7,dpdown:h0.4,righty:a3,leftshoulder:b6,y:b3,x:b4,b:b0,a:b1,dpup:h0.1,back:b10,leftstick:b13,start:b11,lefty:a1,guide:b10,lefttrigger:b8,righttrigger:a4,rightstick:b14,leftx:a0,platform:Linux,
Controller 1: Twin USB Joystick, GUID: bc200055, Mapping: 03000000bc2000000055000011010000,Twin USB Joystick,dpleft:h0.8,rightx:a2,dpright:h0.2,rightshoulder:b7,dpdown:h0.4,righty:a3,leftshoulder:b6,y:b3,x:b4,b:b0,a:b1,dpup:h0.1,back:b10,leftstick:b13,start:b11,lefty:a1,guide:b10,lefttrigger:b8,righttrigger:a4,rightstick:b14,leftx:a0,platform:Linux,
gpath.c:115: assertion 'filename != NULL' failed

gpath.c:115: assertion 'filename != NULL' failed

FNA3D Driver: OpenGL
OpenGL Renderer: Mali-G31
OpenGL Driver: OpenGL ES 3.2 git.c8adbf9.ad00b04c1b60847de257177231dc1a53
OpenGL Vendor: ARM
MojoShader Profile: glsles
Attempting to patch...
Failed to load Patch Object reference not set to an instance of an object
Instanciating "April_Attack_01" SFX (46448 bytes)
Instanciating "April_Attack_02" SFX (44696 bytes)
Instanciating "April_Attack_03" SFX (48040 bytes)
Instanciating "April_Finisher" SFX (48272 bytes)
Instanciating "April_Flyingkick" SFX (96800 bytes)
Instanciating "April_Heavyswing" SFX (168000 bytes)
Instanciating "April_Homerunswing" SFX (103900 bytes)
Instanciating "April_LevelComplete" SFX (166800 bytes)
Instanciating "April_Risingkick_01" SFX (27896 bytes)
Instanciating "April_SpecialAttack_01" SFX (82560 bytes)
Instanciating "April_SpecialAttack_02" SFX (96000 bytes)
Instanciating "April_Whirlwind" SFX (48000 bytes)
Instanciating "Baxter_CannonLaser" SFX (288000 bytes)
Instanciating "Baxter_CannonLaserShot" SFX (151820 bytes)
Instanciating "Baxter_FistPunch" SFX (240000 bytes)
Instanciating "Baxter_GunReload" SFX (247428 bytes)
Instanciating "Baxter_Gunshot" SFX (251688 bytes)
Instanciating "Baxter_Steps_01" SFX (64752 bytes)
Instanciating "Baxter_Steps_02" SFX (63292 bytes)
Instanciating "Baxter_Wings" SFX (768000 bytes)
Instanciating "Bebop_ChargeAntic" SFX (144000 bytes)
Instanciating "Bebop_Gunshoot1" SFX (161512 bytes)
Instanciating "Bebop_Gunshoot2" SFX (59996 bytes)
Instanciating "Bebop_Laser_Antic" SFX (96000 bytes)
Instanciating "Bebop_LooneyRun" SFX (59996 bytes)
Instanciating "Bebop_Punch" SFX (24052 bytes)
Instanciating "Bebop_RunSteps_01" SFX (68540 bytes)
Instanciating "Bebop_RunSteps_02" SFX (72012 bytes)
Instanciating "Boss_BigKO" SFX (288000 bytes)
Instanciating "Boss_ChargeImpact" SFX (131812 bytes)
Instanciating "Boss_Dead" SFX (175048 bytes)
Instanciating "Boss_Down" SFX (71996 bytes)
Instanciating "Boss_Land" SFX (58212 bytes)
Instanciating "Buzzer_BombDrop" SFX (96000 bytes)
Instanciating "Buzzer_Explosion" SFX (116924 bytes)
Instanciating "Buzzer_Shoot" SFX (126496 bytes)
Instanciating "Casey_HeavyAttack" SFX (96000 bytes)
Instanciating "Casey_HomerunSwing" SFX (76428 bytes)
[... 371 lines cut ...]
Instanciating "PizzaMonster_LeapStart" SFX (62412 bytes)
Instanciating "PizzaMonster_RangedMelee" SFX (70216 bytes)
Instanciating "Player_Bounce" SFX (96000 bytes)
Instanciating "Player_Burnt" SFX (169380 bytes)
Instanciating "Player_Charge_01" SFX (73872 bytes)
Instanciating "Player_Charge_02" SFX (72796 bytes)
Instanciating "Player_Clap" SFX (76248 bytes)
Instanciating "Player_Dead" SFX (384000 bytes)
Instanciating "Player_Deflect" SFX (167996 bytes)
Instanciating "Player_DiveKick" SFX (71996 bytes)
Instanciating "Player_DoubleJump" SFX (77800 bytes)
Instanciating "Player_Drophole" SFX (96000 bytes)
Instanciating "Player_Electrified" SFX (245996 bytes)
Instanciating "Player_EyeSpark" SFX (125136 bytes)
Instanciating "Player_Flingslam" SFX (144004 bytes)
Instanciating "Player_Flingtoss" SFX (144052 bytes)
Instanciating "Player_HeldLasso" SFX (119996 bytes)
Instanciating "Player_Hit" SFX (70552 bytes)
Instanciating "Player_HomeRunSnap" SFX (96176 bytes)
Instanciating "Player_IceBreak" SFX (120000 bytes)
Instanciating "Player_Iced" SFX (197596 bytes)
Invalid data being read in SFXPack using list of sounds from SoundsSettings; SoundsSettings & SFXPack are out of sync! Error message: System.Exception: sndSize < 0
  at Paris.Engine.Audio.AudioManager.Init () [0x0019e] in <8f38b78ff7be491f915329c89705d768>:0 
Video /storage/roms/ports/tmntsr/gamedata/Content/Videos/TMNT_GameIntro.ogv does not have an XNB file! Hacking Duration property!
`

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

shantigilbert, 2022-12-17:
Try removing this line https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec-ports/tmntsr/sources/tmntsr.sh#L78 and also the file `${GAMEDIR}/gamedata/.astc_done` if it exists.

clondy999, 2022-12-18:
Deleted, the problem has not been resolved

shantigilbert, 2022-12-18:
> Deleted, the problem has not been resolved

How did you delete it? 

do this via ssh:

```
EMUELEC:~ # cp /usr/bin/tmntsr.sh /emuelec/bin
EMUELEC:~ # nano /emuelec/bin/tmntsr.sh 
```

use nano to delete the line.

clondy999, 2022-12-19:
> > Deleted, the problem has not been resolved
> 
> How did you delete it?
> 
> do this via ssh:
> 
> ```
> EMUELEC:~ # cp /usr/bin/tmntsr.sh /emuelec/bin
> EMUELEC:~ # nano /emuelec/bin/tmntsr.sh 
> ```
> 
> use nano to delete the line.

I edit the file at the compilation stage, delete this line, and see that the file takes effect

shantigilbert, 2022-12-20:
check #1081

closed 2022-12-20 by shantigilbert