EmuELEC/EmuELEC #1056: 8BitDo controllers

John7wg, 2022-11-20:
Hello,

I'm using emuelec 4.5

I have 2 8BitDo controllers, 8BitDo Arcade Stick and 8BitDo M30 Bluetooth.

I have them both connected by Bluetooth, Emulationstation detects them as "XBOX ONE S CONTROLLER".

Retroarch detects correctly as 8BitDo Arcade Stick and 8BitDo M30 Bluetooth.

The problem is that Emulationstation detects them as the same "XBOX ONE S CONTROLLER" and it does not keep the different button configuration for both controllers, the last controller that is configured overwrites the previous configuration of the other controller and every time I use a different one, I have to reconfigure their buttons.

Thank you

Langerz82, 2022-11-24:
I think there may be an issue with the xbox drivers, because my gamesir (T3 and T4 Pro) controllers are picked up as xbox controllers too.

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

Langerz82, 2022-12-24:
To fix this you have to do a few things.
1. In /storage/joypads copy "Xbox One S Controller.cfg" to "8BitDo Arcade Stick.cfg" and "8BitDo M30 Bluetooth.cfg". It would be something like:
```
cp "/storage/joypads/Xbox One S Controller.cfg" "/storage/joypads/8BitDo Arcade Stick.cfg"
cp "/storage/joypads/Xbox One S Controller.cfg" "/storage/joypads/8BitDo M30 Bluetooth.cfg"
```
2. Edit those files and replace input_device = "Xbox One S Controller" with the correct names 8BitDo etc.
3. Reboot
4. When launching Retroarch it should detect the correct files, and not Xbox One S Controller. Then you can change the button assignment and it should alter the correct files, no guarantees though you might have to manually alter the button assignments.

Langerz82, 2022-12-24:
If it works fine paste the controller file assignments and maybe we can include it in ES by default.

John7wg, 2022-12-31:
I try it but It doesn't work. 

Emulationstation still detects them as "XBOX ONE S CONTROLLER".

Langerz82, 2023-01-08:
I should have been more specific, even though ES will reccognize them as Xbox One S Controller, the extra profiles should be picked up by retroarch.

Langerz82, 2023-06-01:
Closing issue. If you still need support let me know.

closed 2023-06-01 by Langerz82