EmuELEC/EmuELEC #1067: SuperTuxKart does NOT work with emuelec 4.6

roisiano, 2022-12-04:
**Describe the bug**

Hello

I have made a fresh install of emuelec 4.6 (X96 Max+; S905X3). 

I have tried to run the port SuperTuxKart, and I started downloading the game (Download and continue? / Yes).
It seems the game downloads OK, but it never starts. When I try to start it again, it tells me again that "data folder does not exists", and we are in a bucle.

Acabo de instalar emuelec 4.6 actualizando desde 4.4 (en X96 Max+; S905X3).

I have other microSD with emuelec 4.2. In that version SuperTuxKart works.
I tried to move that SuperTuxKart (the files) from 4.2 to 4.6, but it does NOT work in 4.6.

Any solution for this?

Thanks in advance and kind regards

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Ports > SuperTuxKart
2. Click on Yes on "Donwload and continue"
3. Bucle 1 and 2

**Expected behavior**
Download the file and work SuperTuxKart.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version [e.g. 22]

**Device (please complete the following information):**
 - SOC: S095x3
 - Brand: X96
 - Model X96 Max+

shantigilbert, 2022-12-04:
Where are the logs?

closed 2022-12-04 by shantigilbert

roisiano, 2022-12-04:
> Where are the logs?

Sorry. The log after trying to install SuperTuxKart:
http://ix.io/4hJv

roisiano, 2022-12-05:
Hello:

I tried to implement this patch:
https://github.com/EmuELEC/EmuELEC/commit/fa91e96063b78197fd59b018d37a9cf9377f1968

In emuelec 4.6, changing the file:
\Storage\roms\ports_scripts\SuperTuxKart.sh

for this one:

```
#!/bin/bash

# SPDX-License-Identifier: GPL-2.0-or-later
# Copyright (C) 2020-present Shanti Gilbert (https://github.com/shantigilbert)

# Source predefined functions and variables
. /etc/profile

PORT="supertuxkart"
# init_port binary audio(alsa. pulseaudio, default)
init_port ${PORT} alsa

/usr/bin/show_splash.sh gameloading "ports" "supertuxkart"

/storage/downloads/run_supertuxkart.sh >> $EE_LOG 2>&1

ret_error=$?

end_port

exit $ret_error
```

and in \storage\downloads I used the file: 
run_supertuxkart.sh

edited in this way:

```
#!/bin/bash

# SPDX-License-Identifier: GPL-2.0-or-later
# Copyright (C) 2021-present Shanti Gilbert (https://github.com/shantigilbert)

# Source predefined functions and variables
. /etc/profile

# This variable has to match the version on package.mk
VERSION="1.4"

ASSETS="https://github.com/supertuxkart/stk-assets-mobile/releases/download/${VERSION}/stk-assets.zip"
DATA="https://github.com/supertuxkart/stk-code/archive/refs/tags/${VERSION}.zip"
DATAFOLDER="/storage/roms/ports/supertuxkart"

mkdir -p "${DATAFOLDER}"
cd "${DATAFOLDER}"

if [ "$EE_DEVICE" == "Amlogic-ng" ]; then 
fbfix
fi

if [ ! -e "${DATAFOLDER}/data/stk_config.xml" ]; then
    text_viewer -y -w -f 24 -t "Data does not exists!" -m "It seems this is the first time you are launching Super Tux Kart or the data folder does not exists\n\nData is about 250 MB total, and you need to be connected to the internet\n\nDownload and continue?"
        if [[ $? == 21 ]]; then
            ee_console enable
            wget "${DATA}" -q --show-progress > /dev/tty0 2>&1
            unzip "${VERSION}.zip" > /dev/tty0
            rm -rf "${DATAFOLDER}/data"
            mv "stk-code-${VERSION}/data" "${DATAFOLDER}" > /dev/tty0
            rm -rf "stk-code-${VERSION}"
            wget "${ASSETS}" -q --show-progress > /dev/tty0 2>&1
            unzip "stk-assets.zip" -d data > /dev/tty0
            rm "stk-assets.zip"
            rm "${VERSION}.zip"
            ee_console disable
            mkdir -p /storage/.config/supertuxkart/config-0.10

cat > /storage/.config/supertuxkart/config-0.10/players.xml << EOF
<?xml version="1.0"?>
<players version="1" >
    <player name="EmuELEC" guest="false" use-frequency="0"
            icon-filename="1.png"
            unique-id="1" saved-session="false"
            saved-user="0" saved-token=""
            last-online-name="" last-was-online="false"
            remember-password="false"
            default-kart-color="0">
    </player>
</players>            
EOF
            SUPERTUXKART_DATADIR="${DATAFOLDER}" supertuxkart
        else
            exit 0
        fi
else
    SUPERTUXKART_DATADIR="${DATAFOLDER}" supertuxkart
fi
```

So, now, SuperTuxKart seems to be "well downloaded"... but it does NOT start, anyway.

Log:
http://ix.io/4hQy

shantigilbert, 2022-12-06:
create the file `/storage/roms/ports/supertuxkart/data/supertuxkart.git`

```
[1;31m[error  ] FileManager: Could not find file 'supertuxkart.git'in any standard location (esp. ../data).
[0;;m[1;31m[error  ] FileManager: Last location checked '/storage/roms/ports/supertuxkart/data/'.
[0;;m[1;31m[fatal  ] FileManager: Set $SUPERTUXKART_DATADIR to point to the data directory.
[0;;m
```

commit fa91e960 (2022-12-04): SupertuxKart: download correct data

fixes #1067