EmuELEC/EmuELEC #1070: SuperTuxKart does NOT work with emuelec 4.6 (#1067), adding log file

roisiano, 2022-12-04:
Please, join or delete with #1067 ( https://github.com/EmuELEC/EmuELEC/issues/1067 ). Added log file link.

Describe the bug

Hello

I have made a fresh install of emuelec 4.6 (X96 Max+; S905X3).

I have tried to run the port SuperTuxKart, and I started downloading the game (Download and continue? / Yes).
It seems the game downloads OK, but it never starts. When I try to start it again, it tells me again that "data folder does not exists", and we are in a bucle.

I have other microSD with emuelec 4.2. In that version SuperTuxKart works.
I tried to move that SuperTuxKart (the files) from 4.2 to 4.6, but it does NOT work in 4.6.

Any solution for this?

Thanks in advance and kind regards

To Reproduce
Steps to reproduce the behavior:

Go to Ports > SuperTuxKart
Click on Yes on "Donwload and continue"
The file seems to download and unpack but does NOT start the game
Bucle 1 and 2
When I try to migrate SuperTuxKart from emuelec 4.2 (it works) to 4.6, it does NOT¨work
Expected behavior
Download the file and work SuperTuxKart.

EmuELEC version:
Version 4.6

Device (please complete the following information):

SOC: S095x3
Brand: X96
Model X96 Max+ 

Logfile:
http://ix.io/4hJv

Kind regards

shantigilbert, 2022-12-04:
No need to open another issue. This has been closed because it has been fixed https://github.com/EmuELEC/EmuELEC/commit/fa91e96063b78197fd59b018d37a9cf9377f1968

closed 2022-12-04 by shantigilbert

roisiano, 2022-12-04:
> No need to open another issue. This has been closed because it has been fixed [fa91e96](https://github.com/EmuELEC/EmuELEC/commit/fa91e96063b78197fd59b018d37a9cf9377f1968)

Thanks again, @shantigilbert . I thought it was closed because there was no log at all.

Kind regards