EmuELEC/EmuELEC #1072: X96 Max + Bluetooth Gamepad Paired, but Not Working

batoceragaming, 2022-12-05:
I want to report that after clean install EmuElec 4.6 in X96 Max + the Bluetooth gamepad was successfully paired. But not responsive at all.

In fact, in EmuElec 4.5 the bluetooth gamepad works perfectly.

I used T3 Gamepad.

Many thanks for the help.

**EmuELEC version:**
 - Version 4.6

**Device (please complete the following information):**
 - SOC: S905x3
 - Brand: X96 Max +
 - Model: 4gb RAM 1gb LAN 64gb ROM

**Additional context**
Add any other context about the problem here.

Langerz82, 2022-12-08:
Did you try plugging it in via USB, wait till it connects, then pull the cable out and the set the controller mapping?

meriox, 2022-12-14:
I have the same problem with Beelink GT King S922x with my Gamesir G3S.
Besides, the second gamepad - 8bitdo Pro 2 takes a long time to connect after reboot.
There were no problems with version 4.5.

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

DavidOliM, 2023-03-10:
You're lucky you actually got bluetooth to connect, I have the same android box and bluetooth and wifi just don't work at all. And I can't connect two bluetooth gamepads using an USB bluetooth adapter because the second one disconnects

Langerz82, 2023-06-01:
Try this:
https://github.com/EmuELEC/EmuELEC/issues/1083#issuecomment-1571479576

Langerz82, 2023-06-03:
Please post in this thread if you have any Bluetooth Controller pairing issues:
https://github.com/EmuELEC/EmuELEC/issues/1194

closed 2023-06-03 by Langerz82

LEKSUS23, 2024-03-06:
Не могу включить блютуз! на emuelec v 4.6 ! На TV BOX A95X F3 Air.Он в меню включен а в реальности -нет! ЕСЛИ СТАВЛЮ ВЕРСИЮ 4.7 ТО БЛЮТУЗ РАБОТАЕТ А 2 ДЖОЙСТИК НЕ РАБОТАЕТ В ИГРАХ А В МЕНЮ EMUELEC ОН РАБОТАЕТ. В ВЕРСИИ 4.6 ВООБЩЕ ПРИСТАВКА НЕ ВИДИТ НИ ОДНО УСТРОЙСТВО БЛЮТУЗ НО ДЕЛАЕТ ВИД ЧТО ОНА ВЕДЕТ ПОИСК!!!

LEKSUS23, 2024-03-06:
I can't turn on Bluetooth! on emuelec v 4.6 ! On the TV BOX A 95 XF 3 Air.It is included in the menu, but in reality it is not! IF I INSTALL VERSION 4.7 BEFORE BLUETOOTH WORKS AND 2 THE JOYSTICK DOES NOT WORK IN GAMES AND IT WORKS IN THE EMUELEC MENU. IN VERSION 4.6, THE CONSOLE DOES NOT SEE ANY BLUETOOTH DEVICES AT ALL, BUT PRETENDS THAT IT IS SEARCHING!!!

LEKSUS23, 2024-03-06:
Какую версию установить ? Что бы рабортал блютуз как положено?
Which version should I install? To make Bluetooth work properly?