EmuELEC/EmuELEC #1076: Bluetooth failed to reconnect, need to reboot

kernelpanicinit0, 2022-12-10:
**To Reproduce**
Steps to reproduce the behavior:
1. Shutdown your Bluetooth controller 
2. Start your Bluetooth controller 
3. It can't reconnect, I need to reboot emuelec to make it connect again.

**Expected behaviour**
Like in 4.5 I expect the controller reconnects again without rebooting the system 

**EmuELEC version:**
 - Version 4.6

 - SOC:  S095x3
 - Model: X96Max
 - Controller: 8bitdo pro 2

This is so annoying when for some reasons you pause the game, go to do something, the controller enters in sleep mode and you need to reboot emuelec using  ssh

sleepingkarma, 2022-12-10:
I am also having problems with my original ps4 gamepad
When restarting Emuelec loses pair and to pair again I have to do "Forget a Bluetooth device" and pair again
When it restarts Emuelec stops communicating by bluetooth and never communicates again until new pairing

meriox, 2022-12-14:
I have the same problem on Beelink GT King S922x with my Gamesir G3S and 8bitdo Pro 2.

worstcase-scenario, 2022-12-16:
Same issue here in EmuELEC v. 4.6 (and also the versions before) using the 8BitDo Receiver (Wireless USB Adapter) and a PS4-Controller. When I boot up my box for the first time and press the PS-Button on my controller, it pairs and connects to the dongle and Emulationstation works without any problems. But when I have to restart the system (softboot), I have to re-insert the dongle everytime I restart Emuelec in order for Emuelec to recognize/connect to the controller again.

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

sleepingkarma, 2022-12-18:
EmuELEC Version 4.6

SOC: Amlogic S922X

Model: Beelink GT King

Controller: Bluetooth PS4 original

Controller:  Ipega 9076 with dongle

my ipega 9076 connected by dongle also does the same to me at least in v4.4 v4.5 and v4.6. 
Especially when the emulec is idle for a long time I go to use it and it has no connectivity with the dongle and only works again after disconnecting and reconnecting the dongle
and as I said before I am also having problems with the PS4 controller over bluetooth when I restart Emuelec

Langerz82, 2022-12-29:
When you cant re-connect with a controller SSH in and type:
```
systemctl status bluetooth
```
also
```
dmesg | grep bluetooth
```
and post the output in a gist, we will see if we can debug the issues.
https://gist.github.com/

worstcase-scenario, 2022-12-29:
Alrighty then. I did as you told, you can find all the output here:

<script src="https://gist.github.com/worstcase-scenario/451e9b034210aed9a33219beecf60e37.js"></script>

Is ist normal that there's no output after the "dmesg"-command?

Langerz82, 2022-12-29:
@worstcase-scenario hello, my bad try the command:
```
dmesg | grep -i bluetooth
```

worstcase-scenario, 2022-12-30:
@Langerz82 No problem, I updated the gist with the new output.

Langerz82, 2022-12-30:
@worstcase-scenario How are you shutting down the device? Because the bluetooth process says it's been running a very long time.

worstcase-scenario, 2022-12-30:
On my android box, there's now power-off-button, so I always have to pull the power plug in order to shut it down completely. But in this case to reproduce the failing bluetooth-connection, I re-started Emeuelec from the system-menu. And when emuelec restarts, the bluetooth-connection is gone.

Langerz82, 2022-12-30:
@shantigilbert 

Bluetooth handling got changed on 4.6, I suspect this change needs some refinement:
https://github.com/EmuELEC/EmuELEC/commit/2d4c2431540a45fa18003aef6ffdacca96777410

Langerz82, 2022-12-30:
emustation-config:
Might need to restart bluetooth as well prior to call to emuelec-bluetooth.
``
BTENABLED=$(get_ee_setting ee_bluetooth.enabled)
if [[ "$BTENABLED" == "1" ]]; then
    # Search for bluetooth gamepads while ES loads. 
    (
        batocera-bluetooth trust
    ) &
fi
```
Change to same as emuelec_autostart:
```
# Start Scanning for Bluetooth Controllers
BTENABLED=$(get_ee_setting ee_bluetooth.enabled)

if [[ "$BTENABLED" != "1" ]]; then
systemctl stop bluetooth
rm /storage/.cache/services/bluez.conf & 
else
systemctl restart bluetooth
emuelec-bluetooth &
fi
```

Langerz82, 2022-12-30:
I doubt that will fix the standby issue, when ES enters standby it might have to call a script like that other python file that loops looking for devices continuously. The standby loop in ES only gets called a few times then not at all in standby mode so it needs an external script to do the looping.

shantigilbert, 2022-12-30:
> @shantigilbert
> 
> Bluetooth handling got changed on 4.6, I suspect this change needs some refinement: [2d4c243](https://github.com/EmuELEC/EmuELEC/commit/2d4c2431540a45fa18003aef6ffdacca96777410)

It could be, but the other option is severely broken.

> I doubt that will fix the standby issue, when ES enters standby it might have to call a script like that other python file that loops looking for devices continuously. The standby loop in ES only [...]

the issue is not when ES goes to sleep, it's when the gamepad goes to sleep (e.g. it disconnects from BT).

Langerz82, 2022-12-30:
@shantigilbert When ES is not on standby it auto detects the controller does it not? What if we had a process that constantly checks the existing bt devices if they are on? Same as emuelec-bluetooth except always running? Let me know if this would be a solution and I can modify the code for a new script to handle this function.

shantigilbert, 2022-12-31:
> When ES is not on standby it auto detects the controller does it not?

At boot it will always scan for BT gamepads.  and will pair any that is found. 

> Same as emuelec-bluetooth except always running?

You mean as a service? I don't think this is the issue, BT should reconnect even with no other help as soon as its enabled. Having the BT scan running all the time might not be a good idea for many .

Langerz82, 2022-12-31:
@shantigilbert 
The 8-bit dongle receiver uses a seperate bluetooth adapter for connectivity (I think). Could it be that bluetooth needs to be restarted in order for it to go through it's pairing process again? I think we need some kind of trigger to restart or to re-initiate the pairing process the dongle uses.

shantigilbert, 2022-12-31:
> @shantigilbert The 8-bit dongle receiver uses a seperate bluetooth adapter for connectivity (I think). Could it be that bluetooth needs to be restarted in order for it to go through it's pairing [...]

I understand what you mean, but there is no way to send a trigger when a gamepad has been disconnected. I think the issue is deeper than just that. Bluetooth stack changed a lot from earlier versions, hence why it as working before, I do not think its related to the emuelec-bluetooth service as that has always been used to pair and connect but re-connect is done by the system itsel(kernel/bluetooth stack)

By "8-bit dongle", you mean the 8bitdo one?

Langerz82, 2022-12-31:
@shantigilbert 
Happy NY mate!! I just woke up and feeling very hungover. haha.

Maybe we just have it launch the search bt process when ES enters standby and kills it off when in normal mode.

I've created a PR for it:
https://github.com/EmuELEC/EmuELEC/pull/1093
and
https://github.com/EmuELEC/emuelec-emulationstation/pull/71

8-bitdo dongle yah!

Langerz82, 2023-01-01:
Can someone with the issue go into ES Main Menu >> Controller and Bluetooth Devices, Forget all bluetooth controllers.
Then in SSH type:
```
emuelec-bluetooth
```
and paste the output here?

worstcase-scenario, 2023-01-02:
Hi. When I go into the Bluetooth-Menu to forget a bluetooth device, there's only the message "No bluetooth devices found".
But the controller still works with the 8bitdo-dongle....

Here's the SSH-output, just in case:

EMUELEC:~ # emuelec-bluetooth
Scanning for available devices for 90 seconds, please wait...
Bluetooth adapter not found

Getting pairable devices, please wait...

Langerz82, 2023-01-02:
@worstcase-scenario 
Okay SSH in and try the following, get in directory "~" (storage), then type:
```
nano test.py
```
copy/paste the following:
```
#!/usr/bin/env python

import time
import logging
from bluetool import Bluetooth

logger = logging.getLogger(__name__)

if __name__ == "__main__":
    try:
        bt = Bluetooth()
    except (bluezutils.BluezUtilError) as error:
            logger.error(str(error) + "\n")
    
    if (bt):
        print('Scanning for available devices for 90 seconds, please wait...')
        bt.start_scanning(60)
        time.sleep(10)
        print('Getting pairable devices, please wait...')
        devices = bt.get_devices_to_pair()
        for device in devices:
            mac = device['mac_address'].decode('utf-8')
            name = device['name'].decode('utf-8')
            icon = bt.get_device_property(mac,'Icon')
            if not icon or (icon and icon == 'input-gaming'):
                print('Found MAC: {}\tName: {}'.format(mac,name))
                print('Found controller {} Name: {}, trusting...'.format(mac,name))
                bt.trust(mac)
                if bt.get_device_property(mac,'Trusted') == 1:
                    print('Trusted {}, quick pause, then pairing...'.format(name))
                    time.sleep(5)
                    bt.pair(mac)
                    if bt.get_device_property(mac,'Paired') == 1:
                        print('Paired {}, quick pause, then connecting...'.format(name))
                        time.sleep(5)
                        bt.connect(mac)
                        if bt.get_device_property(mac,'Connected') == 1:
                            print('Connected {}, exiting...'.format(name))
```
then save the file (Ctrl+x, Y for save), and close. 
give the file exec permissions with:
```
chmod +x test.py
```
then finally run the file with:
```
./test.py
```
it should exit quit quickly as you do not have an internal bluetooth adapter.

worstcase-scenario, 2023-01-02:
I'm quite sure that the android box has an internal bluetooth adapter. When I use it without booting into emuelec, there's no need for the 8bit-dongle when it's running on android. 

I followed your instructions, here's the output:

EMUELEC:~ # ./test.py
Scanning for available devices for 90 seconds, please wait...
Bluetooth adapter not found

Getting pairable devices, please wait...
EMUELEC:~ #

Langerz82, 2023-01-02:
@worstcase-scenario well that's in an issue that it's not picking up your wireless bluetooth adapter. But aside from that the script should also be exiting early for you so it free's up the bluetooth service for the 8bitdo dongle.

worstcase-scenario, 2023-01-02:
Wait - do you want me to run the script when emeuelec has lost the connection to the dongle and see what happens then?

Langerz82, 2023-01-02:
@worstcase-scenario just run it normally, you dont have to wait for anything, and it should exit early.

worstcase-scenario, 2023-01-02:
Running the script (when the connection between dongle and PS4-controller after restarting the emuelec-system has been lost) doesnt't change anything:

EMUELEC:~ # ./test.py
Scanning for available devices for 90 seconds, please wait...
Bluetooth adapter not found

Getting pairable devices, please wait...

Then I removed the dongle, here's the output:

EMUELEC:~ # bluetoothctl
Agent registered
[bluetooth]#

Langerz82, 2023-01-02:
@worstcase-scenario 
Try this file contents in test.py:
https://gist.github.com/Langerz82/6ccb21555cd19af753b479682772a94e

worstcase-scenario, 2023-01-02:
I plugged the dongle back in and emuelec connected again like it does normally. Then I ran your new script:

EMUELEC:~ # ./test.py
Scanning for available devices for 90 seconds, please wait...
Bluetooth adapter not found

Getting pairable devices, please wait...
EMUELEC:~ #

Langerz82, 2023-01-02:
@worstcase-scenario 
try this I updated the script again:
https://gist.github.com/Langerz82/6ccb21555cd19af753b479682772a94e

worstcase-scenario, 2023-01-02:
Hmmm... same output:

EMUELEC:~ # ./test.py
Scanning for available devices for 90 seconds, please wait...
Bluetooth adapter not found

Getting pairable devices, please wait...
EMUELEC:~ #

worstcase-scenario, 2023-01-02:
Just mentioning this because searching for bluetooth-devices seems to work fine....

Langerz82, 2023-01-02:
@worstcase-scenario the issue is that program blocks your dongle from pairing until it is exited, so I'm trying to make it abort early but it's not catching the errors at the start and aborting.

worstcase-scenario, 2023-01-02:
Ok. Tried to pair the PS4-controller via bluetooth without dongle. Emuelec was scanning for devices, but didn't find the PS4-Controller (being in pairing mode).

worstcase-scenario, 2023-01-02:
Isn't it possible that the problem is the USB-connection from Emuelec to the dongle that is somehow lost after a system restart and needs to be kind of "resetted"? Because the bluetooth control is most likely handled by the dongle itself, right?

Langerz82, 2023-01-02:
@worstcase-scenario I am testing out some fixes, that reset the bluetooth service when ES is restarted and when it enters standby mode, just got to make sure it doesnt interfere with normal operations.

worstcase-scenario, 2023-01-02:
Okay. But - as I have written before - in my case, the dongle doesn't loose the Bluetooth Connection to the Controller when I restart Emulationstation. It only loses the connection when I have to restart the system....

sleepingkarma, 2023-01-02:
my ipega 9076 connected by RF dongle especially when the emulec is idle for a long time I go to use it and it has no connectivity with the dongle and only works again after disconnecting and reconnecting the RF dongle even if I exit idle with the keyboard and control Emuelec with the keyboard at least in v3.9 v4.4 v4.5 and v4.6. 

I can pair my PS4 controller over Bluetooth without problems but when I restart Emuelec loses pair and need to remove from bluetooh device and pair it again. This only happens in v4.6

Langerz82, 2023-01-02:
@worstcase-scenario 
@sleepingkarma 
Hey what EmuELEC image files do you use? The Amlogic-ng Generic? If so I can upload a packaged test version for you both. Keep in mind the standby bug isn't fixed, currently there is no workaround.

Langerz82, 2023-01-02:
here it is for testing:
https://github.com/Langerz82/EmuELEC/releases/tag/v4.6bt

worstcase-scenario, 2023-01-03:
Sorry, can't test the image file. I only have one micro-SD-card with Emuelec and all the roms and I don't want to erase that. If I get my fingers on a second card, I'll try the image....

Langerz82, 2023-01-03:
Good news I think. I've made it so when in standby the usb restarts every 60 seconds and the usb stick should go into pairing mode, so the controller can go off, and be turned on again. It will need testing to make sure it doesn't break anything but it's looking promising.

sleepingkarma, 2023-01-04:
@Langerz82 
I have tested the trial version but the PS4 bluetooth controller still has the same problems.  If you restart the EmulationStation you don't lose the pair but if you restart Emuelec you never get connectivity again until you remove the Bluetooth device and pair it again. I remind you that the PS4 controller connects to Beelink without a dongle.

The Ipega 9076 that connects by an RF dongle is also with the same problems. If it is idle for too long it does not get connectivity back until you reconnect the dongle or turn off the box. If I restart the Emuelec it still has no connectivity but I have had this problem since version 3.9

Langerz82, 2023-01-04:
@sleepingkarma 
Can you try a few things for me to try and debug the issue:
Can you do a restart so it loses connectivity then SSH and check the following - 

When you cant re-connect with a controller SSH in and type:
```
systemctl status bluetooth
```
also
```
dmesg | grep -i bluetooth
```
one more thing also check the permissions of the following with:
```
ls /storage/.cache -la
ls /storage/.cache/bluetooth -la
```
and post the output in a gist, we will see if we can debug the issues.
https://gist.github.com/

Also try running:
```
emuelec-bluetooth
```
and see if it will pair the PS4 controller without you having to select forget device.
Thanks in advance. ^-^

worstcase-scenario, 2023-01-04:
Hi. I managed to get an microSD-card and installed your version. At first, everything went okay, but when I restarted the system, the connectivity to the controller was lost again, like in 4.6.

I SSH'd in and copied all the output you wanted into a gist:

https://gist.github.com/worstcase-scenario/7427c59e14c9af7da2e4e6e2da1846d5

sleepingkarma, 2023-01-04:
@Langerz82 
I put the shh output here
https://gist.github.com/sleepingkarma/a18a962d2f43a7ef73b4c329191ee05e

but next I took the PS4 remote and it connected even after the reset to Emuelec. Then I rebooted several times and when I turned on the PS4 remote it connected right away via bluetooth. Even turning the box off. And I couldn't figure out why the day before it wasn't working.  

ao I removed from bluetooth device and then pair again never reconnected the PS4 controller after resetting emuelec

Now output of the shh is this
https://gist.github.com/sleepingkarma/a18a962d2f43a7ef73b4c329191ee05e?permalink_comment_id=4424438#gistcomment-4424438

sleepingkarma, 2023-01-04:
the Ipega 9076 after being paused a few times fails to connect until it connects and disconnects the RF 2.4 Dongle or turns off the box

Langerz82, 2023-01-05:
@worstcase-scenario I got some fixes I think for your case. Which I will upload when I get some time.

@sleepingkarma Now your issue is complex and not figured out. Can you run the following and post output?
```
ls -la .cache/bluetooth/*/*
```

sleepingkarma, 2023-01-05:
here is the output
[LINK](https://gist.github.com/sleepingkarma/a18a962d2f43a7ef73b4c329191ee05e?permalink_comment_id=4426013#gistcomment-4426013)

after removing the bluetooth devices and dongles, just a usb keyboard (because i was using a keyboard with RF 2.4 dongle) i restarted, turned off the box but always lost connectivity 

but magic was done when I did
bluetoothctl
scan on
trust 28:C1:3C:51:3F:CC
pair 28:C1:3C:51:3F:CC
it never lost connectivity again 
then I connected the RF dongle from the Ipega9076, the RF dongle from the keyboard and it never lost connectivity again
I want to remind you that these RF dongles are 2.4 radio and no bluetooth dongles

sleepingkarma, 2023-01-05:
I have been testing and I can confirm that I remove the bluetooth device, pair it again and when I reboot it loses connectivity. 
I make the manual connection by shh and never lose the connectivity in the PS4 gamepad when restarting or disconnecting the box

Langerz82, 2023-01-05:
Hey can you try and make a test.py file in storage (~), with the contents below and try and run it.
```
nano test.py
```
contents: https://gist.github.com/Langerz82/c62ec95847d8d1f3bf34eae88232d2b9
```
chmod +x test.py
```
to run type:   ./test.py

Langerz82, 2023-01-05:
@sleepingkarma sorry the above file had errors which I have hopefully fixed just now I've updated the file.

sleepingkarma, 2023-01-05:
here is the output
https://gist.github.com/sleepingkarma/a18a962d2f43a7ef73b4c329191ee05e?permalink_comment_id=4426209#gistcomment-4426209

Langerz82, 2023-01-05:
This case: ----With Emuelec pair and PS4 disconnected after reset Emuelec
When the program starts the scan make sure you are powering on the PS4 controller.

sleepingkarma, 2023-01-05:
here is the output in this [version](https://github.com/Langerz82/EmuELEC/releases/tag/v4.6bt)
https://gist.github.com/sleepingkarma/a18a962d2f43a7ef73b4c329191ee05e?permalink_comment_id=4426297#gistcomment-4426297

after test.py ps4 gamepad connected but after reset Emuelec lose the connectivity

Langerz82, 2023-01-06:
@sleepingkarma 
Ah good though, it connected the Controller fine, I will fix your restart issue when I get some free time, and upload a new test version.

Langerz82, 2023-01-06:
@sleepingkarma 
I have updated the image for testing see:
https://github.com/Langerz82/EmuELEC/releases/tag/v4.6bt

The scan for new bluetooth devices occurs just before EmulationStation boots, so when it is loading that is when you should turn the controller on.

worstcase-scenario, 2023-01-06:
@Langerz82 

FYI: I tried the updated image, but in my case, the re-connection problem after restarting the system still exists.

Langerz82, 2023-01-06:
@worstcase-scenario did you change the settings in Emulation Station >> EmuELEC Settings, enable USB restart and also set a time restart for USB in standby.

[49 more replies]

closed 2023-06-03 by Langerz82