EmuELEC/EmuELEC #1077: 4.6-TEST-07082022: version glibcxx_3.4.**' not found (required by retroarch32)

brolifen, 2022-12-10:
**Describe the bug**
Error pops up when running psx emulation on the Beelink Super Console X King after upgrading to current test branch.

**Screenshots**
[image]

**EmuELEC version:**
4.6-TEST-07082022

**Device (please complete the following information):**
 - SOC: S922X
 - Brand: Beelink
 - Model Super Console X King

brolifen, 2022-12-10:
Same issue occurs after downgrading to 4.6 stable branch.

brolifen, 2022-12-10:
Resolved this by going in the game settings -> per system advanced settings and changing the cores to "PCSX rearmed" on auto it seems to choose "PCSX REARMED 32B" which then causes the mentioned issue.

miwasp, 2022-12-10:
You will have problems with every 32Bit core in your installation.

At first, 4.6 stable is newer than 4.6 TEST, so this is not a downgrade.

It seems you did an update from an older version and you still have old libs in /emuelec/lib32 that are no longer needed. You have 3 options here:
1. Manually delete all libs in /emuelec/lib32 and reboot
2. Use „Reset scripts and binaries“ from the menu
3. Do a fresh install

After that switch back to pcsx_rearmed_32b as it has better performance.

brolifen, 2022-12-10:
> 2. Use „Reset scripts and binaries“ from the menu

Thank you! This seems to have fixed the problem too and my other open issue too!

https://github.com/EmuELEC/EmuELEC/issues/1078

closed 2022-12-13 by shantigilbert

gingerbeardman, 2022-12-22:
Thanks. This also fixed it for me. 

Odd how my device could get in this state by following stable updates?