EmuELEC/EmuELEC #1079: som gamepad hotkey not work

wang80919, 2022-12-14:
use emuelec ui mapping some gamepad.
hotkey set use SELECT button.
in ppsspp drastic press SELECT+START can quit game.
in RetroArch can not quit game.
in RA MENU INPUT HOTKEY it is not SELECT.
please auto update to EE's hotkey.

wang80919, 2022-12-14:
ee 4.4 4.5 4.6 has this issue.
ee 4.3 not has this bug.

Langerz82, 2022-12-14:
Did you manually configure your controllers buttons in ES Wireless and Controller settings, then assign the HOTKEY to the SELECT button?

wang80919, 2022-12-14:
> Did you manually configure your controllers buttons in ES Wireless and Controller settings, then assign the HOTKEY to the SELECT button?

yes .
4.3 do not has  this bug.

Langerz82, 2022-12-14:
Post your controller config file in /tmp/joypads/"Your Controller".cfg.

Mine is this:
```
  GNU nano 6.4                          Xbox 360 Controller.cfg
input_device = "Xbox 360 Controller"
input_driver = "udev"
input_vendor_id = "1118"
input_product_id = "654"
input_r_y_plus_axis = "+4"
input_left_btn = "h0left"
input_state_slot_decrease_btn = "h0left"
input_r_x_minus_axis = "-3"
input_right_btn = "h0right"
input_state_slot_increase_btn = "h0right"
input_r_btn = "5"
input_save_state_btn = "5"
input_down_btn = "h0down"
input_volume_down_btn = "h0down"
input_r_y_minus_axis = "-4"
input_l_btn = "4"
input_load_state_btn = "4"
input_r_x_plus_axis = "+3"
input_y_btn = "2"
input_x_btn = "3"
input_menu_toggle_btn = "3"
input_b_btn = "0"
input_reset_btn = "0"
input_a_btn = "1"
input_up_btn = "h0up"
input_volume_up_btn = "h0up"
input_select_btn = "6"
input_l3_btn = "9"
input_start_btn = "7"
input_exit_emulator_btn = "7" <--- to exit emulator button
input_l_x_plus_axis = "+0"
input_l_y_minus_axis = "-1"
input_enable_hotkey_btn = "6" <--- the select button
input_l2_axis = "+2"
input_rewind_axis = "+2"
input_r2_axis = "+5"
input_toggle_fast_forward_axis = "+5"
input_l_y_plus_axis = "+1"
input_r3_btn = "10"
input_fps_toggle_btn = "10"
input_l_x_minus_axis = "-0"
```

wang80919, 2022-12-14:
I'm not beside the machine now.

I remember that.

input_enable_hotkey_btn = "8" <--- the select button

I has other gamepad

input_enable_hotkey_btn = "10" <--- the select button

Different gamepad have different values.

wang80919, 2022-12-14:
some gamepad  6 is  home button.
but I do not  want use home button.

Langerz82, 2022-12-14:
Paste your config when you can and I'll take a look. Auto kill on retroarch for 4.6 got disabled from memory. If we can configure it to default to SELECT+START I think we should but I need confirmation from your config file and if shantigilbert agrees we change it.

wang80919, 2022-12-14:
I can press L3+R3 
select 
settings  ->input -> Input Hotkey Binds  -> Hotkey Enable 
press SELECT 。
than.
I can use SELECT+START to quit game now.

but I  want  auto confirmation  this .

wang80919, 2022-12-14:
whan I  change use other gamepad 
I must do it  again.
I hope it is auto .

Langerz82, 2022-12-14:
@shantigilbert what do you think man? Should we enable gptokeyb killthis for retroarch and retrorun? Maybe we make an ES config enabled by default with an option to disable it?

wang80919, 2022-12-14:
> @shantigilbert what do you think man? Should we enable gptokeyb killthis for retroarch and retrorun? Maybe we make an ES config enabled by default with an option to disable it?

why 4.3 ok?

ee 4.3 press hotkey+start
show
press again to quit game.

I want 4.6 same as  4.3 .

shantigilbert, 2022-12-14:
@Langerz82 No. From what I am reading this is not an issue with hotkeys in retroarch. I am not even sure this is even an issue/bug.

wang80919, 2022-12-14:
[image]
[image]
[image]

In 4.3 is not has  this issue.

Langerz82, 2022-12-14:
Confirm Quit needs to be off, and the Quit Controller needs to be set to SELECT+START.

@shantigilbert I do not think the above is the default action though. Enabling gptokeyb would make it more consistent in having it the same way for all emulators. Unless there is cases where in game the SELECT+START combo is needed in-game for a different function?

shantigilbert, 2022-12-14:
On my end this works as expected, RA has its own hotkey functions, and they work as expected. Retroarch (as any other emulator that has a native quit function) does NOT need gptokeyb.  

If this is a cloned/chinese gamepad then I am not going to even bother looking into this, those gamepads are so weird I don't feel like wasting time just because 2 buttons work different. You are free to do so, but keep in mind gptokeyb is NOT needed for Retroarch.

Langerz82, 2022-12-14:
If we use these values in the default retroarch config, the SELECT+START to quit should always work and not rely on the gamepads autoconfig.
```
input_quit_gamepad_combo = "4"
menu_show_quit_retroarch = "true"
quit_on_close_content = "0"
quit_press_twice = "false"
```
I'm not too sure about retrorun though, does it use the ra config too?

shantigilbert, 2022-12-14:
> If we use these values in the default retroarch config, the SELECT+START to quit should always work and not rely on the gamepads autoconfig.
>
> ```
> input_quit_gamepad_combo = "4" [...]

Maybe I am not understanding correctly. Why is this needed? it should not be always SELECT+START it should be HOTKEY+START and right now HOTKEY+START works fine.

Langerz82, 2022-12-14:
HOTKEY is not always correct, In Retroarch the default was assigned value 8, when my HOTKEY in ES was assigned 6 (as SELECT). 8 Doesnt even exist in my auto-config. I guess the issue is the auto-configs are sometimes unreliable, having SELECT+START as a quit is good fixed combination that does rely on controller button codes.

If we change this it might also be worth considering to change the default Menu Toggle Controller combo from L3+R3 to something that can be done with Arcade Controller Sticks as well, (holding SELECT for 2 seconds might be a good option).

shantigilbert, 2022-12-14:
> having SELECT+START as a quit is good fixed combination that does not rely on controller button codes.

I do not agree. the quit combination should always be HOTKEY+START. If there is an issue with the auto-config then that should be the part that should be fixed, but like I said, I cannot reproduce this. for me it works as it should.

Langerz82, 2022-12-14:
If the config is changed I think it would be an alternate combination. HOTKEY+START should still perform as expected.

shantigilbert, 2022-12-14:
As long as HOTKEY+START works as it should and nothing else gets broken for other gamepads (specially official gamepads) I have no issue if this is an option. But like I said, I am not going to do this change as I cannot even reproduce it.

shantigilbert, 2022-12-14:
I see the issue reported with the photos, but again, I cannot reproduce it. For my setup it works as it should. I do not want to deal with gamepads again, especially with so many clones around that work different from each other.

wang80919, 2022-12-16:
> @shantigilbert In the pictures his showing he actually has the same issue I have mentioned. The hotkey in ES is assigned to 6, but when RA starts it thinks that 8 is the hotkey which is non-existent on my controller. So for me to exit I have to go through the menu (and the menu is also have to L3+R3 very non-obvious because hotkey disabled), or change the config, which is not a big deal for me, but for newbies they may have trouble.

yes ,
change the config, which is not a big deal for me, but for newbies they may have trouble.

wang80919, 2022-12-16:
> As long as HOTKEY+START works as it should and nothing else gets broken for other gamepads (specially official gamepads) I have no issue if this is an option. But like I said, I am not going to do this change as I cannot even reproduce it.

in 4.3 it is ok .
in 4.4 4.5 4.6 not ok

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

Langerz82, 2022-12-27:
@wang80919 You do realise that setting the hotkey button as select can be done on top of setting the select button as well. The issue you describe here is only when you do not set the hotkey button, which is kind of silly because it will not conflict with any other button assignments (you have to hold it in + press other buttons for it to work).

wang80919, 2022-12-27:
> @wang80919 You do realise that setting the hotkey button as select can be done on top of setting the select button as well. The issue you describe here is only when you do not set the hotkey button, [...]

https://user-images.githubusercontent.com/19446129/207643665-a51e0480-00d9-4006-a1ac-f673e6828b11.png

Langerz82, 2022-12-27:
> > @wang80919 You do realise that setting the hotkey button as select can be done on top of setting the select button as well. The issue you describe here is only when you do not set the hotkey button, which is kind of silly because it will not conflict with any other button assignments (you have to hold it in + press other buttons for it to work).
> 
> [](https://user-images.githubusercontent.com/19446129/207643665-a51e0480-00d9-4006-a1ac-f673e6828b11.png) button 6 is select button.

Sorry my bad. I also noticed you are making a hotkey for gamepad 2. But retroarch can have only one hotkey set for the 1st controller I think.

Langerz82, 2023-01-21:
I got a PR in the works which would solve your issue. Technically this is not a bug so I will close this issue. For more info -
See: https://github.com/EmuELEC/EmuELEC/pull/1109

closed 2023-01-21 by Langerz82

Khinhphu, 2024-06-24:
your quit controller combo is on none, put on select + start, then you quit game with pressing select + start

wang80919, 2024-06-25:
> your quit controller combo is on none, put on select + start, then you can quit game with pressing select + start

bug is ".config/emulationstation/scripts/retroarch.sh"
the lawawa EmuELEC 4.7.1 fixed it .

shantigilbert, 2024-06-25:
>>  the lawawa EmuELEC 4.7.1 fixed it .

Why not post the fix instead?

wang80919, 2024-06-25:
> > > the lawawa EmuELEC 4.7.1 fixed it .
> 
> Why not post the fix instead?

关我屁事，我有不是 辣娃娃。

wang80919, 2025-01-23:
[image]

wang80919, 2025-01-23:
[image]

Khinhphu, 2025-01-23:
just get a xbox360 controller, or controller with Xinput