EmuELEC/EmuELEC #1080: 4 6 openbor dragonrise zero delay joystick wont map

armstrongstreener, 2022-12-15:
As the title. Dragonrise zero delay boears wont map the joystick in openbor. Emuelec 4.6. 4 3 was working fine

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

armstrongstreener, 2023-01-09:
Having the same problem with a snes style usb controller. Any update on how to fix please?

Langerz82, 2023-02-01:
related issue:
https://github.com/EmuELEC/EmuELEC/issues/1099

armstrongstreener, 2023-02-02:
Still couldnt get this problem fixed

Langerz82, 2023-12-14:
@armstrongstreener It's been awhile, do you still have this issue? Did you upgrade from 4.3 to 4.6? If you have try in EmuELEC Settings >> Danger Zone >> Reset Scripts and Binaries.

If that does not work you could try a new 4.6 clean installation and see if that works.

If you do let me know how it goes. I noticed gptokey is twice loaded with the launch script but I am not sure if that would have and adverse effect.

Langerz82, 2024-01-23:
You could also try and copy the 4-3 master.cfg file into the config directory and see if that works, in SSH do:
```
mv /emuelec/configs/openbor/master.cfg /emuelec/configs/openbor/master.cfg.bak
mv /emuelec/configs/openbor/Saves /emuelec/configs/openbor/Saves.bak
wget -P /emuelec/configs/openbor https://github.com/EmuELEC/EmuELEC/raw/v4.3/packages/sx05re/emulators/openbor/config/master.cfg
```
If it does not work and you need to change the master.cfg back do:
```
mv /emuelec/configs/openbor/master.cfg /emuelec/configs/openbor/master.cfg.43.bak
mv /emuelec/configs/openbor/master.cfg.bak /emuelec/configs/openbor/master.cfg
rm -r /emuelec/configs/openbor/Saves
mv /emuelec/configs/openbor/Saves.bak /emuelec/configs/openbor/Saves
```

Langerz82, 2024-01-23:
Just tested this and my results were bad and the controllers were stuffed! I think the only real solution is to revert the package file to 4-3 and compile a new build.

See:
https://github.com/EmuELEC/EmuELEC/tree/v4.3/packages/sx05re/emulators/openbor

Langerz82, 2024-01-28:
See Pull Request:
https://github.com/EmuELEC/EmuELEC/pull/1287

Langerz82, 2024-01-31:
fix is merged in main dev branch. closing issue.

closed 2024-01-31 by Langerz82