EmuELEC/EmuELEC #1081: TMNT Shredder’s Revenge is broken

djibux, 2022-12-17:
Hello,
The [current tmntsr.sh script](https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec-ports/tmntsr/sources/tmntsr.sh) is broken.  Line 59 it reads
```
  tar –xvzf /emuelec/configs/tmntsr.tar.xz –C "${ROMSDIRECTORY}/ports"
```
There are two issues: first the hyphens should be `-`, not `–`.  Second, the command won’t run.
```bash
# tar -xvzf /emuelec/configs/tmntsr.tar.xz -C /storage/roms/ports
tar: invalid magic
tar: short read
```
I guess the line should read: `tar -xvJf /emuelec/configs/tmntsr.tar.xz -C "${ROMSDIRECTORY}/ports"`.

I am not posting a PR because even with that change I can’t seem to get the game to work.

Log is as follows, referencing a file not found: `/emuelec/configs/Tribute Games/TMNT/ModSettings.dat`.  Indeed it does not exist and I am not sure where to get it from.
```
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.6 (GameForce.aarch64)
umount: can't unmount /storage/roms/ports/mono-6.12.0.122-aarch64.squashfs: Invalid argument
assemblyName: TMNT
appBasePath: /storage/roms/ports/tmntsr/gamedata
appArgs: {}
Redirecting ParisEngine to MONOMODDED_ParisEngine.dll
System.IO.FileNotFoundException: Could not find file "/emuelec/configs/Tribute Games/TMNT/ModSettings.dat"
File name: '/emuelec/configs/Tribute Games/TMNT/ModSettings.dat'
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x001ef] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode) [0x00000] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode)
  at Paris.Game.Menu.Options.LoadModConfig () [0x00006] in <3f0176624b77491b8d88b7eee4f754ce>:0 
Params Set:
> None
Removing Program.CrashHandler handler...
Controller 0: gameforce_gamepad, GUID: 03000300, Mapping: 19000000030000000300000002030000,gameforce_gamepad,platform:Linux,leftstick:b14,rightx:a3,leftshoulder:b4,start:b9,lefty:a0,dpup:b10,righty:a2,a:b1,b:b0,guide:b16,dpdown:b11,rightshoulder:b5,righttrigger:b7,rightstick:b15,dpright:b13,x:b2,back:b8,leftx:a1,y:b3,dpleft:b12,lefttrigger:b6,
gpath.c:115: assertion 'filename != NULL' failed

gpath.c:115: assertion 'filename != NULL' failed

FNA3D Driver: OpenGL
OpenGL Renderer: Mali-G31
OpenGL Driver: OpenGL ES 3.2 v1.r13p0-01rel0.8747d14aec16ec21ac2b2da252c4ec5f
OpenGL Vendor: ARM
MojoShader Profile: glsles
Attempting to patch...
Failed to load Patch Object reference not set to an instance of an object
/usr/bin/tmntsr.sh: line 121:  2521 Killed                  mono ../MMLoader.exe MONOMODDED_${GAMEASSEMBLY} > >(tee ${GAMEDIR}/log.txt) 2>&1
```

djibux, 2022-12-17:
In #1047 the OP seems to be able to run the game with the same missing file, so I guess the _FileNotFoundException_ is irrelevant.  The script still needs fixing.

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like yours.

shantigilbert, 2022-12-17:
> ```shell
> # tar -xvzf /emuelec/configs/tmntsr.tar.xz -C /storage/roms/ports
> tar: invalid magic
> tar: short read
> ```

This is weird, I just tested this and for me it works as it should. 

```
EMUELEC:~ # tar -xvf /emuelec/configs/tmntsr.tar.xz -C /tmp/m
tmntsr/
tmntsr/FNARepacker.exe
...
tmntsr/savedata/
EMUELEC:~ # 
```

`/emuelec/configs/tmntsr.tar.xz` is saved in the SD card so I wonder if there is an issue with your SD card? because if a simple tar command is not working for you then there might be other issues? 

Also I see you are using a gameforce (please fill in the template next time) try removing this line https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec-ports/tmntsr/sources/tmntsr.sh#L78 and also the file `"${GAMEDIR}/gamedata/.astc_done"` if it exists. 

This port was intended to only be used on tvboxes, but the lack of checks was a mistake from me. I think it can run on the Gameforce, I just haven't tested it.

djibux, 2022-12-18:
Hello.
Yes I am on GameForce Chi (and sorry about the template).
`tar -xvf` works as well.  `tar -xvzf` does not.
I really doubt my SD card is corrupted.  It is rather new and the system is very stable.
I am not sure how to remove line 78 as the root file system is readonly.  A basic `mount -o rw,remount /` did not work.

shantigilbert, 2022-12-18:
> `tar -xvf` works as well. `tar -xvzf` does not.

Right, sorry I copied the command from my local copy. 

> I am not sure how to remove line 78 as the root file system is readonly. A basic `mount -o rw,remount /` did not work.

via ssh:

```
EMUELEC:~ # cp /usr/bin/tmntsr.sh /emuelec/bin
EMUELEC:~ # nano /emuelec/bin/tmntsr.sh 
```

You can also edit the file at `/usr/bin/ports/TMNTSR.sh` to make sure the `tmntsr.sh` in /emuelec/bin gets called directly, but it should not be necessary.

djibux, 2022-12-18:
Thanks.
It now goes much further.  I get the first two splashscreens and the video starts playing (with sound).  The image is very choppy and quickly degrades to just weird digital noise, but that’s ok.
The game crashes just after the video, or when I try to skip the video.

I’m happy to help debugging further if you think the Gameforce Chi has any chance of running the game.
Otherwise, don’t bother, it’s OK I can play it on PC :D

```
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.6 (GameForce.aarch64)
umount: can't unmount /storage/roms/ports/mono-6.12.0.122-aarch64.squashfs: Invalid argument
assemblyName: TMNT
appBasePath: /storage/roms/ports/tmntsr/gamedata
appArgs: {}
Redirecting ParisEngine to MONOMODDED_ParisEngine.dll
System.IO.FileNotFoundException: Could not find file "/emuelec/configs/Tribute Games/TMNT/ModSettings.dat"
File name: '/emuelec/configs/Tribute Games/TMNT/ModSettings.dat'
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode, System.IO.FileAccess access, System.IO.FileShare share, System.Int32 bufferSize, System.Boolean anonymous, System.IO.FileOptions options) [0x001ef] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at System.IO.FileStream..ctor (System.String path, System.IO.FileMode mode) [0x00000] in <232d2ec5aa2f429a82060c0ba99debd7>:0 
  at (wrapper remoting-invoke-with-check) System.IO.FileStream..ctor(string,System.IO.FileMode)
  at Paris.Game.Menu.Options.LoadModConfig () [0x00006] in <3f0176624b77491b8d88b7eee4f754ce>:0 
Params Set:
> None
Removing Program.CrashHandler handler...
Controller 0: gameforce_gamepad, GUID: 03000300, Mapping: 19000000030000000300000002030000,gameforce_gamepad,platform:Linux,leftstick:b14,rightx:a3,leftshoulder:b4,start:b9,lefty:a0,dpup:b10,righty:a2,a:b1,b:b0,guide:b16,dpdown:b11,rightshoulder:b5,righttrigger:b7,rightstick:b15,dpright:b13,x:b2,back:b8,leftx:a1,y:b3,dpleft:b12,lefttrigger:b6,
gpath.c:115: assertion 'filename != NULL' failed

gpath.c:115: assertion 'filename != NULL' failed

FNA3D Driver: OpenGL
OpenGL Renderer: Mali-G31
OpenGL Driver: OpenGL ES 3.2 v1.r13p0-01rel0.8747d14aec16ec21ac2b2da252c4ec5f
OpenGL Vendor: ARM
MojoShader Profile: glsles
Attempting to patch...
Failed to load Patch Object reference not set to an instance of an object
Instanciating "April_Attack_01" SFX (46448 bytes)
Instanciating "April_Attack_02" SFX (44696 bytes)
[…]
[…]
[…]
Instanciating "Zorax_Punch" SFX (117996 bytes)
Video /storage/roms/ports/tmntsr/gamedata/Content/Videos/TMNT_GameIntro.ogv does not have an XNB file! Hacking Duration property!

=================================================================
	Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

=================================================================
	Native stacktrace:
=================================================================
	0x5575a549d8 - mono : 
	0x5575a54ce4 - mono : 
	0x5575a0cb18 - mono : 
	0x55759d5a08 - mono : 
	0x7f90fa7680 - linux-vdso.so.1 : __kernel_rt_sigreturn
	0x7f90a8d8e4 - /usr/lib/libc.so.6 : 
	0x7f871e713c - /usr/lib/libSDL2-2.0.so.0 : 
	0x7f87d15754 - Unknown

=================================================================
	Telemetry Dumper:
[... 261 lines cut ...]
=================================================================
	Basic Fault Address Reporting
=================================================================
Memory around native instruction pointer (0x7f90a8d8e4):0x7f90a8d8d4  3f 78 00 f9 3e ff ff 17 1f 20 03 d5 05 40 00 91  ?x..>.... ...@..
0x7f90a8d8e4  a3 00 40 b9 e6 2f 80 52 7f 10 1e 72 63 00 06 0a  ..@../.R...rc...
0x7f90a8d8f4  c1 03 00 54 fe 0f 1f f8 e2 03 00 aa e4 03 01 2a  ...T...........*
0x7f90a8d904  63 03 00 35 5f 08 00 b9 84 02 00 35 a0 00 40 b9  c..5_......5..@.

=================================================================
	Managed Stacktrace:
=================================================================
	  at <unknown> <0xffffffff>
	  at FAudio:FAudioSourceVoice_GetState <0x00007>
	  at Microsoft.Xna.Framework.Audio.DynamicSoundEffectInstance:Update <0x0009f>
	  at Microsoft.Xna.Framework.FrameworkDispatcher:Update <0x00097>
	  at Paris.Engine.ParisEngine:frameworkUpdateWork <0x0000b>
	  at System.Threading.ThreadHelper:ThreadStart_Context <0x000bb>
	  at System.Threading.ExecutionContext:RunInternal <0x001e7>
	  at System.Threading.ExecutionContext:Run <0x0002b>
	  at System.Threading.ExecutionContext:Run <0x00057>
	  at System.Threading.ThreadHelper:ThreadStart <0x00033>
	  at System.Object:runtime_invoke_void__this__ <0x000eb>
=================================================================
/emuelec/bin/tmntsr.sh: line 121:  7097 Aborted                 mono ../MMLoader.exe MONOMODDED_${GAMEASSEMBLY} > >(tee ${GAMEDIR}/log.txt) 2>&1
```

shantigilbert, 2022-12-18:
The intro movie needs to be replaced by an empty one AFAIK

shantigilbert, 2022-12-19:
it should be under `gamedata/Content_hh`

```
cp -f "/storage/roms/ports/tmntsr/gamedata/Content_hh/Videos/TMNT_GameIntro.ogv" "/storage/roms/ports/tmntsr/gamedata/Content/Videos/TMNT_GameIntro.ogv"
```

djibux, 2022-12-19:
Isn't that done in the [install script](https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec-ports/tmntsr/sources/tmntsr.sh#L62) ?

shantigilbert, 2022-12-20:
@djibux not on the version you use (on v4. 6) check the commit date.

djibux, 2022-12-20:
Wow you're right. Sorry about that!

The game runs very well on the device, at least the first few minutes. I'll play some more this evening.

The default mapping is different than on PC (Xbox scheme) but fits the labels on the device butttons (Nintendo's scheme).

shantigilbert, 2022-12-20:
I will close this as fixed via https://github.com/EmuELEC/EmuELEC/commit/991bd30c6395e42ca8617b00384ae7d6d73606ce

closed 2022-12-20 by shantigilbert

djibux, 2022-12-20:
I can confirm that the game runs really well. Load times are longer than on PC, about 30 seconds, but other than that the experience is perfect.

Thanks for making it work!
-- 
djib