EmuELEC/EmuELEC #1082: Emuelec 4.6:Additional PSP system game load fail in ES

edouardlicn123, 2022-12-17:
I tried to add a PSP System with code below in es_systems.cfg

`  <system>
		<name>psp_zh</name>
		<fullname>Playstation Portable Chinese</fullname>
		<manufacturer>Sony</manufacturer>
		<release>2004</release>
		<hardware>portable</hardware>		
		<path>/storage/roms/psp_zh</path>
		<extension>.iso .ISO .cso .CSO .pbp .PBP</extension>
		<command>emuelecRunEmu.sh %ROM% -P%SYSTEM% --core=%CORE% --emulator=%EMULATOR% --controllers="%CONTROLLERSCONFIG%"</command>
		<platform>psp</platform>
		<theme>psp_zh</theme>
		<emulators>
			<emulator name="PPSSPPSDL">
                <cores>
                    <core default="true">PPSSPPSDL</core>
                </cores>
            </emulator>
			<emulator name="libretro">
                <cores>
                    <core>ppsspp</core>
                </cores>
            </emulator>
        </emulators>
	</system>`
Then the Emulator,both SDL and libretro will fail to load the game,with no issus in the log.And when I change the "name" tag back to "psp" it back to normal.It never happen in other systems and I think it should be a bug.

closed 2022-12-17 by shantigilbert

edouardlicn123, 2022-12-17:
Again Shanti feel this is NOT a bug.

shantigilbert, 2022-12-17:
It's not a bug, you are trying to add  a system that relies on other scripts to work. In this case emuelecrun.sh. As I mentioned several times before, use the forum or discord for help.

edouardlicn123, 2022-12-17:
> It's not a bug, you are trying to add a system that relies on other scripts to work. In this case emuelecrun.sh. As I mentioned several times before, use the forum or discord for help.

Other systems could change the "name" tag except the PSP,could you let anyone do something on it but just close the topic and let any other go into the trap again?

And could you please tell others the reason BEFORE you closed the topic?That is the way you always like.

shantigilbert, 2022-12-17:
> And could you please tell others the reason BEFORE you closed the topic?That is the way you always like.

https://github.com/EmuELEC/EmuELEC/issues/604

See I ask the same, could you read the pinned post BEFORE you post? it works both ways.