EmuELEC/EmuELEC #1083: PS5 Controllers Bluetooth error(br-connection-unknown) in 4.6

ChuLiqiang, 2022-12-17:
[NEW] Device D0:BC:C1:4C:D9:C4 Wireless Controller
[bluetooth]# trust D0:BC:C1:4C:D9:C4
[CHG] Device D0:BC:C1:4C:D9:C4 Trusted: yes
Changing D0:BC:C1:4C:D9:C4 trust succeeded
[bluetooth]# pair D0:BC:C1:4C:D9:C4
Attempting to pair with D0:BC:C1:4C:D9:C4
[CHG] Device D0:BC:C1:4C:D9:C4 Connected: yes
[CHG] Device D0:BC:C1:4C:D9:C4 Bonded: yes
[CHG] Device D0:BC:C1:4C:D9:C4 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device D0:BC:C1:4C:D9:C4 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device D0:BC:C1:4C:D9:C4 ServicesResolved: yes
[CHG] Device D0:BC:C1:4C:D9:C4 Paired: yes
Pairing successful
[CHG] Device D0:BC:C1:4C:D9:C4 ServicesResolved: no
[CHG] Device D0:BC:C1:4C:D9:C4 Connected: no
[bluetooth]# connect D0:BC:C1:4C:D9:C4
Attempting to connect to D0:BC:C1:4C:D9:C4
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[bluetooth]# connect D0:BC:C1:4C:D9:C4
Attempting to connect to D0:BC:C1:4C:D9:C4
[CHG] Device D0:BC:C1:4C:D9:C4 Connected: yes
Failed to connect: org.bluez.Error.Failed br-connection-unknown
[Wireless Controller]# connect D0:BC:C1:4C:D9:C4
Attempting to connect to D0:BC:C1:4C:D9:C4
Failed to connect: org.bluez.Error.Failed br-connection-unknown

edouardlicn123, 2022-12-17:
Sugget you to send info to the FORUM for Shanti will ALWAYS CLOSE THE TOPIC like you.

ChuLiqiang, 2022-12-22:
4.3 version is ok.

Langerz82, 2023-06-01:
1. SSH into your device.
2. Copy/paste the following commands one at a time:
```
cd /emuelec/scripts
wget https://gist.githubusercontent.com/Langerz82/2eae038ba57aecbb8d7b7c989e2d8fd5/raw/4f7d49f64106d27c2bcc8259c8887838c9d62adc/emuelec-bluetooth
chmod +x /emuelec/scripts/emuelec-bluetooth
```
3. Reboot then when the ES Loading Splash occurs power on your PS5 controller.
4. You may need to also on ES loaded and at least 30 seconds has past then plug in the PS5 Controller with a cable to your device.
5. Unplug and see if it pairs then.

Langerz82, 2023-06-03:
Please post in this thread if you have any Bluetooth Controller pairing issues:
https://github.com/EmuELEC/EmuELEC/issues/1194

closed 2023-06-03 by Langerz82