EmuELEC/EmuELEC #1085: bluetooth BUGs exist emuelec3. x-4.6

junm6802030, 2022-12-21:
1.Bluetooth controller BUG

/storage/cache/bluetooth
The file and folder users in the cache/bluetooth subdirectory are changed without reason. After the Bluetooth handle is successfully matched, you will see that the user and owner of the Bluetooth handle directory are both root. After the system is restarted and switched, the permissions of the directory and owner become 1023

After the system is switched on and off every time, if SSH goes in and uses the command hciconfig, it will be found that the Bluetooth controller of the box will directly jump around in hci0 and hci1. I don't know whether this causes the next problem

Solution:/storage/ Config Modify customer_ start.sh   The command chown - R root: root/storage
 
Because the Android system is too destructive and has modified too many permissions, it was originally only used to repair the permissions of the Bluetooth directory. When it was found that it was not working, it was completely repaired in a fit of pique.

2.Bluetooth handle sleeps, unable to wake up BUG

Emuelec The Bluetooth handle connection method is the official method. The official method lacks a command. If you do not use this command, you will lose a linkkey, and the handle cannot be connected after sleep.

command：pair “BTMAC”

The above BUGs exist emuelec3. x-4.6

rudboy62100, 2022-12-22:
Il y a pas que ça qui bug 
Le WiFi c'est pareille chez moi
Pas de Bluetooth pas de WiFi dégoûté

Langerz82, 2022-12-29:
@shantigilbert If the bluetooth is going to sleep we could change the service details to restart on stop with in:
packages/network/bluez/system.d/bluetooth.service
```
Restart=on-failure
```
too:
```
Restart=always
```

shantigilbert, 2022-12-30:
> @shantigilbert If the bluetooth is going to sleep we could change the service details to restart on stop with in: packages/network/bluez/system.d/bluetooth.service
> 
> ```
> Restart=on-failure
> ```
> 
> too:
> 
> ```
> Restart=always
> ```

I am not sure if it's going to sleep, I haven't really looked much into this. Have you tried this? 

The original issue has to do with Android it seems, so I am not sure how this could even be fixed if using Android while the same SD as EmuELEC is inserted. 

It cannot be permissions as you already know, EmuELEC is always running as ROOT. so even if the permissions changed, it should not matter, right?

Langerz82, 2023-01-06:
See Fixes:
https://github.com/EmuELEC/EmuELEC/issues/1076#issuecomment-1373208320

Langerz82, 2023-06-01:
Fixes read here onwards:
https://github.com/EmuELEC/EmuELEC/issues/1076#issuecomment-1568636815

Any issue post there.

closed 2023-06-01 by Langerz82