EmuELEC/EmuELEC #1094: SAMBA share "-" character in roms folders

b0b33140, 2023-01-01:
Hello,

I want to mount the pico-8 and tic-80 roms folders with the instructions from the wiki:
https://github.com/EmuELEC/EmuELEC/wiki/ROMS-on-CIFS-SAMBA-shares

file : storage-roms-pico-8.mount

```
[Unit]
Description=Mounting Pico8 share on samba
Requires=network-online.service
After=network-online.service
Before=emustation.service

[Mount]
What=//192.168.1.19/Stockage/emulation/roms/pico-8
Where=/storage/roms/pico-8
#The following option might not be required if your CIFS/SAMBA share is shared publicly without requiring any credentials) 
Options=username=b0b33140,password=Aurel33140!,vers=3.0
Type=cifs

[Install]
WantedBy=multi-user.target
```

file : storage-roms-tic-80.mount

```
[Unit]
Description=Mounting TIC80 share on samba
Requires=network-online.service
After=network-online.service
Before=emustation.service

[Mount]
What=//192.168.1.19/Stockage/emulation/roms/tic80
Where=/storage/roms/tic-80
#The following option might not be required if your CIFS/SAMBA share is shared publicly without requiring any credentials) 
Options=username=b0b33140,password=Aurel33140!,vers=3.0
Type=cifs

[Install]
WantedBy=multi-user.target
```

When I type the following command:
systemctl enable storage-roms-pico-8.mount && systemctl start storage-roms-pico-8.mount
the terminal gives me the following error:
Failed to start storage-roms-pico-8.mount: Unit storage-roms-pico-8.mount has a bad unit file setting.

I think the "-" character is problematic in the folder name.

7Ji, 2023-01-01:
The unit name should be `storage-roms-pico\x2d8.mount` instead. Several things to note here:
 
1. Several special characters including `-` have their specific usage in a systemd unit name and should be escaped: https://www.freedesktop.org/software/systemd/man/systemd.unit.html#String%20Escaping%20for%20Inclusion%20in%20Unit%20Names. The `-` in a mount unit is used to split path, so your `storage-roms-pico-8.mount` was for `/storage/roms/pico/8`, not `/storage/roms/pico-8`
2. This is a systemd usage problem and not an EmuELEC issue, the rule about the name applies to all modern Linux distros that use systemd as their init system. If you're using e.g. Ubuntu for your desktop, you could test about this there first.
3. If you're using >= 4.6 stable, you do not need to `enable` the mount unit, you just need to `start` it for test. All mount units will be handled by the new mount handler eemount so SAMBA shares could co-exist with external ROMs drive.

closed 2023-01-02 by shantigilbert

b0b33140, 2023-01-03:
It works, thanks.
I think a note should be added in the wiki.