EmuELEC/EmuELEC #1097: EmuELEC 4.6 > Duckstation > 'no such file or directory' when trying to run the game

sanyco92, 2023-01-06:
Hello, guys!
Installed EmuELEC 4.6, copied some games to PSX ROMs folder, added files to BIOS folder
For Playstation 1 changed the emulator to Duckstation. But when trying to run the game, getting the following error:

[image]

It looks like it's not installed or something. Can anyone help, how can I add Duckstation?
Unfortunately, I don't have wi-fi on my device, so everything I can do, is add files via microSD

Langerz82, 2023-01-06:
Did you do a clean install from scratch or upgrade? You need to mount the sd-card ext4 storage partition, using linux or an external windows app, and check that you have valid files in: /storage/.config/emuelec/configs/duckstation

sanyco92, 2023-01-06:
Yes, I did a clean install. 
Where can I get those duckstation files?

Langerz82, 2023-01-06:
they should automatically copy across, something doesn't seem right, maybe it was your installation process that error'd.

sanyco92, 2023-01-10:
Could anyone share the files that should be in 
 /storage/.config/emuelec/configs/duckstation
/storage/.local/share/duckstation

I'll try to put files in those directories. Maybe it will help. Because I don't know where to get these files. Thanks!

Langerz82, 2023-01-20:
Try this in SSH:
```
cp -r /usr/config/emuelec/configs/duckstation /emuelec/configs/duckstation
```

Langerz82, 2023-05-01:
Considering this fixed as others do not have this issue.

AngelTorresGarcia, 2023-07-24:
> @sanyco92 -
> Try this in SSH:
> ```
> cp -r /usr/config/emuelec/configs/duckstation /emuelec/configs/duckstation
> ```
> 

Hello!! After that what i have to do? It says press some commands but i don't know if i put the right command, can u help me?

AngelTorresGarcia, 2023-07-24:
[image]

Langerz82, 2023-07-25:
Whats is your device specifications? Also in ES, press start then go to EmuELEC settings >> danger zone >> Reset script and Binaries.
Did you update EmuELEC, or is this a clean image?

AngelTorresGarcia, 2023-07-25:
> Whats is your device specifications? Also in ES, press start then go to EmuELEC settings >> danger zone >> Reset script and Binaries.
> Did you update EmuELEC, or is this a clean image?
> 

Amlogic s905x3, I was configuring the emulator and I think leaving it in fullscreen was the main problem since it did start fine but the problem was after I had configured it

fantasygamessb, 2023-07-28:
do this in ES, press start then go to EmuELEC settings >> danger zone >> Reset script and Binaries.

AngelTorresGarcia, 2023-07-29:
> do this in ES, press start then go to EmuELEC settings >> danger zone >> Reset script and Binaries.

I already did it and it did not give me results, what I did was enter the local network from my cell phone and delete the duckstation folder, then I did emuelec on another sd card and copy the duckstation folder

Kahoned, 2024-01-09:
Hi!

I have the same problem
Can I put the directory manually?
Like I'm going to create a folder or is there a way to put this directory automatic?

Langerz82, 2024-01-31:
If you need to reset the settings, SSH in and run the following commands:
```
rm -r "/storage/.local/share/duckstation"
rm -r "/emuelec/configs/duckstation"
```
then re-run duckstation standalone.

Closing issue.

closed 2024-01-31 by Langerz82