EmuELEC/EmuELEC #1099: Daphne hypseus mapping issue. Emuelec 4.6

armstrongstreener, 2023-01-10:
Amlogic s905x3 4gb

Using dragonrise zero delay encoders. When in dragons lair or dragons lair 2 the sword button does not work. Tried remapping in the .ini file with no luck. All was working fine on emuelec 4.3.

Something has changed as having the same problem but with joysticks not mapping up and right in openbor

armstrongstreener, 2023-01-15:
Fixed by adding using .ini file from batocera

Langerz82, 2023-02-01:
Can you list the steps you took to fix the issue and maybe we can include the configs from batocera in EE for future versions, thanks.

armstrongstreener, 2023-02-02:
> 

Hi i replaced the hypseus.ini file using the one from batocera 3.5

Langerz82, 2023-02-02:
Have you got a link to the file you mentioned?

Langerz82, 2024-01-15:
test with the latest emuelec build and re-try. If it still doesn't work let me know here so I can re-open the issue.

closed 2024-01-15 by Langerz82