EmuELEC/EmuELEC #1100: (S905x3 running 4.6) Duckstation does not save controller and hotkey settings

soulbrix, 2023-01-12:
I updated my S905x3 box from 3.9 to 4.6 last night and it all went well and I like that DuckStation is now available, but it keeps forgetting my controller bindings and hotkey settings when I close it. All other options (display, for example) seem to be saved correctly, but not those. I wonder if it could be trying to write on a Read Only folder - do you know anything about this issue?

**EmuELEC version:**
 - 4.6
 - 
**Device (please complete the following information):**
 - SOC: S095x3
 - Brand:  Sunnzo
 - Model S9 Max

Langerz82, 2023-01-15:
Have you disabled the Duckstation controller auto config in emulationstation, emuelec settings, scrolldown to gamepad config >> disable auto config duckstation gamepad.

Langerz82, 2023-05-01:
No reply, closing issue. If it still persists, re-open.