EmuELEC/EmuELEC #1102: [Emuelec 4.6] Dolphin does not detect more than 2 controllers

mrabssalom, 2023-01-14:
**Describe the bug**
Dolphin does not recognize more than 2 controllers, tested with different controllers, connections (bluetooth and usb) and games

**To Reproduce**
Steps to reproduce the behavior:
1. Connect more than 2 controllers
2. Open any gamecube game
3. Only two controllers works as player 1 & 2

**Expected behavior**
All controllers conected must appear in the game

**EmuELEC version:**
 - Version 4.6

**Device (please complete the following information):**
 - SOC: s922x
 - Brand: Beelink
 - Model GT KING Pro

Langerz82, 2023-02-01:
Can you connect all the controllers you were using and try to connect with the auto controller config enabled and SSH in and get the output of:
```
cat /emuelec/logs/emuelec.log
```
and also the controllers configs in:
```
cat /proc/bus/input/devices
```
And I will try and figure out where the issue lies, thanks.

Langerz82, 2023-02-20:
@mrabssalom I would like to try and debug the issue, but I need more information...

shantigilbert, 2024-01-25:
Closing, no response

closed 2024-01-25 by shantigilbert