EmuELEC/EmuELEC #1107: EmuELEC 4.6 > When try to load some console games

harvic3, 2023-01-20:
Hi guys, I have a retro console with different game roms that was on EmuELEC version 4.5 and everything worked fine, but after applying an automatic update to version 4.6 some consoles stopped working and get error messages saying that libraries are missing to run the games, for example the one in this picture below trying to load a `DreamCast` game, but this happens with `PS1`, `PSP` and others. 
Nintendo, Family and basic consoles like this work fine!!

[image]

Transcription of the image for when it is deleted:
```console
Run Command is: 
retroarch32 -v -L /tmp/cores/flycast_32b_libreto.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/<CONSOLE>/<GAME_NAME> 
Emulator Output is: 
retroarch32: /emulec/lib32/libc.so.6: version 'GLIBC_2.34' not found (required by retroarch32)
retroarch32: /emulec/lib32/libc.so.6: version 'GLIBC_2.33' not found (required by retroarch32)
retroarch32: /emulec/lib32/libc++.so.6: version 'GLIBCXX_3.4.29' not found (required by retroarch32)
exit 1
exit bios 0
```

So, I trying to find these libraries in repos and put it into /emulec/lib32 but it not works, can any one help us with this issue? Ç
Thanks!!

shantigilbert, 2023-01-20:
Please tee the wiki

closed 2023-01-20 by shantigilbert

meynon007, 2023-02-03:
I looked at the wiki as you suggested "How to update" and it just talks to how to manually update EmuELEC. Can you please be a little more specific on how we go about resolving this specific issue? 

As when I ran the auto-update on my KinHank Super Console X3 Plus from 4.5 -> 4.6 it appeared to work flawlessly. The error It looks like the EmuELEC is looking for an older version of a libs GLIBC_2.34,  GLIBC_2.33 and GLIBCXX_3.4.29 for these emulators which may have been updated to a newer lib versions in 4.6.

Please any additional information would be greatly appreciated as I has only had my KinHank Super Console X3 Plus 3 days and seem to have bricked at least 3 of the best emulators (Dreamcast, PS1 and PSP.   : (

Run Command is: 
retroarch32 -v -L /tmp/cores/flycast_32b_libreto.so --config /storage/.config/retroarch/retroarch.cfg /storage/roms/<CONSOLE>/<GAME_NAME> 
Emulator Output is: 
retroarch32: /emulec/lib32/libc.so.6: version 'GLIBC_2.34' not found (required by retroarch32)
retroarch32: /emulec/lib32/libc.so.6: version 'GLIBC_2.33' not found (required by retroarch32)
retroarch32: /emulec/lib32/libc++.so.6: version 'GLIBCXX_3.4.29' not found (required by retroarch32)
exit 1
exit bios 0

shantigilbert, 2023-02-03:
Pleas read the wiki on supported devices, we do not support devices that come with emuelec pre-installed.