EmuELEC/EmuELEC #1110: Pico-8 won’t work since I tried installing the latest version

djibux, 2023-01-22:
**Describe the bug**
I tried installing the latest version of Pico-8 (0.2.5e) and Pico-8 now refuses to start: it goes back to emulationstation.  Reverting to previous versions does not solve the issue (I tried 2.5d, 2.4c, 2.3, 2.2 then gave up). 

**To Reproduce**
Steps to reproduce the behavior:
1. Install a version of Pico-8 as explained [here](https://github.com/EmuELEC/EmuELEC/wiki/PICO-8-Setup-Guide)
2. Upgrade to the latest version using the same process.
3. Pico-8 won’t start anymore.

**Expected behavior**
I would expect Pico-8 to start.

**Log**
I get the following 
```
EmuELEC Run Log
##############################################
#                 EmuELEC                    #
#     Retro emulation for Amlogic Devices    #
#             Based on CoreELEC              #
##############################################
EmuELEC (Official - EmuELEC is FREE! NOT FOR SALE!): 4.6 (GameForce.aarch64)
PLATFORM: pico8
ROM NAME: /storage/roms/pico-8/splore.p8
BASE ROM NAME: splore.p8
USING CONFIG: /storage/.config/retroarch/retroarch.cfg
1st Argument: /storage/roms/pico-8/splore.p8
2nd Argument: -Ppico8
3rd Argument: --core=Pico-8
4th Argument: --emulator=Pico-8
Full arguments: /storage/roms/pico-8/splore.p8 -Ppico8 --core=Pico-8 --emulator=Pico-8 --controllers=-p1index 0 -p1guid 19000000030000000300000002030000 
Run Command is:
/usr/bin/bash pico8.sh /storage/roms/pico-8/splore.p8
Emulator Output is:
/emuelec/bin/pico-8/pico8_dyn: symbol lookup error: /emuelec/lib32/libpthread.so.0: undefined symbol: _dl_make_stack_executable, version GLIBC_PRIVATE
exit 0
```

**EmuELEC version:**
 - Version 4.6

**Device:**
 Gameforce Chi

miwasp, 2023-01-22:
In a fresh installation of version 4.6 the dir /emuelec/lib32 is empty. So your installation must be an update of an older version.
1. You have to „Reset scripts and binaries“
2. Pico-8 version 0.2.5e has a bug and does not work on several systems. Either wait for 0.2.5f or use 0.2.5c

djibux, 2023-01-22:
Spot on!  Thanks.
As a side effect, resetting scripts and binaries removed my custom collections, but I had a quite recent backup.

closed 2023-01-22 by djibux