EmuELEC/EmuELEC #1111: Wifi hotspot name

alabotski, 2023-01-28:
**Describe the bug**
I have Wi-Fi with name — **¯\_(ツ)_/¯ 5 GHz**
CoreELEC & Android TV work fines with it
Problems only with EmuELEC 4.5 & 4.6
It can't connect

Temporarily I created a hotspot on my phone and EmuELEC without spaces connected to it
Apparently there are problems in the name of the Wi-Fi point?
But why does CoreELEC work with it without difficulties?

**Expected behavior**
Set up a connection with my home wifi hotsopt

**EmuELEC version:**
 - Version 4.5
 - Version 4.6

**Device (please complete the following information):**
 - SOC: S095x3
 - Brand: KinHank
 - Model: Super Console X3 Plus

Langerz82, 2023-02-01:
The smiley character isnt part of ASCII, so try removing that and it should work.

alabotski, 2023-02-02:
The rest of the systems work fine, the same CoreElec

Langerz82, 2023-02-03:
The password is saved in a plain text file which does not support other languages and that code is part of Japanese text. You will need to remove it for it to work, this isnt really a bug it's more a feature request.

Langerz82, 2023-04-17:
Closing as not really a bug, but a feature and use discourse forum for features.

closed 2023-04-17 by Langerz82

shantigilbert, 2023-04-17:
it is not a bug, escape your special chars with a backslash : `$ => \$` `\ => \\` etc you might need to manually edit emuelec.conf