EmuELEC/EmuELEC #1114: No Wifi Networks Found - S905x4 - EmuELEC 4.6

gheatgheat, 2023-01-31:
**Describe the bug**
I can't find the wifi network. I have reset many times but the device still says "No Wifi Networks Found". I also tried plugging in the network cable, but the device still say no network.

**Expected behavior**
Find the wifi network and connect to it

**Screenshots**
[image]

**EmuELEC version:**
 - Version 4.6

**Device (please complete the following information):**
 - SOC: S905x4
 - Brand: VONTAR
 - Model: X96X4
 - Ram 4GB
 - LAN 100Mb

Langerz82, 2023-02-01:
Have you tried also re-booting your router? Sometimes channel interference can prevent the wifi being picked up. Make sure its also broadcasting its SSID.

gheatgheat, 2023-02-01:
> Have you tried also re-booting your router? Sometimes channel interference can prevent the wifi being picked up. Make sure its also broadcasting its SSID.

I tried rebooting the router, tried plugging in the network cable and using the phone to broadcast wifi. All are not working, it still says no wifi net works found. 

My device is using ATV (android television) firmware not default firmware. I'm trying to reinstall the default firmware hoping emuELEC will connect to wifi.

Note: My android box with ATV firmware is still connecting to the wifi network

Langerz82, 2023-02-01:
What DTB file are you using? If you are using the 1GBit network trying the 100mbit dtb file.

gheatgheat, 2023-02-01:
> What DTB file are you using? If you are using the 1GBit network trying the 100mbit dtb file.

I use "sc2_s905x4_4g.dtb" then also try the "sc2_s905x4_4g_1gbit.dtb" version. As a result, both can't connect to wifi

Langerz82, 2023-02-01:
You could try using an earlier version like 4.3 and see if you have any luck with the BT. It works it's most likely driver support for the newer CoreElec version.

gheatgheat, 2023-02-01:
> You could try using an earlier version like 4.3 and see if you have any luck with the BT. If it works it's most likely driver support for the newer CoreElec version.

SOC: S905x4 is only supported on version 4.6 :((

rudboy62100, 2023-04-12:
bonjour avez vous trouvé la solution ?

badar47, 2023-05-29:
i'h same problem with my box A95X F4 there is no wifi and bluetooth works

Langerz82, 2023-06-03:
You can try these possibly updated dtb files, it's the latest dtb files from CoreELEC-20.1 Nexus.
[device_trees.zip](https://github.com/EmuELEC/EmuELEC/files/11642704/device_trees.zip)

GMGuimaraes, 2023-06-23:
I have the same problem but with wifi on GameStick GD10
[image]

SkillGodAk, 2023-09-20:
> You can try these possibly updated dtb files, it's the latest dtb files from CoreELEC-20.1 Nexus. [device_trees.zip](https://github.com/EmuELEC/EmuELEC/files/11642704/device_trees.zip)

Where did CoreELEC find device_trees.zip? I want to find the latest version of dtb.img installation

Langerz82, 2023-09-20:
compile coreelec-20 then get the dtb's from the img output file.
https://github.com/CoreELEC/CoreELEC/tree/coreelec-20

SkillGodAk, 2023-09-20:
> 編譯 coreelec-20，然後從 img 輸出文件中獲取 dtb。 https://github.com/CoreELEC/CoreELEC/tree/coreelec-20

thanks

closed 2023-11-07 by shantigilbert

rpascalegh, 2024-01-05:
TP link TL-WN725N  Just plug it in and your done.  Emuelec 4.6 found wifi no problem.