EmuELEC/EmuELEC #1122: Build failed

kylongmu, 2023-02-08:
Follow the guide, clone the code and build with "PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC make image", but failed with this error.  host is ubuntu 20.04

checking rpc/rpc.h usability... no
checking rpc/rpc.h presence... no
checking for rpc/rpc.h... no
configure: error: Header file rpc/rpc.h not found - maybe try building with --enable-tirpc
FAILURE: scripts/build nfs-utils:host during configure_host (default)
*********** FAILED COMMAND ***********
${PKG_CONFIGURE_SCRIPT} ${HOST_CONFIGURE_OPTS} ${PKG_CONFIGURE_OPTS_HOST}
**************************************
FAILURE: scripts/build nfs-utils:host has failed!

shantigilbert, 2023-02-08:
did you follow these? 

https://github.com/EmuELEC/EmuELEC#build-prerequisites

https://github.com/EmuELEC/EmuELEC/wiki/DEV:-How-to-compile-toolchain-only#building-the-toolchain
instead of `build toolchain` use `make image`

kylongmu, 2023-02-08:
Yes, I followed with https://github.com/EmuELEC/EmuELEC#build-prerequisites

7Ji, 2023-02-08:
`nfs-utils:host` is a host package and it depends things on host. In this case the `rpc.h` is missing. Search with your distro's package manager to check which package provides it, e.g. `apt-file search rpc.h`. 

The build instructions really need to be updated considering how many missing deps are there...

kylongmu, 2023-02-08:
> `nfs-utils:host` is a host package and it depends things on host. In this case the `rpc.h` is missing. Search with your distro's package manager to check which package provides it, e.g. `apt-file [...]

~/EmuELEC$ apt-file search rpc/rpc.h
dietlibc-dev: /usr/include/diet/rpc/rpc.h 
krb5-multidev: /usr/include/mit-krb5/gssrpc/rpc.h
libkrb5-dev: /usr/include/gssrpc/rpc.h
libntirpc-dev: /usr/include/ntirpc/rpc/rpc.h
libtirpc-dev: /usr/include/tirpc/rpc/rpc.h

I installed all the 5 packages, but the problem is same.

shantigilbert, 2023-02-08:
did you tried what is suggested? `- maybe try building with --enable-tirpc`

7Ji, 2023-02-08:
Did a quick build with latest commit and nothing wrong happened. Maybe you could bump your local tree and try again?

kylongmu, 2023-02-08:
> nfs-utils

> did you tried what is suggested? `- maybe try building with --enable-tirpc`
Yes, but failed also, I'll pull again with the latest commit.

7Ji, 2023-02-11:
Now I think about it, are you building under Ubuntu20.04? If you're using Debian or Ubuntu22.04 or anything other than Ubuntu 20.04 then this is a known problem we found somewhere last year. 

It should also be the cause for https://github.com/EmuELEC/EmuELEC/issues/1125

AFAIK there's no plan bumping the expected build env to Ubuntu 22 in upstream LibreELEC.

The solution would be to switch your distro. Yeah it sounds stupid and I really don't like to keep Ubuntu 20 around. But for me,  I just got tired of fixing these wierd host library lookup issues after numerous build failures on my Arch server and put everything LE-derived into a Ubuntu20.04 VM.

kylongmu, 2023-02-11:
> Now I think about it, are you building under Ubuntu20.04? If you're using Debian or Ubuntu22.04 or anything other than Ubuntu 20.04 then this is a known problem we found somewhere last year.
>
> It [...]

I tried with ubuntu20.04, ubuntu22.04, debian10, debian11, no one can success.
And now I got some new problem with the new pull, I issued it.

shantigilbert, 2023-02-11:
> AFAIK there's no plan bumping the expected build env to Ubuntu 22 in upstream LibreELEC.

We don't need to follow that step, I compiled EmuELEC on Ubuntu 22.04 before (not recent) and I just had to patch a few files, but it was not hard nor impossible.

kylongmu, 2023-02-11:
> > AFAIK there's no plan bumping the expected build env to Ubuntu 22 in upstream LibreELEC.
> 
> We don't need to follow that step, I compiled EmuELEC on Ubuntu 22.04 before (not recent) and I just had to patch a few files, but it was not hard nor impossible.

Can you release your patch to a docker image? so we can follow it to build.

kylongmu, 2023-02-11:
I'll try again with this docker image :https://hub.docker.com/r/coreelec/coreelec-builder

shantigilbert, 2023-02-11:
> Can you release your patch to a docker image?

I have never used docker, sorry. and like I said this was a while ago, just testing, I don't think I saved the patches, sorry, was just making a comment that it is possible.

kylongmu, 2023-02-15:
The package "qt-everywhere-5.15.0" need gcc g++ version 10 to compile, the ubuntu22 default install gcc-11.
Try to  "apt remove gcc-11 g++-11" and "apt install gcc-10 g++-10" can solve this problem

shantigilbert, 2023-02-23:
Pull latest changes, I just build several images using Ubuntu 22.

closed 2023-02-23 by shantigilbert