EmuELEC/EmuELEC #1124: Support Xbox Wireless Adapter

pureair, 2023-02-10:
**Is your feature request related to a problem? Please describe.**
Xbox Wireless Adapter (dongle for first-gen xbox one controller that has no bluetooth) is not supported.
After inserting the adapter, dmesg reads:
```
[  813.316853@0]- usb 1-1.3: new high-speed USB device number 4 using xhci-hcd
[  813.465942@0]- usb 1-1.3: New USB device found, idVendor=045e, idProduct=02fe
[  813.465951@0]- usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  813.465954@0]- usb 1-1.3: Product: XBOX ACC
[  813.465957@0]- usb 1-1.3: Manufacturer: Microsoft Inc.
[  813.465960@0]- usb 1-1.3: SerialNumber: 441051
[  813.467495@0]- usb 1-1.3: Unsupported device
```

**Describe the solution you'd like**
Ship EmuELEC with the [xone kernel driver](https://github.com/medusalix/xone) which provides support to the xbox wireless adapter.

**Additional context**
Users can not install the driver by themselves because the install script requires DKMS which is not present in EmuELEC. It also requires Linux kernel 4.15+.

Langerz82, 2023-02-16:
hmm it's really up to CoreELEC if they want to update Linux Kernels which may take a long time as we are using linux 4.9 atm even though we are using a newer CoreELEC. Have you found any other driver details which can support older kernels?
Might be waiting a long time, consider buying some PS3 like Controllers to use.

7Ji, 2023-02-28:
Langerz82... We're using vendor kernel for Amlogic devices and it's not an easy upgrade both for us and for CoreELEC. 

Amlogic as a SoC vendor aimed at Android market follows the Android kernel route and not the Linux kernel route, they only choose the "stablest" Android kernel when they design a new family of SoCs to add their stuffs and stick to that until Google forces them to use newer kernel by raising the minimum kernel requirement for newer Android version.

What that ends up is giant gaps between their kernels, so far the major versions CE has used are only 3.14, 4.9 and 5.4, in which 5.4 currently only supports SoCs from SC2 (S905X4) up. 

Since that driver requires kernel 4.15+, we will either need to maintain it for -ne project for post-SC2 devices (but I guess OP is not using a post-SC2 device) which just adds more burden to maintainance and really does not give much gain for our main purpose emulation, or try to backport that driver and dependencies back to 4.9.

So OP.. You will not be able to use your receiver on EmuELEC on Amlogic platform in short time until someone here backports that driver. I don't have time for that now so possibly not me.

elflayu, 2023-11-24:
Hi devs. It would be great if you could add xow as it was in previous versions, or give a chance to users to add it by their own. This would be a great solutions to keep using our xbox controllers. Thanks a lot!

Langerz82, 2024-02-03:
https://github.com/EmuELEC/EmuELEC/blob/v4.6/packages/sx05re/emuelec/package.mk#L23
change:
```
PKG_DEPENDS_TARGET+=" $PKG_TOOLS $PKG_EMUS $PKG_EXPERIMENTAL emuelec-ports"
```
to add xow
```
PKG_DEPENDS_TARGET+=" $PKG_TOOLS $PKG_EMUS $PKG_EXPERIMENTAL emuelec-ports xow"
```
then build your emuelec image.

Langerz82, 2024-02-03:
try this diff file and rebuild the image:
[xbox.txt](https://github.com/EmuELEC/EmuELEC/files/14151776/xbox.txt)

shantigilbert, 2024-02-03:
I think the right fix would be to upgrade to https://github.com/medusalix/xone but both require kernel 5.x so I doubt it will work on EE

Langerz82, 2024-02-06:
Xow is included in the latest build to run it, in SSH add the following line in /storage/.config/custom_start.sh
https://github.com/EmuELEC/EmuELEC/blob/dev/packages/sx05re/emuelec/config/custom_start.sh#L13

```
xow &
```

closed 2024-02-26 by shantigilbert