EmuELEC/EmuELEC #1135: Brightness slider very slow to change

gingerbeardman, 2023-02-15:
**Describe the bug**
Brightness slider very slow to change

**To Reproduce**
Steps to reproduce the behavior:
1. Start > System Settings > Brightness
2. Use dpad left/right to change

**Expected behavior**
Should be a quick process

**Screenshots**

https://imgur.com/a/sxR8Q30

**EmuELEC version:**
 - Version: 4.6

**Device (please complete the following information):**
 - SOC: RK3326
 - Brand: GameForce
 - Model: Chi
 - Version: n/a

**Additional context**
- Currently after playing brightness is maxed out at 100% so I'm having to use this slider a lot

Langerz82, 2023-02-16:
It sends a command to change it every time the value changes. Would you prefer it to be a dropdown control in increments of say 10% and when the value changes it sets the value?

gingerbeardman, 2023-02-16:
Increments of 10% would be good for me. Would it need to be a drop down? Would this slider still be slow if there were only 10 steps compared to 100?

Could it perhaps not send the command until the slider remains at the same value for more than a fraction of a second? But then it would lose realtime updating (not that it is realtime now).

Langerz82, 2023-02-17:
Good ideas, Increments of 10 would probably be the easiest way. And if it still causes lock ups perhaps we make the command async so it doesnt lock up the screen or find a better way to handle it. You mentioned you would also like it to apply to retroarch's screen brightness do you still want that to occur or not. Let me know and I can do the fixes.

gingerbeardman, 2023-02-18:
Thanks.

Not sure about RetroArch, what are your thoughts?

Langerz82, 2023-02-18:
If you have to change the brightness again for the backlight when in retroarch I think it's worth matching the values to keep sync.
First we try and fix why brightness is not saving in Gameforce, and if it poses an issue the retroarch config not matching the ES config then we can sync those values together (it would only be 1-line of code so it's an easy fix).
ODroidGoAdvance is working with brightness, except the shortcuts don't work. So I think it's just a package config issue that the Gameforce is not working as well.

gingerbeardman, 2023-02-19:
OK, I agree (I think!)

One other thing to note, it was discovered in #1137 that the hotkeys change in steps of 5% so let's keep with that.

closed 2023-02-19 by gingerbeardman