EmuELEC/EmuELEC #1136: Brightness set to 100% after RetroArch

gingerbeardman, 2023-02-15:
**Describe the bug**
Brightness set to 100% after RetroArch

**To Reproduce**
Steps to reproduce the behavior:
1. Start > System Settings > Brightness > 10%
2. Run a game using RA
3. Quit
4. Brightness is now 100%

**Expected behavior**
Brightness should remain as set and not change unexpectedly 

**Screenshots**
N/A

**EmuELEC version:**
 - Version: 4.6

**Device (please complete the following information):**
 - SOC: RK3326
 - Brand: GameForce
 - Model: Chi
 - Version: n/a

**Additional context**
-

gingerbeardman, 2023-02-15:
Resetting RetroArch config did nothing

Langerz82, 2023-02-16:
Try in RA Settings >> Configuration >> Save on Exit.

Where are you changing brightness? I couldn't find the option in the main menu.

In /storage/.config/retroarch/retroarch.cfg you can manually change the number under:
```
screen_brightness = "100"
```

Langerz82, 2023-02-16:
It seems it might reset in the code, as AmberELEC has a patch to disable the reset here:
https://github.com/AmberELEC/AmberELEC/blob/dev/packages/games/tools/retroarch/patches/retroarch-disable_brightness_setting.patch

gingerbeardman, 2023-02-16:
> Where are you changing brightness? I couldn't find the option in the main menu.

In ES: Start > System Settings > Brightness

Sadly none of your suggestions offer the solution.

Langerz82, 2023-02-17:
@shantigilbert 
Will probably have to look into this one, because we usually have a patch that should disable the resetting of screen brightness located here:
packages/sx05re/libretro/retroarch/patches/OdroidGoAdvance/9999-emuelec-unify-brightness.patch

Gameforce should also use these patches. Maybe they are not applying properly.

Langerz82, 2023-02-17:
I used an similar Odroid-Go Advance device and the brightness settings were being applied. Shanti might have to take a look at why Gameforce is not working the same.

miwasp, 2023-02-18:
The patch still works fine for oga/ogs and gf, it just isn‘t applied to the 32bit version of retroarch. This needs to be fixed.

gingerbeardman, 2023-02-18:
Aha! That would do it.

Only reason I am using it is that it's the only version that I could get to run Beetle Adventure Racing

closed 2023-02-19 by gingerbeardman