EmuELEC/EmuELEC #1137: Brightness keys not mapped on GameForce Chi

gingerbeardman, 2023-02-15:
**Is your feature request related to a problem? Please describe.**

Brightness keys not mapped on GameForce Chi

- #1135
- #1136

**Describe the solution you'd like**
Additional key mappings
- L3 + L1 brightness-
- L3 + R1 brightness+

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Langerz82, 2023-02-16:
hmm, I believe retroarch by default does not have shortcuts for it to be changed, so this will be a feature request. We would have to make an external listener that waits for input and changes it accordingly. Most likely gptokeyb.
I can fix the other issues, but this might have to be put on hold, unless someone else decides to implement the feature.

Langerz82, 2023-02-17:
You are right sorry, my bad.

I have an Odroid Go Advance like device so I'll see if it has the same issue.

miwasp, 2023-02-18:
Setting brightness is mapped to GHK + left/right. GHK on GF is „1“. This works but no overlay is displayed.

What is L3?

gingerbeardman, 2023-02-18:
Ah! On GameForce Chi the lower left rectangular face buttons are marked 1 & 2.

I incorrectly assumed GHK was the Chi home button.

I can confirm that pressing GHL "1" and left/right changes brightness in both ES and RA.

So, I consider this closed! 

thanks

closed 2023-02-18 by gingerbeardman

Cathodeus, 2023-03-15:
@gingerbeardman Your assumption was right the "User Manual" says that GHK button is the Home Button, And pressing it with left/right should drive the brightness. But it doesn't work !!! We have to press another button ... the "1" button ... and as it isn't what says the user manual than it is confusing !!! I really would like to be hable to drive beightness with the good button !

Cathodeus, 2023-03-15:
This issue should be re opened. Based on my previous post explanations i guess ...

shantigilbert, 2023-03-15:
> This issue should be re opened. Based on my previous post explanations i guess ...

No, we do not make the GameForce, and we are not responsible for the manual, we however have the hotkeys listed here: 

https://github.com/EmuELEC/EmuELEC/wiki/HotKeys-for-Odroid-Go-Advance,-Odroid-Go-Super-and-GameForce-Chi

and those are the ones that will stay.

Cathodeus, 2023-03-16:
Hello, 
Thanks for your informations. I totally agree with those shortcuts, we're not asking to change them.

Here is a picture to clarify and maybe find a solution together :
Please have a look
[(https://www.dropbox.com/s/0rg5kwifjgra5ai/2023-03-16_11-23-45.pdf?dl=0)]

What i'm trying to understand here is : 

- Why do all these shortcuts works with the home button wich is the official GHK button from user manual, while only the brightness shortcuts doesn't ... instead of changing the screen brightness it change the save state slot.

Couple additional informations/questions :

-Why do Emuelec do not use the same button as the one in the user manual.
-Is there a way user wise to re assign the GHK button to the official one.
-Why do all shortcuts works with the official button, but the brightness screen one do not ...