EmuELEC/EmuELEC #1146: WiFi/BT module 5BK does not work in ODROID-M1

Pyohwan, 2023-02-26:
**Describe the bug**
A clear and concise description of what the bug is.
WiFi/BT module 5BK does not work in ODROID-M1.

**To Reproduce**
After running EmuELEC, plug WiFi/BT module 5BK into the device.
Alternatively, run EmeuELEC after plugging WiFi/BT module 5BK into the device.

**Expected behavior**
I expect the network connection and bluetooth to work.

**Screenshots**
There are no screenshots because I can't connect to WiFi and Bluetooth.

**EmuELEC version:**
 - EmuELEC v.4.6

**Device (please complete the following information):**
 - SOC: Rockchip RK3568B2
 - Brand: ODROID
 - Model ODROID-M1, WiFi Module 5BK
 - Version None

**Additional context**
Add any other context about the problem here.

There is a workaround on the Odroid forum.
https://forum.odroid.com/viewtopic.php?f=212&t=45114

shantigilbert, 2023-03-09:
I think your link is wrong, can you post it again, or the workaround?

Pyohwan, 2023-03-12:
I've never build emuelec, so I may be inaccurate.
It need an OS provided by the HARDKERNEL.
https://wiki.odroid.com/odroid-m1/os_images/ubuntu/ubuntu
https://forum.odroid.com/viewtopic.php?f=217&t=44346

```
sudo apt update
sudo apt install linux-firmware
```
I think EmuELEC image should include linux-firmware.
After installing linux-firmware, WiFi Module 5BK works normally.

WiFi Module 5BK was created by HARDKERNEL.
https://www.hardkernel.com/shop/wifi-module-5bk/

shantigilbert, 2023-03-17:
So its just a matter of updating the firmware? have you checked with logs that this is in fact the issue?

Pyohwan, 2026-02-21:
@road-t 
I flashed the X96X6 image from the link you provided to a microSD card and plugged it into my odroid-m1.
But it wouldn't even boot. This is probably because it's not an image for the ODROID-M1 (RK3568).

Referring to your commit, I don't know how to create EmuELEC image by adding RTL8821CU driver.
If you could give me the ODROID-M1 image, I'd be happy to test it.

But aren't the RTL8822CS and RTL8821CU similar but different chipsets? You included the RTL8822CS driver in your image, but I'd like to check for myself if it works with the RTL8821CU Wi-Fi dongle.

Unfortunately, this issue hasn't been resolved yet.

road-t, 2026-02-21:
> [@road-t](https://github.com/road-t) I flashed the X96X6 image from the link you provided to a microSD card and plugged it into my odroid-m1. But it wouldn't even boot. This is probably because it's not an image for the ODROID-M1 (RK3568).

The image would work, but the dtb is wrong. If you edit the boot config and place your working dtb in the microSD root, it will boot for sure.

> Referring to your commit, I don't know how to create EmuELEC image by adding RTL8821CU driver. If you could give me the ODROID-M1 image, I'd be happy to test it.

I can try to make a build with RTL8821CU driver, but if you give me your correct dtb, it helps a lot since I don't have an Odroid M1 here.

Pyohwan, 2026-02-22:
[rk3568-odroid-m1.dtb.tar.gz](https://github.com/user-attachments/files/25465014/rk3568-odroid-m1.dtb.tar.gz)

After copying the .dtb file from the EmuELEC v4.8 ODROID-M1 image to the X96X6 image and modifying the configuration file, EmuELEC booted successfully. However, the RTL8821CU Wi-Fi/BT dongle didn't work.

I uploaded the rk3568-odroid-m1.dtb file, which is included in the EmuELEC v4.8 ODROID-M1 image.

Please add the RTL8821CU driver and let me know.

road-t, 2026-02-22:
Damn. I've missed that you were talking about a DONGLE. In this case there's no need to edit dts (dtb) nor adding 8821CU drivers. There should be some problem in USB detection or the module initialization. It's a bit hard to debug w/o the device(