EmuELEC/EmuELEC #1155: pocketcdg plays only distorted noise

worstcase-scenario, 2023-03-24:
EMUELEC 4.6 (Amlogic-ng.aarch64)
Machine model: Amlogic

When I’m trying to run a .cdg-file, pocketcdg is being launched via retroarch, but instead of playing the mp3-music, it only produces kind of a distorted, oldschool-modem-white noise… (while showing the karaoke text and all). 

This happens both when starting the .cdg-file from a gamelist in Emulationstation and directly loading the file in retroarch. 

When I remember correctly, this issue didn't happen on EmuElec 4.5... has there been an update / change to the pocketcdg-core? 

I'd be glad for any ideas, suggestions and solutions! Thanks.

shantigilbert, 2023-03-25:
If the Retroarch volume is too high (above 100%) please try it with the volume around 70%

worstcase-scenario, 2023-03-25:
The only audio-option I can find in retroarch is under Audio -- Volume Gain (dB)
And that is set on 0.0 - so normal, right?

Should I change this number?

shantigilbert, 2023-03-27:
Maybe try copying the Pocketcdg core from 4.,5 to 4.6 (to /tmp/cores) to see if it's a core update issue.

shantigilbert, 2024-01-23:
Is this still an issue?

worstcase-scenario, 2024-01-24:
I would have to try it out again, after discovering EmuELEC has mplayer, I deleted the pocketcd-roms...

shantigilbert, 2024-01-24:
confirmed working

closed 2024-01-24 by shantigilbert