EmuELEC/EmuELEC #1167: EMUELEC RK3566 GREEN SCREEN

CODEJZX, 2023-04-18:
After flashing the latest EMUELEC 4.6 version （[EmuELEC-RK356x.aarch64-4.6-rk356x.img.gz](https://github.com/EmuELEC/EmuELEC/releases/download/v4.6/EmuELEC-RK356x.aarch64-4.6-rk356x.img.gz)） on the RK3566 model, the screen turns green and I cannot enter the system. I have tried three different TF cards and the same thing happens. What could be the reason for this?
The device is firefly station M2，HELP！

worstcase-scenario, 2023-04-18:
Could be a problem with the HDMI resolution? Try to change it on your TV...

CODEJZX, 2023-04-18:
> Could be a problem with the HDMI resolution? Try to change it on your TV...
I have tried three different TV and Display screens，the same thing happens.

CODEJZX, 2023-04-18:
Have none of you encountered this problem before?

CODEJZX, 2023-04-18:
> I have tried three different TV and Display screens，the same thing happens.

I have tried three different TV and Display screens，the same thing happens.

worstcase-scenario, 2023-04-18:
Do you have the right dtb.img? The box doesn't seem to boot into loading/installing Emuelec. Did you do all the steps from the Wiki? https://github.com/EmuELEC/EmuELEC/wiki/How-to-Install

worstcase-scenario, 2023-04-18:
Sorry, I have another box. One last idea: Did you try to use the generic EmuELEC-image?
https://github.com/EmuELEC/EmuELEC/releases/download/v4.6/EmuELEC-Amlogic-ng.aarch64-4.6-Generic.img.gz

Perhaps this will work on your box...

CODEJZX, 2023-04-18:
> Sorry, I have another box.
> 
> Have you tried rebooting into CoreElec from android?
> 
> One last idea: Did you try to use the generic EmuELEC-image? https://github.com/EmuELEC/EmuELEC/releases/download/v4.6/EmuELEC-Amlogic-ng.aarch64-4.6-Generic.img.gz
> 
> Perhaps this will work on your box...

Thank you very much regardless, but the CPU in my box is RK3566 which is quite obscure. The EMUELEC image for Amlogic will definitely not work. Let's see if any friends have encountered similar problems. Thank you again!

CODEJZX, 2023-04-18:
> Hope you can find a solution. Or you can think about trying lakka (https://www.lakka.tv/) or batocera (https://batocera.org/), perhaps these Emulators will work on your device?

OKOK，THANKS VERY MUCH！

Langerz82, 2023-04-19:
@CODEJZX Just wondering when does it turn green? Straight away when powered on?

CODEJZX, 2023-04-19:
> @CODEJZX Just wondering when does it turn green? Straight away when powered on?

YES

worstcase-scenario, 2023-04-19:
When I had problems installing EmuElec on my cheapo Android Box, I tried to install CoreElec (https://coreelec.org/), in order to see if the box itself is able to run / boot into another system (using the toothpick-method...) By that I could work out which dtb.img / Device tree to use and then I used that knowledge to install EmuElec.

CODEJZX, 2023-04-19:
> When I had problems installing EmuElec on my cheapo Android Box, I tried to install CoreElec (https://coreelec.org/), in order to see if the box itself is able to run / boot into another system (using the toothpick-method...) By that I could work out which dtb.img / Device tree to use and then I used that knowledge to install EmuElec.
> 
> Update: searching the net, I found some forums that talk about your system and issues with multiboot... So perhaps that is the root of your problem... seems you have to update the u-boot in eMMC... You should read into this:
> 
> https://www.armbian.com/station-m2/
> 
> https://forum.armbian.com/topic/24753-firefly-station-m2p2-edk2-uefigrub/

Thank you very much.I'll try it!

Langerz82, 2024-02-03:
Needs testing, I dont have a RK3566 so I cannot test it:
https://github.com/EmuELEC/EmuELEC/pull/1295