EmuELEC/EmuELEC #1194: [Bugs][Bluetooth] Post all Bluetooth Controller Bugs Here!!

Langerz82, 2023-06-03:
Hello,
In an effort to de-clutter the bugs list, post all bugs related to Bluetooth Pairing here thank-you.

junm6802030, 2023-06-05:
patch works

s905x4+ Fake PS4 Controller

After the device is restarted, turn on the SHARE + PS button to pair successfully instantly.

God worked hard
Thank you so much

junm6802030, 2023-06-07:
Problems I encountered after adding the patch

It cannot be directly and automatically recognized after restarting
But you can connect in seconds after turning on the pairing mode on the handle, which is a significant improvement over the previous ones. Can you add a script to execute the command emuelec-bluetooth every time you boot?
and that solves my problem.

After many tests, there are two solutions
1. Enter the command after the box starts: emuelec-bluetooth
2. After the box starts, enter the command: bluetoothctl and then enter connect target handle MAC

                                    Machine model: Amlogic
                                   CoreELEC dt-id: sc2_s905x4_4g_1gbit

EMUELEC:~ # emuelec-bluetooth
Scanning available devices for 60 seconds, with interval 15, please wait...
Starting scan 1, please wait...
detected device 小米手机, mac: C4:0B:CB:66:22:33, icon: phone, paired: False
detected device BRAVIA, mac: B0:10:41:51:95:5C, icon: audio-card, paired: False
detected device , mac: 43:6D:C4:C6:69:B2, icon: , paired: False
detected device , mac: 79:B2:6B:1D:A9:C3, icon: , paired: False
detected device , mac: 70:7F:C0:47:99:FF, icon: , paired: False
detected device , mac: 71:F3:06:CF:02:2F, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
found device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
pairing a paired device Wireless Controller
failed to pair Wireless Controller, try again
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 2, please wait...
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: False
found device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: False
successfully paired Wireless Controller
detected device 小米手机, mac: C4:0B:CB:66:22:33, icon: phone, paired: False
detected device BRAVIA, mac: B0:10:41:51:95:5C, icon: audio-card, paired: False
detected device , mac: 43:6D:C4:C6:69:B2, icon: , paired: False
detected device , mac: 79:B2:6B:1D:A9:C3, icon: , paired: False
detected device , mac: 70:7F:C0:47:99:FF, icon: , paired: False
detected device , mac: 71:F3:06:CF:02:2F, icon: , paired: False
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 3, please wait...
detected device , mac: 43:6D:C4:C6:69:B2, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device 小米手机, mac: C4:0B:CB:66:22:33, icon: phone, paired: False
detected device BRAVIA, mac: B0:10:41:51:95:5C, icon: audio-card, paired: False
detected device , mac: 79:B2:6B:1D:A9:C3, icon: , paired: False
detected device , mac: 70:7F:C0:47:99:FF, icon: , paired: False
detected device , mac: 71:F3:06:CF:02:2F, icon: , paired: False
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True

ebeem, 2023-06-07:
@junm6802030 the advantage of pairing over just connecting is that you can auto connect later. Can you let us know if you're able to reconnect after you put your controller to sleep? Or in other words, if you pair using the script and get connected, then you disconnect the controller and try to connect it again, does that work without needing to pair again?
Also, do you have different controllers to test? And what's your current controller?

btw, emuelec-bluetooth is executed on boot

Langerz82, 2023-06-07:
@ebeem In the first connect it looks like it's not dropping the pair properly in the forget function.

ebeem, 2023-06-07:
@Langerz82 can you explain more please? i am not sure if I can see the failure in forgetting.

I am checking on my phone screen so the output might not look great.

The first pairing attempt failed because the controller is probably already paired using some other script, the fact that a try again message was printed indicates an execution of the BluetoothCTL forget command.

The second attempt succeeded, which means that the device was successfully removed from trusted and paired devices and was paired again, if it wasn't the case it will not be able to pair and it will fail.

Langerz82, 2023-06-07:
@ebeem In your script just after the device found you check for a pre-existing pair, and call forget, couldnt you just add the untrust() before forget() and it should pair the 1st time correctly?

See:
```
            if (
                not dev.connected
                and dev.rssi != 0
                and "input-" in dev.icon
                and len(dev.name) > 0
            ):
                print(
                    "found device {}, mac: {}, icon: {}, paired: {}".format(
                        dev.name, dev.mac, dev.icon, dev.paired
                    )
                )

                # handle if a device is trying to pair while it's already paired
                if dev.paired:
                    print("pairing a paired device {}".format(dev.name))
                    bt.untrust(dev)
                    bt.forget(dev)
```
Is that no good?

junm6802030, 2023-06-07:
@ebeem 
@Langerz82 
My controller failed to wake up after reconnecting after sleep, because it is a fake PS4 controller. Before patching, I have been manually pairing through SHH, and how many pairings are needed, it often takes a minute or two to connect multiple times.

I have 2 handles, but both are of the same model, so it's not worth testing.

junm6802030, 2023-06-07:
@Langerz82 @ebeem 
After the box is restarted, the hands are in the sleeping state

Type emuelec-bluetooth

I can scan the handle but fail excitedly

C3:C2:08:A7:61:E1
71:4B:F4:A4:B6:7E
These two are my MAC in hand

EMUELEC:~ # emuelec-bluetooth
Scanning available devices for 60 seconds, with interval 15, please wait...
Starting scan 1, please wait...
detected device , mac: 7C:02:D0:67:A2:46, icon: , paired: False
detected device , mac: 63:C6:32:F4:E2:CE, icon: , paired: False
detected device , mac: 7B:25:87:45:C7:4F, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 2, please wait...
detected device , mac: 7C:02:D0:67:A2:46, icon: , paired: False
detected device , mac: 63:C6:32:F4:E2:CE, icon: , paired: False
detected device , mac: 7B:25:87:45:C7:4F, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 3, please wait...
detected device , mac: 7C:02:D0:67:A2:46, icon: , paired: False
detected device , mac: 63:C6:32:F4:E2:CE, icon: , paired: False
detected device , mac: 7B:25:87:45:C7:4F, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 4, please wait...
detected device , mac: 7C:02:D0:67:A2:46, icon: , paired: False
detected device , mac: 63:C6:32:F4:E2:CE, icon: , paired: False
detected device , mac: 7B:25:87:45:C7:4F, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True

Langerz82, 2023-06-07:
@ebeem hey man, the untrust probably does nothing, my bad. If a controller is listed as paired couldn't just directly call connect() ?

junm6802030, 2023-06-07:
DEBUG=True

Device reboots twice

The result after emuelec-bluetooth 6 times still can not wake up sleep

EMUELEC:~ # emuelec-bluetooth
> bluetoothctl power on
Changing power on succeeded

> bluetoothctl agent on
> bluetoothctl discoverable on
Changing discoverable on succeeded

> bluetoothctl pairable on
Changing pairable on succeeded

Scanning available devices for 60 seconds, with interval 15, please wait...
Starting scan 1, please wait...
> bluetoothctl devices
Device 47:E2:62:CF:D4:97 47-E2-62-CF-D4-97

Device 5F:35:75:A2:5F:33 5F-35-75-A2-5F-33

Device 48:A4:BB:B6:35:9D 48-A4-BB-B6-35-9D

Device 6B:AB:1C:CF:AC:80 6B-AB-1C-CF-AC-80

Device 51:8E:D4:EC:D8:BC 51-8E-D4-EC-D8-BC

Device 71:4B:F4:A4:B6:7E Wireless Controller

Device C3:C2:08:A7:61:E1 Wireless Controller

> bluetoothctl info 47:E2:62:CF:D4:97
Device 47:E2:62:CF:D4:97 (random)

	Alias: 47-E2-62-CF-D4-97

	Paired: no

	Bonded: no

	Trusted: no

	Blocked: no

	Connected: no

	LegacyPairing: no

	ManufacturerData Key: 0x004c

	ManufacturerData Value:

  10 05 7d 1c 9a 3c 79                             ..}..<y         

	RSSI: -90

	TxPower: 7

['Device 47:E2:62:CF:D4:97 (random)\n', '\tAlias: 47-E2-62-CF-D4-97\n', '\tPaired: no\n', '\tBonded: no\n', '\tTrusted: no\n', '\tBlocked: no\n', '\tConnected: no\n', '\tLegacyPairing: no\n', '\tManufacturerData Key: 0x004c\n', '\tManufacturerData Value:\n', '  10 05 7d 1c 9a 3c 79                             ..}..<y         \n', '\tRSSI: -90\n', '\tTxPower: 7\n']
> bluetoothctl info 5F:35:75:A2:5F:33
Device 5F:35:75:A2:5F:33 (random)

	Alias: 5F-35-75-A2-5F-33

	Paired: no

	Bonded: no
[... 870 lines cut ...]

	Blocked: no

	Connected: no

	WakeAllowed: no

	LegacyPairing: no

	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)

	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)

	Modalias: usb:v054Cp09CCd0100

['Device C3:C2:08:A7:61:E1 (public)\n', '\tName: Wireless Controller\n', '\tAlias: Wireless Controller\n', '\tClass: 0x00002508\n', '\tIcon: input-gaming\n', '\tPaired: yes\n', '\tBonded: yes\n', '\tTrusted: yes\n', '\tBlocked: no\n', '\tConnected: no\n', '\tWakeAllowed: no\n', '\tLegacyPairing: no\n', '\tUUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)\n', '\tUUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)\n', '\tModalias: usb:v054Cp09CCd0100\n']
detected device , mac: 59:E1:A6:C8:CD:44, icon: , paired: False
detected device , mac: 50:6F:7F:78:80:4C, icon: , paired: False
detected device , mac: 5F:35:75:A2:5F:33, icon: , paired: False
detected device , mac: 48:A4:BB:B6:35:9D, icon: , paired: False
detected device , mac: 6B:AB:1C:CF:AC:80, icon: , paired: False
detected device , mac: 51:8E:D4:EC:D8:BC, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
EMUELEC:~ #

ebeem, 2023-06-07:
@Langerz82 that will work if it's only trusted I think. It's not gonna work if it's paired.

As far as I am concerned, there's no unpair command in bluetoothctl, but maybe they introduced a one recently, I am not sure.

But only issuing an untrust command will probably break the code and make pairing impossible if the device is already paired.

ebeem, 2023-06-07:
@Langerz82 regarding issuing a connect command, not a good idea because we want the controller to be able to issue that, not the device itself or bluetoothctl.

This code will only be executed by the user after all

ebeem, 2023-06-07:
@junm6802030 your controller has 1 of the mac addresses not trusted for some reason.

Can you try this after pairing?
bluetoothctl trust 71:4B:F4:A4:B6:7E

After that put controller into sleep and try to reconnect

Langerz82, 2023-06-07:
@junm6802030 Hey, can you remove that long output you added in this post: https://github.com/EmuELEC/EmuELEC/issues/1194#issuecomment-1579956558
and add it in a gist with the link, https://gist.github.com

Try downloading and using this script:
SSH in go to /emuelec/scripts then:
```
rm /emuelec/scripts/emuelec-bluetooth
wget https://gist.githubusercontent.com/Langerz82/41cef0d791181dc8bc74a6a463129a2d/raw/9e8509536060883647dc9cd587ea11f1841d278e/gistfile1.txt
chmod +x /emuelec/scripts/emuelec-bluetooth
```

Langerz82, 2023-06-07:
@junm6802030 I'll do that again.
Open SSH and enter these commands:
```
cd /emuelec/scripts
rm /emuelec/scripts/emuelec-bluetooth
wget https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/c3972b9e93bbe5ef7c1e3e46fb56eeb3e08c4e84/emuelec-bluetooth
chmod +x /emuelec/scripts/emuelec-bluetooth
```
Then reboot and re-test.

junm6802030, 2023-06-07:
> 

> @junm6802030我会再做一次。 打开SSH并输入以下命令：
> 
> ```
> cd /emuelec/scripts
> rm /emuelec/scripts/emuelec-bluetooth
> wget https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/c3972b9e93bbe5ef7c1e3e46fb56eeb3e08c4e84/emuelec-bluetooth
> chmod +x /emuelec/scripts/emuelec-bluetooth
> ```
> 
> 然后重新启动并重新测试。

我测试了这个  依旧无法唤醒   上面的那个是否需要测试 ？

Langerz82, 2023-06-07:
It's the same, can you post the output when calling the command from SSH?

junm6802030, 2023-06-07:
@Langerz82 
EMUELEC:~ # emuelec-bluetooth
Scanning available devices for 60 seconds, with interval 15, please wait...
Starting scan 1, please wait...
detected device , mac: 77:2B:85:F3:8C:CB, icon: , paired: False
detected device , mac: 5B:3A:92:E9:DC:B1, icon: , paired: False
detected device , mac: 7B:10:BD:66:4C:EF, icon: , paired: False
detected device , mac: 7F:5B:76:19:01:7D, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 2, please wait...
detected device , mac: 77:2B:85:F3:8C:CB, icon: , paired: False
detected device , mac: 5B:3A:92:E9:DC:B1, icon: , paired: False
detected device , mac: 7B:10:BD:66:4C:EF, icon: , paired: False
detected device , mac: 7F:5B:76:19:01:7D, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 3, please wait...
detected device , mac: 77:2B:85:F3:8C:CB, icon: , paired: False
detected device , mac: 7B:10:BD:66:4C:EF, icon: , paired: False
detected device , mac: 7F:5B:76:19:01:7D, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting scan 4, please wait...
detected device , mac: 5B:3A:92:E9:DC:B1, icon: , paired: False
detected device , mac: 77:2B:85:F3:8C:CB, icon: , paired: False
detected device , mac: 7B:10:BD:66:4C:EF, icon: , paired: False
detected device , mac: 7F:5B:76:19:01:7D, icon: , paired: False
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
EMUELEC:~ #

Langerz82, 2023-06-07:
@junm6802030  Try this:
```
cd /emuelec/scripts
rm /emuelec/scripts/emuelec-bluetooth
wget https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/03cf812491bc99768c2e99a4bc7d381be6eeb698/emuelec-bluetooth
chmod +x /emuelec/scripts/emuelec-bluetooth
```
try and run in SSH and post output. It has a small change to help us debug.

junm6802030, 2023-06-07:
@Langerz82 Tested can't wake up
EMUELEC:~ # emuelec-bluetooth
Scanning available devices for 60 seconds, with interval 15, please wait...
Starting scan 1, please wait...
detected device , mac: 6C:8F:16:2C:D4:D9, icon: , paired: False
BluetoothDevice(mac='6C:8F:16:2C:D4:D9', name='', alias='6C-8F-16-2C-D4-D9', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
detected device , mac: 45:19:A5:C6:83:39, icon: , paired: False
BluetoothDevice(mac='45:19:A5:C6:83:39', name='', alias='45-19-A5-C6-83-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-82)
detected device , mac: 61:42:4F:A7:F7:DC, icon: , paired: False
BluetoothDevice(mac='61:42:4F:A7:F7:DC', name='', alias='61-42-4F-A7-F7-DC', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)
detected device , mac: 7A:5F:04:2C:C8:69, icon: , paired: False
BluetoothDevice(mac='7A:5F:04:2C:C8:69', name='', alias='7A-5F-04-2C-C8-69', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
Starting scan 2, please wait...
detected device , mac: 6C:8F:16:2C:D4:D9, icon: , paired: False
BluetoothDevice(mac='6C:8F:16:2C:D4:D9', name='', alias='6C-8F-16-2C-D4-D9', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
detected device , mac: 45:19:A5:C6:83:39, icon: , paired: False
BluetoothDevice(mac='45:19:A5:C6:83:39', name='', alias='45-19-A5-C6-83-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-82)
detected device , mac: 61:42:4F:A7:F7:DC, icon: , paired: False
BluetoothDevice(mac='61:42:4F:A7:F7:DC', name='', alias='61-42-4F-A7-F7-DC', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)
detected device , mac: 7A:5F:04:2C:C8:69, icon: , paired: False
BluetoothDevice(mac='7A:5F:04:2C:C8:69', name='', alias='7A-5F-04-2C-C8-69', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
Starting scan 3, please wait...
detected device , mac: 6C:8F:16:2C:D4:D9, icon: , paired: False
BluetoothDevice(mac='6C:8F:16:2C:D4:D9', name='', alias='6C-8F-16-2C-D4-D9', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
detected device , mac: 45:19:A5:C6:83:39, icon: , paired: False
BluetoothDevice(mac='45:19:A5:C6:83:39', name='', alias='45-19-A5-C6-83-39', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-82)
detected device , mac: 61:42:4F:A7:F7:DC, icon: , paired: False
BluetoothDevice(mac='61:42:4F:A7:F7:DC', name='', alias='61-42-4F-A7-F7-DC', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-76)
detected device , mac: 7A:5F:04:2C:C8:69, icon: , paired: False
BluetoothDevice(mac='7A:5F:04:2C:C8:69', name='', alias='7A-5F-04-2C-C8-69', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
detected device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
[... cut ...]
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
EMUELEC:~ #

junm6802030, 2023-06-07:
@Langerz82 The problem is still unable to wake up. As long as I have time, I can keep testing. I also hope to wake up the controller to improve the gaming experience.

> bluetoothctl info 71:4B:F4:A4:B6:7E
Device 71:4B:F4:A4:B6:7E (public)

	Name: Wireless Controller

	Alias: Wireless Controller

	Class: 0x00002508

	Icon: input-gaming

	Paired: no

	Bonded: no

	Trusted: no

	Blocked: no

	Connected: yes

	LegacyPairing: no

['Device 71:4B:F4:A4:B6:7E (public)\n', '\tName: Wireless Controller\n', '\tAlias: Wireless Controller\n', '\tClass: 0x00002508\n', '\tIcon: input-gaming\n', '\tPaired: no\n', '\tBonded: no\n', '\tTrusted: no\n', '\tBlocked: no\n', '\tConnected: yes\n', '\tLegacyPairing: no\n']
> bluetoothctl info C3:C2:08:A7:61:E1
Device C3:C2:08:A7:61:E1 (public)

	Name: Wireless Controller

	Alias: Wireless Controller

	Class: 0x00002508

	Icon: input-gaming

	Paired: no

	Bonded: no

	Trusted: no

	Blocked: no

	Connected: yes

	LegacyPairing: no

Failed to pair: org.bluez.Error.AuthenticationCanceled

detected device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: False
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
trusting device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: False
> bluetoothctl trust C3:C2:08:A7:61:E1
Changing C3:C2:08:A7:61:E1 trust succeeded

> bluetoothctl pair C3:C2:08:A7:61:E1
Attempting to pair with C3:C2:08:A7:61:E1

Failed to pair: org.bluez.Error.ConnectionAttemptFailed

Langerz82, 2023-06-07:
@ebeem , @junm6802030 
The problem with the controller waking up, is it's configuration of that controller does not allow waking to occur (wake_allowed=False). We would have to have a scan running in the background polling constantly to check for devices searching to connect again. unless there is another way, maybe ebeem knows a solution?

ebeem, 2023-06-07:
@Langerz82 
I am not sure if that will not introduce extra input lag, most likely it will.

The wake allowed should allow a device to wake the system from suspend/sleep. It should have nothing to do with auto connect.

@junm6802030 
A friend of mine tested original ps4 controller and it worked, so the question is, does this generic controller support different modes? Some manufacturers support different modes, like xbox, ps, switch, etc. Is it the case for this controller of yours? Maybe that's why you get multiple devices.

If you have Android, Windows, or Linux, try to pair the controller and see if it can auto connect. If the problem is with the controller, then I don't think we should try to handle it here. However, if it works on one of the former OSes, then we should investigate more.

junm6802030, 2023-06-07:
> @Langerz82 I am not sure if that will not introduce extra input lag, most likely it will.
>
> The wake allowed should allow a device to wake the system from suspend/sleep. It should have nothing to do [...]
@Langerz82 @ebeem 
i think you are right
If there is a problem with the handle itself, it may not even support wake-up, then we will not be able to solve it.

Langerz82, 2023-06-07:
Here is what I have so far:
https://gist.github.com/Langerz82/6838a6c44232902a47182658c55cc00f

It enables the script to enter a loop or just execute once, which will be needed in ES.
When the emuelecRunEmu.sh is called it will update a flag file in temp which basically makes the script silent.

syntax:
emuelec-bluetooth SCAN_INTERVAL SCAN_TIME SCAN_LOOP SCAN_LOOP_INTERVAL
emuelec-bluetooth 10 10 1 <-- should run the script only once.
emuelec-bluetooth 10 10 -1 30 <-- should run the loop once for every 30 seconds

Let me know what you think.

ebeem, 2023-06-07:
@Langerz82 
I already run the script on my machine whenever the user is not in game, I can share the script later if you need it, but I don't know if there are any drawbacks to this approach, to me it's working completely fine and I can pair anytime without using the menu

Langerz82, 2023-06-07:
A better way might be to save the process-id in a file using pythons os.getpid(), then when emuelecRunEmu runs in the script I make it halt the process by id, then resume when ES re-launches.

Langerz82, 2023-06-07:
@ebeem Your right a total timeout is better than loop count. I will also make polling configurable in ES, and the frequency in which it checks.

ebeem, 2023-06-07:
pass `-1` for total time argument and add your changes of handling the special controller.

````python
#!/usr/bin/env python

from subprocess import Popen, PIPE
from dataclasses import dataclass
import time
import sys
import os
import re

DEBUG = False

@dataclass
class BluetoothDevice:
    mac: str
    name: str
    alias: str
    clas: str
    icon: str
    paired: bool
    bounded: bool
    trusted: bool
    blocked: bool
    connected: bool
    wake_allowed: bool
    legacy_pairing: bool
    rssi: int

class ShellIO:
    @staticmethod
    def execute(cmd: list[str]) -> list[str]:
        if DEBUG:
            print("> " + " ".join(x for x in cmd))
        out: list[str] = []
        with Popen(cmd, stdout=PIPE, bufsize=1, universal_newlines=True) as p:
            for line in p.stdout:
                out.append(line)
                if DEBUG:
                    print(line)
        return out

    @staticmethod
    def execute_lookup(cmd: list[str], match: str) -> str:
        out = ShellIO.execute(cmd)
        for o in out:
            if match in o:
                return o
        return ""

    @staticmethod
    def execute_async(cmd: list[str]):
        Popen(cmd, stdout=PIPE, bufsize=1, universal_newlines=True)

class BluetoothCTL:
    def _parse_device_info(self, mac: str):
        info = ShellIO.execute(["bluetoothctl", "info", mac])
        dev = BluetoothDevice(
            mac=mac,
            name="",
            alias="",
            clas="",
            icon="",
            paired=False,
            bounded=False,
            trusted=False,
            blocked=False,
            connected=False,
            wake_allowed=False,
[... 164 lines cut ...]
                not dev.connected  # exclude already connected devices
                and dev.rssi != 0  # exclude inactive devices (saved)
                and "input-" in dev.icon  # exclude any non-input device
                and len(dev.name.strip()) > 0  # exclude unnamed devices
            ):
                print(
                    "found device {}, mac: {}, icon: {}, paired: {}".format(
                        dev.name, dev.mac, dev.icon, dev.paired
                    )
                )

                # handle if a device is trying to pair while it's already paired
                if dev.paired:
                    print("pairing a paired device {}".format(dev.name))
                    bt.forget(dev)

                # trust then pair and finally connect
                if bt.trust(dev) and bt.pair(dev) and bt.connect(dev):
                    print("successfully paired {}".format(dev.name))
                else:
                    # if for any reason one step fails, forget the device so next time it can be paired
                    bt.forget(dev)
                    print("failed to pair {}, try again".format(dev.name))

````

Langerz82, 2023-06-07:
@ebeem Hey mate, here are some changes I made. Check if the changes do not break compatibility with your code.
See:
https://gist.github.com/Langerz82/6838a6c44232902a47182658c55cc00f/revisions

ebeem, 2023-06-07:
`ignoreList` isn't cleared anywhere, so basically through the whole lifetime of a system, your type of controller will only be able to pair once, repairing isn't possible? that's how it looks like to me from the code.

it seems like you have  changed the conditions in the if here?
````python
        # decide whether we should stop or resume scanning and pairing
        if is_in_game():
            print("emustation is inactive now, stopping scan")
            bt.scan_stop()
        else:
            print("emustation is active now, resuming scan")
            bt.scan_async(timeout=SCAN_TIME)
````

I think this will result in a scan and a kill attempt every iteration, the `scan_async`  with `-1` timeout means an infinite scan `bluetoothctl scan on`, I don't think you're supposed to run this multiple times.
I haven't tested, can you check and confirm based on the output you get? this will probably keep printing
````
emustation is inactive now, stopping scan
emustation is inactive now, stopping scan
emustation is inactive now, stopping scan
emustation is inactive now, stopping scan
emustation is inactive now, stopping scan
````
or 
````
emustation is active now, resuming scan
emustation is active now, resuming scan
emustation is active now, resuming scan
emustation is active now, resuming scan
`````

This shouldn't happen, this should only be printed on a status change, which is a game gets activated or closed.

Langerz82, 2023-06-08:
I changed the flow a little bit and renamed some variables. Here are the parameters explained:

SCAN_TIME = 10 # time bt scans occurs for in seconds
CONNECT_ATTEMPTS = 2 # amount of device connection attempts in a scan.
SCAN_INTERVAL = 30 # the time delay in between a scan.
LIFETIME = 60 # the overall lifetime of the process before it exits.

here is the script:
code: https://gist.github.com/Langerz82/6838a6c44232902a47182658c55cc00f
diff: https://gist.github.com/Langerz82/6838a6c44232902a47182658c55cc00f/revisions

Let me know what you think, Cheers.

ebeem, 2023-06-08:
@Langerz82 I don't think I can follow up with all these changes. Please proceed with what you think works best.

Regards

Langerz82, 2023-06-08:
@junm6802030  Hello try this script hopefully it all works ok.

```
cd /emuelec/scripts
rm /emuelec/scripts/emuelec-bluetooth
wget https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/2b9281d2aeda3eb863d48ac58facfb73abeab1e1/emuelec-bluetooth
chmod +x /emuelec/scripts/emuelec-bluetooth
```
And reboot, go into ES, Emuelec Settings, and Forget Devices. Then try pairing normally, also try waking up the controller while the script is running and see if it connects during a scan.

junm6802030, 2023-06-08:
> @junm6802030 你好试试这个脚本希望它一切正常。
> 
> ```
> cd /emuelec/scripts
> rm /emuelec/scripts/emuelec-bluetooth
> wget https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/2b9281d2aeda3eb863d48ac58facfb73abeab1e1/emuelec-bluetooth
> chmod +x /emuelec/scripts/emuelec-bluetooth
> ```
> 
> 然后重新启动，进入 ES、Emuelec 设置和忘记设备。然后尝试正常配对，也尝试在脚本运行时唤醒控制器并查看它是否在扫描期间连接。
> 
> @ebeem别担心，如果你有机会测试它会很好，否则感谢你为脚本所做的所有工作。希望它适用于所有情况哈哈。

@Langerz82 没问题 我现在测试一下 PS+SHARE键  是否正常配对

junm6802030, 2023-06-08:
[image]
@Langerz82 

This latest script experience is not good
Under the Emuelec Settings, the normal pairing is not possible. It is particularly difficult to match only once out of 10 times. Personally, I think the scanning time is too long. If the device cannot be successfully matched, the long scanning time is useless.

I used the mobile app developed by Chinese Emuelec enthusiasts to connect smoothly.

The attached app has been uploaded to see if it is of reference value, please change the jpg suffix to apk.

Langerz82, 2023-06-08:
@junm6802030 Try:
```
emuelec-bluetooth 10 2 10 60
```
See if that works.
Also do you have the source code to this other script you use?
I can take a look and make any needed changes.

junm6802030, 2023-06-08:
EMUELEC:~ # emuelec-bluetooth 10 2 10 60
Scanning available devices for 10 seconds, with interval 10, please wait...
emustation is active now, resuming scan
Starting scan 1, please wait...
Starting connect attempt 0, please wait...
Starting connect attempt 1, please wait...
Scanning available devices for 10 seconds, with interval 10, please wait...
Starting scan 2, please wait...
Starting connect attempt 0, please wait...
Starting connect attempt 1, please wait...
Scanning available devices for 10 seconds, with interval 10, please wait...
Starting scan 3, please wait...
Starting connect attempt 0, please wait...
Starting connect attempt 1, please wait...
EMUELEC:~ # emuelec-bluetooth 10 2 10 60
Scanning available devices for 10 seconds, with interval 10, please wait...
emustation is active now, resuming scan
Starting scan 1, please wait...
Starting connect attempt 0, please wait...
Starting connect attempt 1, please wait...
Scanning available devices for 10 seconds, with interval 10, please wait...
Starting scan 2, please wait...
Starting connect attempt 0, please wait...
Starting connect attempt 1, please wait...
Scanning available devices for 10 seconds, with interval 10, please wait...
Starting scan 3, please wait...
Starting connect attempt 0, please wait...
Starting connect attempt 1, please wait...
EMUELEC:~ # 
@Langerz82 

The problem is still

Does the app have no reference value?

The following content is the original words of the app, and provides ideas for app development. Some bluetooth controller players in emuelec in China use this app to connect and use.

Because emuelec needs to enter a lot of commands to connect to the Bluetooth handle, the operation is more complicated. In addition, my Bluetooth controller often needs to switch back and forth between Switch and Emuelec, so I made an app for quick pairing and connection (actually, Bluetooth commands are processed through SSH, so the SSH function must be turned on, and the phone must be guaranteed Connect to the same network as the emuelec device)

PS: The connection principle comes from https://www.bilibili.com/read/cv3743842/

Langerz82, 2023-06-08:
did you use the file here:
https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/2e413001d5d1bda812bd9e9d831f1a89a741487d/emuelec-bluetooth

The scan length is 10 seconds with a 10 second break between it, It picks up my controller fine. Are you sure you are using the right file?

junm6802030, 2023-06-08:
> 你在这里使用文件了吗： https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/2e413001d5d1bda812bd9e9d831f1a89a741487d/emuelec-bluetooth
> 
> 扫描长度为 10 秒，中间有 10 秒的间隔，它可以很好地拾取我的控制器。你确定你使用的是正确的文件吗？

Maybe the write failed?
I delete the file and try to rewrite the build.

junm6802030, 2023-06-08:
EMUELEC:~ # emuelec-bluetooth 10 2 10 60
Scanning available devices for 10 seconds, with interval 10, please wait...
emustation is active now, resuming scan
Starting scan 1, please wait...
Starting connect attempt 0, please wait...
BluetoothDevice(mac='54:E1:77:60:34:D3', name='', alias='54-E1-77-60-34-D3', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-84)
BluetoothDevice(mac='7D:45:C2:14:4F:94', name='', alias='7D-45-C2-14-4F-94', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-84)
BluetoothDevice(mac='51:F3:C2:0B:00:45', name='', alias='51-F3-C2-0B-00-45', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-84)
BluetoothDevice(mac='7C:A9:14:96:43:3D', name='', alias='7C-A9-14-96-43-3D', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-80)
BluetoothDevice(mac='7C:25:DA:4A:75:99', name='', alias='7C-25-DA-4A-75-99', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
pairing and connecting device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting connect attempt 1, please wait...
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='54:E1:77:60:34:D3', name='', alias='54-E1-77-60-34-D3', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='7D:45:C2:14:4F:94', name='', alias='7D-45-C2-14-4F-94', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='51:F3:C2:0B:00:45', name='', alias='51-F3-C2-0B-00-45', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='7C:A9:14:96:43:3D', name='', alias='7C-A9-14-96-43-3D', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
pairing and connecting device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Scanning available devices for 10 seconds, with interval 10, please wait...
Starting scan 2, please wait...
Starting connect attempt 0, please wait...
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='54:E1:77:60:34:D3', name='', alias='54-E1-77-60-34-D3', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='7D:45:C2:14:4F:94', name='', alias='7D-45-C2-14-4F-94', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='51:F3:C2:0B:00:45', name='', alias='51-F3-C2-0B-00-45', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='7C:A9:14:96:43:3D', name='', alias='7C-A9-14-96-43-3D', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
pairing and connecting device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
Starting connect attempt 1, please wait...
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='54:E1:77:60:34:D3', name='', alias='54-E1-77-60-34-D3', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
[... cut ...]
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
pairing and connecting device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True

@Langerz82 
Normal scan after rewrite

Difficult to pair, restart 2 times, scan 4 times, only one of the 2 handles is successfully paired and once

The success rate of connecting with the app is 100%.

Langerz82, 2023-06-08:
It was trying to pair when already paired. I updated the code, try this file:
https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/34948358820add6ea141709961d4826e95e2f9cb/emuelec-bluetooth

junm6802030, 2023-06-08:
EMUELEC:~ # emuelec-bluetooth 10 2 10 60
/emuelec/scripts/emuelec-bluetooth: line 1: ​#!/usr/bin/env: No such file or directory
/emuelec/scripts/emuelec-bluetooth: line 3: from: command not found
/emuelec/scripts/emuelec-bluetooth: line 4: from: command not found
Version: ImageMagick 7.0.10-62 Q16 aarch64 2021-02-07 https://imagemagick.org
Copyright: (C) 1999-2021 ImageMagick Studio LLC
License: https://imagemagick.org/script/license.php
Features: Cipher DPC HDRI 
Delegates (built-in): bzlib fontconfig freetype jng jpeg lzma png xml zlib
Usage: import [options ...] [ file ]

Image Settings:
  -adjoin              join images into a single multi-image file
  -border              include window border in the output image
  -channel type        apply option to select image channels
  -colorspace type     alternate image colorspace
  -comment string      annotate image with comment
  -compress type       type of pixel compression when writing the image
  -define format:option
                       define one or more image format options
  -density geometry    horizontal and vertical density of the image
  -depth value         image depth
  -descend             obtain image by descending window hierarchy
  -display server      X server to contact
  -dispose method      layer disposal method
  -dither method       apply error diffusion to image
  -delay value         display the next image after pausing
  -encipher filename   convert plain pixels to cipher pixels
  -endian type         endianness (MSB or LSB) of the image
  -encoding type       text encoding type
  -filter type         use this filter when resizing an image
  -format "string"     output formatted image characteristics
  -frame               include window manager frame
  -gravity direction   which direction to gravitate towards
  -identify            identify the format and characteristics of the image
  -interlace type      None, Line, Plane, or Partition
  -interpolate method  pixel color interpolation method
  -label string        assign a label to an image
  -limit type value    Area, Disk, Map, or Memory resource limit
  -monitor             monitor progress
  -page geometry       size and location of an image canvas
  -pause seconds       seconds delay between snapshots
  -pointsize value     font point size
  -quality value       JPEG/MIFF/PNG compression level
  -quiet               suppress all warning messages
  -regard-warnings     pay attention to warning messages
  -repage geometry     size and location of an image canvas
  -respect-parentheses settings remain in effect until parenthesis boundary
  -sampling-factor geometry
                       horizontal and vertical sampling factor
  -scene value         image scene number
  -screen              select image from root window
  -seed value          seed a new sequence of pseudo-random numbers
  -set property value  set an image property
  -silent              operate silently, i.e. don't ring any bells 
  -snaps value         number of screen snapshots
  -support factor      resize support: > 1.0 is blurry, < 1.0 is sharp
  -synchronize         synchronize image to storage device
  -taint               declare the image as modified
  -transparent-color color
                       transparent color
  -treedepth value     color tree depth
  -verbose             print detailed information about the image
  -virtual-pixel method
                       Constant, Edge, Mirror, or Tile
  -window id           select window with this id or name
                       root selects whole screen

Image Operators:
  -annotate geometry text
[... 321 lines cut ...]
import: delegate library support not built-in '' (X11) @ error/import.c/ImportImageCommand/1297.
/emuelec/scripts/emuelec-bluetooth: line 10: DEBUG: command not found
/emuelec/scripts/emuelec-bluetooth: line 13: @dataclass: command not found
/emuelec/scripts/emuelec-bluetooth: line 14: class: command not found
/emuelec/scripts/emuelec-bluetooth: line 15: mac:: command not found
/emuelec/scripts/emuelec-bluetooth: line 16: name:: command not found
/emuelec/scripts/emuelec-bluetooth: line 17: alias:: command not found
/emuelec/scripts/emuelec-bluetooth: line 18: clas:: command not found
/emuelec/scripts/emuelec-bluetooth: line 19: icon:: command not found
/emuelec/scripts/emuelec-bluetooth: line 20: paired:: command not found
/emuelec/scripts/emuelec-bluetooth: line 21: bounded:: command not found
/emuelec/scripts/emuelec-bluetooth: line 22: trusted:: command not found
/emuelec/scripts/emuelec-bluetooth: line 23: blocked:: command not found
/emuelec/scripts/emuelec-bluetooth: line 24: connected:: command not found
/emuelec/scripts/emuelec-bluetooth: line 25: wake_allowed:: command not found
/emuelec/scripts/emuelec-bluetooth: line 26: legacy_pairing:: command not found
/emuelec/scripts/emuelec-bluetooth: line 27: rssi:: command not found
/emuelec/scripts/emuelec-bluetooth: line 30: class: command not found
/emuelec/scripts/emuelec-bluetooth: line 31: @staticmethod: command not found
/emuelec/scripts/emuelec-bluetooth: line 32: syntax error near unexpected token `('
/emuelec/scripts/emuelec-bluetooth: line 32: `    def execute(cmd: list[str]) -> list[str]:'
EMUELEC:~ # 

@Langerz82 
Serious error?

Langerz82, 2023-06-08:
sorry, typo in header, try this file:
https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/65b52e8efbb20db38f7ec5ef70227ed4bc1c0596/emuelec-bluetooth

junm6802030, 2023-06-08:
> @junm6802030 sorry, typo in header, try this file: https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/65b52e8efbb20db38f7ec5ef70227ed4bc1c0596/emuelec-bluetooth
> 
> edit: Both controllers should connect ok with this one.

还是刚才那个严重错误。

Langerz82, 2023-06-08:
Still not working on latest file?

Langerz82, 2023-06-08:
try this it had a space in bad header.

Fixed file:
https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/4fcb05b3cd079d0e16c81f0e7c348d9cbff56526/emuelec-bluetooth

junm6802030, 2023-06-08:
> try this it had a space in bad header.
> 
> Fixed file: https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/4fcb05b3cd079d0e16c81f0e7c348d9cbff56526/emuelec-bluetooth

This patch tested fine except it doesn't wake up
SHH enter the command emuelec-bluetooth

PS key + SHARE to enter the pairing mode, and it will be connected in two or three seconds

I'll restart it and test it a few more times.

junm6802030, 2023-06-08:
> emuelec

EMUELEC:~ # emuelec-bluetooth
Scanning available devices for 10 seconds, with interval 30, please wait...
emustation is active now, resuming scan
Starting scan 1, please wait...
Starting connect attempt 1, please wait...
BluetoothDevice(mac='7B:F9:46:4E:5D:BB', name='', alias='7B-F9-46-4E-5D-BB', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
BluetoothDevice(mac='75:B3:F5:3F:BA:D4', name='', alias='75-B3-F5-3F-BA-D4', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-100)
BluetoothDevice(mac='55:7E:AF:75:EA:1A', name='', alias='55-7E-AF-75-EA-1A', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
BluetoothDevice(mac='5A:B4:D5:79:7B:BC', name='', alias='5A-B4-D5-79-7B-BC', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-68)
BluetoothDevice(mac='65:F3:18:3D:95:3A', name='', alias='65-F3-18-3D-95-3A', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
BluetoothDevice(mac='5F:FC:4F:7A:FF:C8', name='', alias='5F-FC-4F-7A-FF-C8', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
pairing and connecting device Wireless Controller, mac: 71:4B:F4:A4:B6:7E, icon: input-gaming, paired: True
successfully paired Wireless Controller
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
pairing and connecting device Wireless Controller, mac: C3:C2:08:A7:61:E1, icon: input-gaming, paired: True
successfully paired Wireless Controller
Starting connect attempt 2, please wait...
BluetoothDevice(mac='7B:F9:46:4E:5D:BB', name='', alias='7B-F9-46-4E-5D-BB', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
BluetoothDevice(mac='75:B3:F5:3F:BA:D4', name='', alias='75-B3-F5-3F-BA-D4', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-100)
BluetoothDevice(mac='55:7E:AF:75:EA:1A', name='', alias='55-7E-AF-75-EA-1A', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
BluetoothDevice(mac='5A:B4:D5:79:7B:BC', name='', alias='5A-B4-D5-79-7B-BC', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-68)
BluetoothDevice(mac='65:F3:18:3D:95:3A', name='', alias='65-F3-18-3D-95-3A', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-86)
BluetoothDevice(mac='5F:FC:4F:7A:FF:C8', name='', alias='5F-FC-4F-7A-FF-C8', clas='', icon='', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=-78)
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
Scanning available devices for 10 seconds, with interval 30, please wait...
Starting scan 2, please wait...
Starting connect attempt 1, please wait...
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
Starting connect attempt 2, please wait...
BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
BluetoothDevice(mac='C3:C2:08:A7:61:E1', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=True, bounded=False, trusted=True, blocked=False, connected=True, wake_allowed=False, legacy_pairing=False, rssi=0)
EMUELEC:~ # 

@Langerz82 

At present, it is almost perfectly paired, restarted three times, and successfully paired within 3 seconds three times.

But need SHH to enter the command emuelec-bluetooth

Does this mean that every time I start the emuelec settings, just search for the paired controller?

My controller wake-up function is probably flawed, and the current patch is almost perfect for me.

junm6802030, 2023-06-08:
Re-tested and does not require SSH commands

PS key + SHARE can directly open the matching mode to pair with Bluetooth

Langerz82, 2023-06-08:
@junm6802030 Good to hear, it will run automatically at boot and in ES option to keep it polling so if you need re-connect it can do so.

junm6802030, 2023-06-08:
> @junm6802030 Good to hear, it will run automatically at boot and in ES option to keep it polling so if you need re-connect it can do so.

Thank you for your hard work, our two-day test was not in vain

Emuelec Bluetooth pairing finally ushered in good news

It is estimated that this patch can fix more than 80% of the Bluetooth handle pairing difficulties.

Looking forward to the release of the new version of emuelec, is there any plan to release version 4.7 in the near future?

Langerz82, 2023-06-08:
It's really up to project leader but he is on a bit of a break, and he might want to update CoreElec stuff before releasing new version. I guess we wait and see.

ebeem, 2023-06-08:
> @junm6802030 It was trying to pair when already paired. I updated the code, try this file:
> https://gist.githubusercontent.com/Langerz82/6838a6c44232902a47182658c55cc00f/raw/34948358820add6ea141709961d4826e95e2f9cb/emuelec-bluetooth
> 
> edit:
> @ebeem 
> Problem with second controller is it's not connected or trusted and has a rssi = 0.
> ```
> BluetoothDevice(mac='71:4B:F4:A4:B6:7E', name='Wireless Controller', alias='Wireless Controller', clas='0x00002508', icon='input-gaming', paired=False, bounded=False, trusted=False, blocked=False, connected=False, wake_allowed=False, legacy_pairing=False, rssi=0)
> ```
> How do you think we should handle this case?
> 
> 

I am not sure what that other controller is, if it's an active Bluetooth, then it should show up with rssi, I think it might be just saved from previous pairing or something

Anyways, @junm6802030 was able to pair from the beginning but wasn't able to reconnect. So the workaround that fixed his problem was just to keep running the scan for longer in the background

Langerz82, 2023-06-08:
@junm6802030, @ebeem 
I added a little extra condition if it's not trusted yet has an rssi==0.
```
and (dev.rssi != 0 or (not dev.trusted and dev.rssi == 0)) # exclude inactive devices (saved)
```
Seems to work. Hope it hasn't otherwise affected ur code (it shouldnt).

Thanks to the both of you for all your coding and testing. I've tested on mine and it pairs fast. I will include a emuelec.conf check to enable/disable polling, disable it by default, as only some people with certain controllers will need the polling.

Good job guys!! :dancing_women:

ebeem, 2023-06-08:
@Langerz82 I think this extra condition will actually break the active Bluetooth devices check, because by default, devices are not trusted. So this condition doesn't filter inactive devices anymore

This will probably cause infinite failed pairing attempts in case an input controller is saved but not paired (trying to pair a device that's actually not active). It will probably happen if scan is active without pairing or pairing script completed before pairing with the controller

Langerz82, 2023-06-08:
@ebeem, hmm do you have any alternative solution for this case? It's the only thing I can think of to get it working for that case.

ebeem, 2023-06-08:
@Langerz82 I am not quite sure what's that other controller is tbh. It's really hard to debug without having the hardware on hand.

But I am not sure if trusting that other controller did actually fix something, I think it will be worth solving if that extra trust enabled auto reconnect without new pairing

junm6802030, 2023-06-08:
EMUELEC:~ # hciconfig -a
hci1:	Type: Primary  Bus: UART
	BD Address: 43:39:00:00:1F:AC  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING 
	RX bytes:1455 acl:0 sco:0 events:87 errors:0
	TX bytes:4449 acl:0 sco:0 commands:87 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x8f
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'EMUELEC'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.1 (0x7)  Revision: 0x29f
	LMP Version: 4.1 (0x7)  Subversion: 0x6109
	Manufacturer: Broadcom Corporation (15)

hci0:	Type: Primary  Bus: USB
	BD Address: 8C:88:2B:65:1F:2B  ACL MTU: 1021:5  SCO MTU: 255:11
	UP RUNNING PSCAN 
	RX bytes:3216 acl:0 sco:0 events:240 errors:0
	TX bytes:28990 acl:0 sco:0 commands:214 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'EMUELEC #1'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 5.1 (0xa)  Revision: 0xd9a
	LMP Version: 5.1 (0xa)  Subversion: 0x2883
	Manufacturer: Realtek Semiconductor Corporation (93)

EMUELEC:~ # 

@Langerz82 @ebeem 
I am a bluetooth adapter with two active

There seems to be a problem with the built-in Bluetooth in the box, and the Bluetooth device cannot be scanned normally. I also purchased a USB Bluetooth adapter and have been using the external Bluetooth 8761 to search for Bluetooth devices.

Is this one of the reasons why rssi is having issues?

Can Emuelec disable a certain Bluetooth adapter, for example, if the built-in Bluetooth is defective but cannot be disabled, will the simultaneous use of two Bluetooth devices cause conflicts and form bugs?

ebeem, 2023-06-08:
> > @Langerz82 I am not quite sure what's that other controller is tbh. It's really hard to debug without having the hardware on hand.
> > 
> > But I am not sure if trusting that other controller did actually fix something, I think it will be worth solving if that extra trust enabled auto reconnect without new pairing
> 
> I think the rssi 0 is occuring without trust enabled because thats the value when being initialized and it's not supplying an rssi of it's own. Im not sure what you mean exactly.

Devices in bluetoothctl represents all devices, remembered (active or inactive) and also ones that were active previously but stopped sending signals (even if not trusted). Bluez deletes these inactive ones that aren't trusted or paired after sometime if they remain inactive, but you still get more than what you expect in bluetoothctl devices

[196 more replies]

closed 2024-01-18 by Langerz82