EmuELEC/EmuELEC #1196: HS2734A wifi bluetooth module not working in 4.6 but does work in 4.3

keitsang, 2023-06-05:
**Describe the bug**
HS2734A Wifi/bluetooth module not found/supported

**To Reproduce**
Installation on MXQ Pro 4k Pro+ ceases to have wifi or bluetooth on version 4.6 but is fine on 4.3

**Expected behavior**
Wifi and bluetooth both work fine in 4.3

**EmuELEC version:**
 - 4.6 (likely 4.4 and above), maybe the new CE kernel doesn't support this module

**Device (please complete the following information):**
 - SOC: S905X
 - Brand: MXQ
 - Model: 4K Pro+

**Additional context**
Using P212 1G/8G DTB, I have tried the latest coreelec v20 nightly and it doesn't pick up this module either.

Langerz82, 2023-06-06:
Try this:
https://github.com/EmuELEC/EmuELEC/issues/1194

keitsang, 2023-06-10:
Unfortunately that has not worked.

junm6802030, 2023-06-11:
@Langerz82 

He seems to have the same problem as me. It is because the built-in Bluetooth adapter has no driver or the driver is incompatible and incomplete. The easiest solution is to buy an external Bluetooth 8761 adapter, right?

Langerz82, 2024-02-03:
I have no idea this will work, try:

Under this line:
https://github.com/EmuELEC/EmuELEC/blob/v4.6/projects/Amlogic-ce/devices/Amlogic-ng/options#L147

add:
```
		ADDITIONAL_DRIVERS+=" openvfd-driver wifi_dummy-aml media_modules-aml ap6xxx-aml ssv6xxx-aml RTL8822BU-aml RTL8822CS-aml RTL8152-aml"

```
And re-build and see if it works.

Langerz82, 2024-07-24:
@keitsang 
Is this still an issue the EmuELEC 4.7 or higher?

Langerz82, 2024-09-02:
Update to EmuELEC 4.8TEST and re-try.

kagebarton, 2026-09-14:
Did a fix ever make it into the 4.8 release? I have the same wifi chip and it is not working.

kagebarton, 2026-09-15:
Thanks for reopening this! Let me know if there's any logs I can pull for your information.

Langerz82, 2026-09-16:
From Google AI. Looks like is not as simple as adding a driver to EmuELEC.

> The HS2734A is a combination Wi-Fi and Bluetooth chip (commonly found in budget Android TV boxes like the [MXQ Pro 4K](https://www.google.com/search?ibp=oshop&prds=pvt:hg,pvo:48,imageDocid:8296722324805067310,headlineOfferDocid:14175981327253687899,productDocid:14175981327253687899&q=product&sa=X&ved=2ahUKEwi83J-2jfOWAxWs1DQHHfxAASAQxa4PegoIAggACAAIEBAC), M9X Max, or Mini M8S III). It is a repackaged Broadcom BCM4334 or BCM4330 SDIO chipset rather than a Realtek (RTL) component. [[1](https://forum.armbian.com/topic/6887-m9x-max-under-mate-no-wifi/)] (https://forum.armbian.com/topic/6887-m9x-max-under-mate-no-wifi/), [[2](https://discourse.coreelec.org/t/better-wireless-support-for-old-boxes/18607)] (https://discourse.coreelec.org/t/better-wireless-support-for-old-boxes/18607), [[3](https://github.com/EmuELEC/EmuELEC/issues/1196)] (https://github.com/EmuELEC/EmuELEC/issues/1196), [[4](https://discourse.coreelec.org/t/ap6234-wifi-bt-treiber-feature-request/6948)] (https://discourse.coreelec.org/t/ap6234-wifi-bt-treiber-feature-request/6948)Because of this, trying to use a Realtek RTL driver (like rtlwifi or rtl8xxxu) will not work. You need the Broadcom brcmfmac wireless driver or the dhd driver module. [[1](https://forum.armbian.com/topic/6887-m9x-max-under-mate-no-wifi/)] (https://forum.armbian.com/topic/6887-m9x-max-under-mate-no-wifi/), [[2](https://github.com/temperednetworks/rtlwifi)] (https://github.com/temperednetworks/rtlwifi)Driver Implementation by Operating System1. Linux & Custom OS (CoreELEC / LibreELEC / EmuELEC)Wi-Fi: The HS2734A requires the brcmfmac kernel driver. However, because it operates over an SDIO interface, it also requires a specific firmware blob (brcmfmac4334-sdio.bin or similar) and an NVRAM configuration file (.txt) matched to your device’s device-tree (.dtb) configuration. [[1](https://discourse.coreelec.org/t/better-wireless-support-for-old-boxes/18607)] (https://discourse.coreelec.org/t/better-wireless-support-for-old-boxes/18607), [[2](https://forum.armbian.com/topic/6887-m9x-max-under-mate-no-wifi/)] (https://forum.armbian.com/topic/6887-m9x-max-under-mate-no-wifi/), [[3](https://github.com/EmuELEC/EmuELEC/issues/1196)] (https://github.com/EmuELEC/EmuELEC/issues/1196)Bluetooth: Bluetooth connectivity for this module is notoriously difficult to configure on mainline Linux. It requires loading a Broadcom patch file (.hcd) via a utility like brcm_patchram_plus over an internal UART interface. [[1](https://github.com/EmuELEC/EmuELEC/issues/1196)] (https://github.com/EmuELEC/EmuELEC/issues/1196), [[2](https://www.microcontrollertips.com/serialethernet-wi-fi-embedded-module-provides-dual-band-802-11n-wireless/)] (https://www.microcontrollertips.com/serialethernet-wi-fi-embedded-module-provides-dual-band-802-11n-wireless/)Note: If you are running emulation or media builds, older OS versions like LibreELEC test builds or EmuELEC 4.3 have better legacy support for this chip than versions running newer Linux kernels (such as EmuELEC 4.6+ or CoreELEC 20+), which dropped the outdated Broadcom SDIO code. [[1](https://discourse.coreelec.org/t/better-wireless-support-for-old-boxes/18607)] (https://discourse.coreelec.org/t/better-wireless-support-for-old-boxes/18607), [[2](https://github.com/EmuELEC/EmuELEC/issues/1196)] (https://github.com/EmuELEC/EmuELEC/issues/1196)

kagebarton, 2026-09-16:
Claude managed to get the wifi working. I'm still having trouble connecting a DS3 via bluetooth. Will report back:

## Root cause and fix: `gxl_p212_1g` DTB is missing `dhd_static_buf`

**Tested on:** MXQ Pro+ 4K (S905X, 1 GB). Wifi/BT chip marked HS2734A, which identifies as BCM4334 rev 3 (SDIO `02d0:4334`).
**Software:** EmuELEC 4.8 (`BUILD_ID=5a3e0877546318dd70cd001563e57bfb78328a6b`), Amlogic-ng, kernel `4.9.269`, `dhd` 1.363.59.144.10 from `ap6xxx-aml`.

### Root cause

The `wifi` node in the shipped `gxl_p212_1g.dtb` has no `dhd_static_buf` property:

```
/proc/device-tree/wifi: compatible dev_name interrupt_pin irq_trigger_type name pinctrl-0 pinctrl-names power_on_pin pwm_config status
```

The 4.9 kernel uses that property to decide how much memory to reserve for the Broadcom driver (`drivers/amlogic/wifi/wifi_dt.c`):

```c
if (of_get_property(pdev->dev.of_node, "dhd_static_buf", NULL)) {
    WIFI_INFO("dhd_static_buf all setup\n");
    bcmdhd_init_wlan_mem(1);
} else {
    WIFI_INFO("dhd_static_buf little setup\n");
    bcmdhd_init_wlan_mem(0);
}
```

- **Partial allocation without the property.** With `all_buf == 0`, `bcmdhd_init_wlan_mem()` (`drivers/amlogic/wifi/dhd_static_buf.c`) skips `wlan_static_prot`, `wlan_static_osl_buf` and `wlan_static_dhd_info_buf`. It still prints `prealloc ok`, just with a smaller size.
- **Silent NULL.** For those sections `bcmdhd_mem_prealloc()` returns the NULL pointers without logging an error.
- **`dhd` needs exactly those buffers.** It's built with `CONFIG_DHD_USE_STATIC_BUF`, so `dhd_attach` fails, retries, and gives up:

```
bcmdhd_init_wlan_mem : prealloc ok: 3188736(3114K)          <- partial setup
...
F1 signature OK, socitype:0x1 chip:0x4334 rev:0x3 pkg:0x0
wifi_platform_prealloc: failed to alloc static mem section 3  <- DHD_PREALLOC_OSL_BUF
wifi_platform_prealloc: failed to alloc static mem section 7  <- DHD_PREALLOC_DHD_INFO
wifi_platform_prealloc: failed to alloc static mem section 0  <- DHD_PREALLOC_PROT
dhd_prot_attach: kmalloc failed
bcmsdh_sdmmc: probe of sdio:0001:2 failed with error -12
...
dhd_module_init: Failed to load driver max retry reached**
dhd_module_init: Exit err=-62
```

Each failed attempt also triggers `WARNING: ... at kobject_put+0x78/0x80` in the `dhd_detach` error path.

Of the 75 DTBs in `/flash/device_trees/` on the 4.8 image, 48 have `dhd_static_buf`, including `gxl_p212_2g*`, `gxl_p212_3g`, `gxl_p230_*`, `gxl_p231_*` (1g and 2g) and `gxl_p281_*` (1g and 2g). `gxl_p212_1g` does not.

Ruled out:
- Power-up, SDIO enumeration and firmware files all work.
- `dhd.ko` does link to the kernel's `bcmdhd_mem_prealloc`.
- `dhd_sdio.ko` (100.10.545.8) can't be used instead, because it has no `4334` SDIO alias.

### Fix

Add `dhd_static_buf;` to the `wifi` node of the `gxl_p212_1g` device tree:

```dts
wifi {
	compatible = "amlogic, aml_wifi";
	dev_name = "aml_wifi";
	status = "okay";
	interrupt_pin = <&gpio GPIOX_7 GPIO_ACTIVE_HIGH>;
	irq_trigger_type = "GPIO_IRQ_LOW";
	dhd_static_buf;
	power_on_pin = <&gpio GPIOX_6 GPIO_ACTIVE_HIGH>;
	pinctrl-names = "default";
	pinctrl-0 = <&wifi_32k_pins>;
	pwm_config = <&wifi_pwm_conf>;
};
```

I haven't found where in the build the property is lost for this DTB: the DTS source, a patch, or the DTB generation step. The other 26 DTBs without it are worth checking too if they're used with AP6xxx/BCM SDIO modules.

The driver could also be hardened: `dhd` could fall back to `kmalloc` when a static section isn't available, instead of failing attach.

### Verification

I added only the `dhd_static_buf` property to the stock `gxl_p212_1g.dtb`. A property-by-property diff of the patched file against the original shows `/wifi:dhd_static_buf = <empty>` as the only difference. Booting it as `/flash/dtb.img`:

```
bcmdhd_init_wlan_mem : prealloc ok: 8162304(7971K)          <- full setup
alloc static buf at ...!
F1 signature OK, socitype:0x1 chip:0x4334 rev:0x3 pkg:0x0
Register interface [wlan0]
dhd_module_init: Exit err=0
Final fw_path=/usr/lib/firmware/brcm/fw_bcm4334b1_ag.bin
Final nv_path=/usr/lib/firmware/brcm/nvram_bcm4334.txt
dhdsdio_write_vars: Download, Upload and compare of NVRAM succeeded.
Firmware: wl0: Jun 26 2012 11:32:33 version 6.10 (TOB) (r335311)
wl_android_wifi_on: Success
dhd_open: Exit ret=0
```

- **Memory:** `MemTotal: 995000 kB`, which is correct for 1 GB.
- **Wifi:** 2.4 GHz and 5 GHz networks are visible. Connected over wifi on 5580 MHz at -43 dBm with a 150 Mbit/s TX rate. No kernel warnings.
- **Bluetooth:** still works. `hci0` is UP RUNNING over UART at 2000000 baud with 0 errors, and a scan finds nearby devices.

**Don't use `gxl_p212_2g` as a workaround.** It has `dhd_static_buf` and wifi works with it, but on a 1 GB box it tells the kernel there is 2 GB of RAM (`MemTotal: 2032528 kB`) and reserves the framebuffer at `0x7f800000`.

The patched DTB (`gxl_p212_1g_dhd_static_buf.dtb`) is attached as a stopgap for affected 1 GB boxes.

[gxl_p212_1g_dhd_static_buf.zip](https://github.com/user-attachments/files/32293563/gxl_p212_1g_dhd_static_buf.zip)

kagebarton, 2026-09-16:
Claude has fixed bluetooth with DS3 also, and this sounds like a fix that might all DS3 users (it can open a PR if desired):

## Root cause and fix: the Bluetooth agent is never shipped, and when added it is orphaned at boot

**Tested on:** MXQ Pro+ 4K (S905X, 1 GB), `gxl_p212_1g`. Onboard Bluetooth on `hci0` (UART, BD address `43:34:B0:00:1F:AC`). Three Sony PLAYSTATION(R)3 Controllers (`054C:0268`).
**Software:** EmuELEC 4.8 (`BUILD_ID=5a3e0877546318dd70cd001563e57bfb78328a6b`), Amlogic-ng, kernel `4.9.269`, BlueZ 5.72.

### Symptom

The documented PS3 flow ([wiki: Bluetooth § Notes / PS3 controllers](https://github.com/EmuELEC/EmuELEC/wiki/Bluetooth#notes--ps3-controllers)) — plug the pad in over USB, press the PS button, unplug — does nothing. Pairing by hand in `bluetoothctl` works, but only if you answer the

```
[agent] Authorize service 00001124-0000-1000-8000-00805f9b34fb (yes/no):
```

prompt yourself, and it does not survive a reboot.

There are two independent defects behind this. The second one is why the first isn't simply fixed by adding the missing unit.

### Root cause 1: `bluetooth-agent.service` is referenced but never shipped

`emuelecRunEmu.sh` stops and starts a `bluetooth-agent` unit around every game launch:

```sh
# packages/sx05re/emuelec/bin/emuelecRunEmu.sh:22
systemctl stop bluetooth-agent
# packages/sx05re/emuelec/bin/emuelecRunEmu.sh:548
systemctl start bluetooth-agent
```

That unit does not exist. Grepping the tree finds only three hits for `bluetooth-agent` — those two lines plus the agent script's own log path — and no `.service` file. `packages/sx05re/emuelec/package.mk:127` enables only `emuelec-autostart.service` and `emuelec-disable_small_cores.service`. The running system agrees:

```
/storage/.config/emuelec/logs/es_launch_stderr.log:
Failed to stop bluetooth-agent.service: Unit bluetooth-agent.service not loaded.
Failed to start bluetooth-agent.service: Unit bluetooth-agent.service not found.
```

`/usr/bin/batocera/batocera-bluetooth-agent` ships and runs fine (its `dbus`, `pgi` and `bluetool` imports all resolve), but nothing launches it — `/emuelec/logs/bluetooth-agent.log` did not exist at all until it was started by hand.

BlueZ's sixaxis plugin does its half of cable pairing correctly; it just needs an agent to authorize the result. With none registered, bluetoothd is explicit about it:

```
bluetoothd[3904]: Authentication attempt without agent
bluetoothd[3904]: profiles/input/server.c:auth_callback() Access denied: org.bluez.Error.Rejected
```

`emuelec-bluetooth`, started from `emuelec_autostart.sh`, does not fill the gap:

```python
# packages/sx05re/emuelec/bin/batocera/emuelec-bluetooth:144
def agent(self, value):
    ShellIO.execute(["bluetoothctl", "agent", "on" if value else "off"])
# ...called once at line 323
bt.agent = True
```

That is a one-shot `bluetoothctl` invocation — it exits immediately and the agent registration dies with the process. Its scan-based loop also never sees a PS3 pad, which doesn't answer inquiry scans.

### Root cause 2: the agent is orphaned by a bluetoothd restart at boot

Adding the missing unit is not sufficient. `emuelec_autostart.sh` restarts bluetoothd *after* the agent has already registered against it:

```sh
# packages/sx05re/emuelec/bin/emuelec_autostart.sh:129-132
rm /storage/.cache/services/bluez.conf &
else
systemctl restart bluetooth
emuelec-bluetooth ${BTSCANTIME} &
fi
[... 98 lines cut ...]
Vendor=1356
Product=616
Version=0
```

Across a reboot, with all three pads left `Trusted=false`, `PartOf=` behaves as intended:

```
17:47:13 systemd[1]: Started bluetooth.service.
17:47:13 systemd[1]: Started bluetooth-agent.service.
17:47:19 systemd[1]: Stopping bluetooth-agent.service...   <- follows the restart
17:47:19 systemd[1]: Stopped bluetooth.service.
17:47:19 systemd[1]: Started bluetooth.service.
17:47:19 systemd[1]: Started bluetooth-agent.service.      <- re-registers against the new daemon
```

```
/emuelec/logs/bluetooth-agent.log:
17:47:21 agent registered            (no subsequent "agent: Release")
17:47:37 agent: AuthorizeService
17:47:40 agent: AuthorizeService
17:47:46 agent: AuthorizeService
```

All three controllers reconnected on a PS button press, zero `Authentication attempt without agent` or `Refusing` entries in the journal for the whole boot, and all three records still present and unchanged.

Langerz82, 2026-09-17:
There is already a fix for bluetooth pairing, see:
https://github.com/EmuELEC/EmuELEC/pull/1667