EmuELEC/EmuELEC #1198: emuelec

junm6802030, 2023-06-11:
emuelec.conf

/storage/.config/emuelec/configs/emuelec.conf 下的audio.volume

------------ C - Audio ------------
Set the audio device (auto, hdmi, jack, speakers, headphones) mostly for OdroidGoAdvance
audio.device=auto

Set system volume (0..100)
audio.volume=50

I encountered a BUG, ​​the EE setting system sound cannot be saved, every time the box is restarted, it will be restored, how to fix it?

For example: the system volume defaults to 100, manually set to 50, and then changes to 100 after restarting.

ilikeulinux, 2023-08-30:
Skyworth E900V22C 2+8 also has the same bug

nevmind, 2023-10-27:
odroid n2 plus also has the same bug

Langerz82, 2023-12-14:
PR Fix here for testing:
https://github.com/EmuELEC/EmuELEC/pull/1266

alealeks, 2025-06-13:
I have the same problem on EmuELEC v4.7 (EmuELEC-Amlogic-ng.aarch64-4.7-Generic.img)

Langerz82, 2025-06-16:
I think it might have to do with this line:
https://github.com/EmuELEC/EmuELEC/blob/facb14bb3048a32d6b3c03226d90fc386a7e3ef6/projects/Rockchip/devices/OdroidGoAdvance/packages/odroidgoa-utils/sources/headphone_sense.sh#L11
```
[[ "$GPIO" == "1" ]] && set_ee_setting "audio.device" "headphone" || set_ee_setting "audio.device" "speakers"
```
It will always fallback to speakers if the gpio is not set.

Langerz82, 2025-06-23:
I tested on the latest dev version and it doesn't have this issue. When 4.8 is released if the issue still persists let me know and I'll re-open the issue.

worstcase-scenario, 2025-06-23:
That's strange. Are you able to install 4.8 test on another SD card and check if the issue happens then, too?

Langerz82, 2025-06-24:
@alealeks Can you list here the exact filename of the build file you used? Also can you SSH (https://github.com/EmuELEC/EmuELEC/wiki/How-to-SSH-into-EmuELEC) in and type:
```
echo $EE_DEVICE
```
and post the value here?

Thanks.

Langerz82, 2025-06-25:
This is a long shot, so I'm not sure it will work. SSH in and copy/paste the following commands:
```
cd /emuelec/bin
wget https://gist.githubusercontent.com/Langerz82/3c4426fddabc646d3e8c287c6c44444c/raw/d16f7bea2db00f7f65f4694418d8e423f44f4db1/rr_audio.sh
chmod +x rr_audio.sh
```
Then re-try setting the volume and getting into the same game system that caused the bug, exit and re-check.

Langerz82, 2025-06-25:
Ah no worries, It was a long shot to begin with.

worstcase-scenario, 2025-06-26:
When you run a game in retroarch?
What happens if you change the volume there to 50% for example?

alealeks, 2025-06-26:
> When you run a game in retroarch?

The volume after launching the game in Retroarch also reset to 100%.

worstcase-scenario, 2025-06-26:
What kind of audio driver have you set in retroarch?

Langerz82, 2025-06-26:
@alealeks 
Couple of things to help, as worstcase said list your retroarch audio driver.

Also, what happens when you launch a non retroarch emulator but a standalone emulator like AdvanceMame? same issue.

What about Ports? Can you try a couple port games and see if it has the same issue? Please list the port games you try as well.

That way we can try to isolate what could be causing the issue.

Cheers.

alealeks, 2025-06-27:
> What kind of audio driver have you set in retroarch?

I carefully checked all the available drivers and here are the test results:
- **alsa** sound is good, but settings are not saved
- **alsathread** (was the default) sound is good, but settings are not saved
- **tinyalsa** sound is hoarse, **settings are saved (!)**
- **oss** sound is gone, settings are saved
- **sld2** sound is good, but settings are not saved
- **pulse** sound is gone, settings are saved

Apparently the problem is in the sound driver. So this is a bug on Retroarch's side?

alealeks, 2025-06-27:
> Then re-try all the test cases you done above again and let me know how it goes.

Unfortunately, nothing has changed.

Langerz82, 2025-06-27:
@alealeks I got one last attempt at fixing this issue. Try the following commands and repeat the test process. If it does not work for Alsa don't worry about testing all the others.

```
cd /emuelec/bin
rm /emuelec/bin/rr_audio.sh
wget https://gist.githubusercontent.com/Langerz82/650cf49602a15a72e23c19c487dc526e/raw/9edf3d44fa9f442d39f65551b01c5b7c6d5f0b23/rr_audio.sh
chmod +x /emuelec/bin/rr_audio.sh
```

Langerz82, 2025-06-28:
@alealeks 
I found something really strange, if you change the music volume as well, the audio volume setting started saving for me. I tested on an older device. Try it and see.

alealeks, 2025-06-28:
> if you change the music volume as well, the audio volume setting started saving for me. I tested on an older device. Try it and see.

I changed the System Volume and Music Volume, rebooted the device, went into the game. Then opened the settings. The system volume was reset to 100%, and the music volume remained at the correct level. In Retroarch, alsathread is selected.

Langerz82, 2025-06-28:
> I changed the System Volume and Music Volume, rebooted the device, went into the game. Then opened the settings. The system volume was reset to 100%, and the music volume remained at the correct [...]

@alealeks How I got it to work was set the system volume to say 90% and the music volume to 90%, then launched a SNES game and quit from retroarch and check back in ES and it preserved the settings. But tbh it does not matter as this will be fixed in future versions properly.

alealeks, 2025-06-28:
For example, when I use "tinyalsa" driver, system volume settings are saved correctly after reboot and launch any game. But this driver produces a hoarse sound.

closed 2025-07-03 by Langerz82