EmuELEC/EmuELEC #1207: Unable to Save Custom Scaling Configuration

BiancaRerre, 2023-07-05:
I can't save my scale setting,
I necessarily need to use "custom", so that it is to my liking, however the custom number that I define is not being saved, is there any way for me to deactivate that emuelec overwrites the config file? or disable the configuration via the emuelec menu?
it's crap this business of everything I have to choose from the menu, because several things don't work,

I even tried using "per folder configuration", per core, per rom,
but nothing is saved, I need help!

I am using "EmuELEC-OdroidGoAdvance.aarch64-4.6-odroidgo2.img"

Langerz82, 2023-07-07:
Scale setting? Do you mean Integer scaling? Or adjusting the video resolution? Currently we do not support resolution changes with Rockchip handhelds, I think it's fixed.

BiancaRerre, 2023-07-07:
these numbers here, I want to leave it like this, but they don't want to be saved

[image]

Retroarch:
Settings>video>scaling
Aspect ratio: custom
Custom aspect ratio x, y, widht, height

-----

there in the emuelec menu, there is the option to choose the aspect ratio, but when I put it in custom, it does not appear to choose the numbers I want to use. if I try to edit manually in retroarch, the number is not saved, because it seems that emuelec always replaces the number

Langerz82, 2023-07-22:
solution posted if this still is an issue let me know.

closed 2023-07-22 by Langerz82