EmuELEC/EmuELEC #1210: Question about splash videos length / Auto-stop

worstcase-scenario, 2023-07-15:
Hello, I have a small issue/question and hope that someone can help me with it:

I wanted to have a splash-video running before starting games from each systems, did that and it works great (by putting the .mp4-files into the /roms/splash-directory).

But as far as I experienced it, EmuElec always plays the WHOLE splash video to the end before starting the game - even though the internal "pre-loading" time is already done (I know that because I have checked the loading time without having a video but just a splash-screen). 

Is there any way that EmuElec automatically can stop the splash video when finished loading and go on to start the emulation?
Or do I just have to deal with it by having only ten second long splash videos or so?

Looking forward for any suggestions or answers! Thanks in advance!

Langerz82, 2023-12-14:
As explained you would need a hook into retroarch or modify retroarch to execute a kill command to the splash video process. Retroarch only starts loading the rom once the process has started.

shantigilbert, 2023-12-14:
With the current implementation this is not really possible, at least not in a simple way. The splash screen is a separate process than the emulator itself. What happens is something like this:  The Launcher (ES) calls a script, this script in turn sets any of the custom settings, screen ratio, sound, resolution, bezels, etc, etc, then in the end it calls the splash screen, since we use the frame buffer for display, only one thing can be shown at a time, so we need to finish the splash screen before the script continues and then we call anything else (in this case Retroarch), otherwise the splash screen would go into the background, but the display will try to load 2 things at the time  and it will look all scrambled/mixed. 

So in other words, the splash screen is a lie, it is NOT pre-loading the game in the background, the game starts loading only after the video/splash finishes. 

Unfortunately there is no real easy way to have multiple emulators/ports/etc control a splash screen. I guess a custom splash loader than would block the display until the emulator is loaded completely then ends the splash is doable...in theory, in practice its a whole different story, and not something in the plans or to do list.

worstcase-scenario, 2023-12-15:
Thank you very much for your explanation! 
It all makes sense. I guess the only question/option would be if the splash screen format could be a .gif? Is that possible?
This way it could show an animated picture (like a rotating hour glass, for example) to fake a video...
But this whole feature is really not that important, it was just an aesthetic idea. You guys have so many more important technical issues to solve... thank you again for your incredible work! It's very much appreciated!