EmuELEC/EmuELEC #1213: Problems with the game MarioGolf64. I've tried changing all emulators. He starts the game, and on the first shot, the RED screen appears

Mario64Golf64, 2023-08-16:
# BEFORE YOU OPEN AN ISSUE MAKE SURE IT IS NOT ALREADY RESOLVED IN THE WIKI https://github.com/EmuELEC/EmuELEC/wiki

# MAKE SURE YOUR DEVICE IS A [SUPPORTED DEVICE](https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices)  

# PLEASE DO NOT POST GENERAL HELP QUESTIONS! EITHER USE THE FORUM (https://emuelec.org) OR USE DISCORD FOR THAT (https://discord.gg/cbgtJTu), THIS IS ONLY FOR BUG/ISSUES REPORTING!

# DELETE EVERYTHING ABOVE THIS LINE (INCLUDING THIS LINE) BEFORE SUBMITTING TO MAKE SURE YOU'VE READ THIS!

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**EmuELEC version:**
 - Version [e.g. 22]

**Device (please complete the following information):**
 - SOC: [e.g. S095x3]
 - Brand: [e.g. Beelink]
 - Model [e.g. mini g1]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Mario64Golf64, 2023-08-16:
Problems with the game MarioGolf64. I've tried changing all emulators. He starts the game, and on the first shot, the RED screen appears

Kelvfimer, 2023-08-16:
Pls follow the template with those values not filled in. We can't help. So we need the emuelec version, board you are using etc

If not we have ti close the request.

Mario64Golf64, 2023-08-16:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mario64Golf64, 2023-08-16:
Do you need any more information?

Kelvfimer, 2023-08-16:
Tested in the latest version 4.6. It is working in all emulators except mupen64 32bit. It is an emulator issue not Emuelec issue.

Btw mupen64plus is not recommended to be used as the development was stopped in 2019.

Pls, for more info read

https://github.com/libretro/mupen64plus-libretro

You must use mupen64plus next or parallel. I tested on both of them and it works. 

Finally, remember we do NOT support custom systems with non official Emuelec embeded as the one you are using from Vayava. So be careful if you follow any official upgrade guide, you can brick your device and it will no longer work.

With an official Emuelec image, the recomendation is to be in the latest version but this Vayava device it is not official and unknown for us.

Thx

closed 2023-08-26 by Langerz82