EmuELEC/EmuELEC #1220: Compiling EE code with  `ninja: error: loading 'build.ninja': No such file or directory` ERROR.

yingw, 2023-08-25:
I was building EE image from source code, the toolchain build succefully:
`PROJECT=Rockchip` DEVICE=GameForce ARCH=aarch64 DISTRO=EmuELEC ./scripts/build toolchain`

But when build image using:
`PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC make image`

It continue throw a error: `ninja: error: loading 'build.ninja': No such file or directory`, I search google for days and have no progress, re-download the project and re-compile it with the same error. Any idea?

Im using ubuntu 22.04 (WSL) on Windows 11, with the newest code from git and dev branch.

[image]

Kelvfimer, 2023-08-25:
Hello

Pls can you do the following?

sudo apt-get update
sudo apt-get install gobject-introspection

delete the build folder, yes | rm -r build.EmuELEC-Amlogic-ng.aarch64-4

then execute again PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC make image

yingw, 2023-08-26:
hi, @Kelvfimer it's working, thanks, so what's this amazing magic about `gobject-introspection`? 
But, after hours compiling, it occurs ag, but this time it was at step 290 core-info:target (previously was at 088 glibc:init) , it's kind of weird, I tried to delete the folder in `build.EmuELEC-Amlogic-ng.aarch64-4/image/.stamps/core-info/` then ran ag, but it failed ag.

So, what's this Meson or ninja for , and why it fails randomly... and what I should do ? Delete the whole build folder then run ag?

[image]

Kelvfimer, 2023-08-26:
Ninja and meson are build systems. I don't see clear error there.

Pls do 

sudo apt-get update
sudo apt-get upgrade

yes | rm -r build.EmuELEC-Amlogic-ng.aarch64-4
Yes |rm -r sources

PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC make image

Thx

yingw, 2023-08-26:
Thank you, @Kelvfimer , I deleted it ag, this time both 088 and 290 step about meson and ninja goes well.

closed 2023-08-26 by yingw