EmuELEC/EmuELEC #1221: Compile with error about evtest:target step.

yingw, 2023-08-26:
Compile EE with newest dev branch, and got this error at step317 evtest:target 
```
[312/628] [DONE] install xmlstarlet:target
<<< evtest:target seq 318 <<<
BUILD      evtest (target)
    TOOLCHAIN      autotools
Executing (target): /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/configure --host=aarch64-libreelec-linux-gnu --build=x86_64-linux-gnu --prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --disable-static --enable-shared
configure: WARNING: unrecognized options: --disable-static, --enable-shared
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for aarch64-libreelec-linux-gnu-strip... /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-strip
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for aarch64-libreelec-linux-gnu-gcc... /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc accepts -g... yes
checking for /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc option to enable C11 features... none needed
checking whether /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc... gcc3
checking for xmlto... /usr/bin/xmlto
checking for asciidoc... /usr/bin/asciidoc
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure: WARNING: unrecognized options: --disable-static, --enable-shared
Executing (target): make
make[1]: Entering directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
make  all-am
make[2]: Entering directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/python3: No module named asciidoc
make[2]: *** [Makefile:855: evtest.xml] Error 1
make[2]: Leaving directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
make[1]: *** [Makefile:316: all] Error 2
make[1]: Leaving directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
FAILURE: scripts/build evtest:target during make_target (default)
*********** FAILED COMMAND ***********
make ${PKG_MAKE_OPTS_TARGET}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install evtest:target has failed!

The following log for this failure is available:
  /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/318.log

>>> evtest:target seq 318 >>>
[313/628] [FAIL] install evtest:target

The following log for this failure is available:
  /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/318.log

Parallel build failure - see log for details. Time of failure: Sat Aug 26 12:23:58 CST 2023
make: *** [Makefile:10: image] Error 1
```

It gives me a warning : `/toolchain/bin/python3: No module named asciidoc` but the Ubuntu (WSL) had asciidoc installed. not sure need install in build folder?

and the 318.log don't have too much useful info...

```
yinguowei@N220R-WIN11:~/EmuELEC$ tail -100 build.EmuELEC-Amlogic-ng.aarch64-4/.threads/logs/318.log
BUILD      evtest (target)
    TOOLCHAIN      autotools
Executing (target): /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/configure --host=aarch64-libreelec-linux-gnu --build=x86_64-linux-gnu --prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --disable-static --enable-shared
configure: WARNING: unrecognized options: --disable-static, --enable-shared
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for aarch64-libreelec-linux-gnu-strip... /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-strip
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for aarch64-libreelec-linux-gnu-gcc... /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc accepts -g... yes
checking for /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc option to enable C11 features... none needed
checking whether /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/aarch64-libreelec-linux-gnu-gcc... gcc3
checking for xmlto... /usr/bin/xmlto
checking for asciidoc... /usr/bin/asciidoc
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
configure: WARNING: unrecognized options: --disable-static, --enable-shared
Executing (target): make
make[1]: Entering directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
make  all-am
make[2]: Entering directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/toolchain/bin/python3: No module named asciidoc
make[2]: *** [Makefile:855: evtest.xml] Error 1
make[2]: Leaving directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
make[1]: *** [Makefile:316: all] Error 2
make[1]: Leaving directory '/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/.aarch64-libreelec-linux-gnu'
FAILURE: scripts/build evtest:target during make_target (default)
*********** FAILED COMMAND ***********
make ${PKG_MAKE_OPTS_TARGET}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install evtest:target has failed!
yinguowei@N220R-WIN11:~/EmuELEC$
```

Tried delete the build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35 and sources/evtest folders and compile ag , same error.

Kelvfimer, 2023-08-26:
Hello

you should do pip install asciidoc, but the important thing is to understand the number of python framework you have installed

https://unix.stackexchange.com/questions/410579/change-the-python3-default-version-in-ubuntu

maybe is picking apython framework that doesn't have it install.

On the other hand with so many errors that you have. Delete everything and start from scratch

apt install gcc make git unzip wget xz-utils libsdl2-dev libsdl2-mixer-dev libfreeimage-dev libfreetype6-dev libcurl4-openssl-dev rapidjson-dev libasound2-dev libgl1-mesa-dev build-essential libboost-all-dev cmake fonts-droid-fallback libvlc-dev libvlccore-dev vlc-bin texinfo premake4 golang libssl-dev curl patchelf xmlstarlet default-jre xsltproc

apt install tzdata xfonts-utils lzop

git clone https://github.com/EmuELEC/EmuELEC.git
cd EmuELEC
git checkout dev
PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC make image

Langerz82, 2023-08-26:
I've noticed some packages fail with an unknown module when they are nested multiple times in directories. In the directory of the build of failing package try entering:
```
git submodule update --init --recursive
```

yingw, 2023-08-29:
> Hello
>
> you should do pip install asciidoc, but the important thing is to understand the number of python framework you have installed [...]

Thank you ag, @Kelvfimer , yes you are right, it's kind of a mess of my project content, everytime a Delete my build folder or whole project the next compile will be some new issue.

But, I might finally find a way not to delete the whole project or build folder to wait another hours to compile to the step. I can delete these folders:
```
./build.EmuELEC-Amlogic-ng.aarch64-4/build/evtest-1.35/
./sources/evtest-1.35/
./build.EmuELEC-Amlogic-ng.aarch64-4/.ccache
```
and run `PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC ./scripts/build evtest` to build a single component again.

Thanks anyway.

And @Langerz82 thank you too, this git command helped me a lot to complete the source files

closed 2023-08-29 by yingw