EmuELEC/EmuELEC #1222: Can't load theme list in batocera-es-theme

yingw, 2023-08-29:
Dev branch, newest code, compile and build successfully.

Can't load theme list, defined in `batocera-es-theme`, it show "no item", and nothing useful in the backend log "es-log.txt"

[image]

[image]

[image]

[image]

closed 2023-11-24 by yingw

Kelvfimer, 2023-11-24:
@yingw known issue the way the themes are downloaded changed. We need to fix it. thx

shantigilbert, 2023-11-25:
This should be fixed now.