EmuELEC/EmuELEC #1225: Failed to start rtkbt-firmware-aml service

IOsetting, 2023-09-02:
**Describe the bug**

Failed to start rtkbt-firmware-aml service

**To Reproduce**

In EmuELEC  ssh command line, run command `systemctl start rtkbt-firmware-aml` to start rtkbt-firmware-aml service will show error
```
× rtkbt-firmware-aml.service - Attach /dev/ttyS1 to BlueZ stack using rtk_h5 type
     Loaded: loaded (/usr/lib/systemd/system/rtkbt-firmware-aml.service; static)
     Active: failed (Result: exit-code) since Sat 2023-09-02 14:24:17 CST; 4min 6s ago
   Duration: 547ms
    Process: 7304 ExecStartPre=/usr/sbin/rfkill unblock bluetooth (code=exited, status=0/SUCCESS)
    Process: 7305 ExecStart=/usr/bin/rtk_hciattach -n -s 115200 /dev/ttyS1 rtk_h5 (code=exited, status=1/FAILURE)
    Process: 7306 ExecStopPost=/usr/sbin/rfkill block bluetooth (code=exited, status=0/SUCCESS)
   Main PID: 7305 (code=exited, status=1/FAILURE)

Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth :Firmware/config: rtl8761b_fw, rtl8761b_config
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth :Couldnt open extra config /opt/rtk_btconfig.txt, No such file or directory
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth :Couldnt access customer BT MAC file /storage/.config/btaddr
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth ERROR: Can't access Config file: /lib/firmware/rtlbt/rtl8761b_config, No such file or directory
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth ERROR: Read Config file error, use eFuse settings
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth ERROR: Can't access firmware /lib/firmware/rtlbt/rtl8761b_fw, No such file or directory
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth ERROR: Read Bluetooth firmware error
Sep 02 14:24:17 EMUELEC rtk_hciattach[7305]: Realtek Bluetooth ERROR: Can't initialize device 2, No such file or directory
Sep 02 14:24:17 EMUELEC systemd[1]: rtkbt-firmware-aml.service: Main process exited, code=exited, status=1/FAILURE
Sep 02 14:24:17 EMUELEC systemd[1]: rtkbt-firmware-aml.service: Failed with result 'exit-code'.
```
These two file `/lib/firmware/rtlbt/rtl8761b_config` and `/lib/firmware/rtlbt/rtl8761b_fw` are missing, but they actually exist in /usr/lib/kernel-overlays/base/lib/firmware/rtl_bt/
```bash
ll /usr/lib/kernel-overlays/base/lib/firmware/rtl_bt/
total 908
drwxrwxr-x    2 root     root           844 Nov 19  2022 ./
drwxrwxr-x   26 root     root          6442 Nov 20  2022 ../
...
-rw-rw-r--    1 root     root            25 Nov 19  2022 rtl8761b_config.bin
-rw-rw-r--    1 root     root         37740 Nov 19  2022 rtl8761b_fw.bin
```

**Expected behavior**

Manually adding the links will fix this problem
```bash
ln -s /usr/lib/kernel-overlays/base/lib/firmware/rtl_bt/rtl8761b_fw.bin /lib/firmware/rtlbt/rtl8761b_fw
ln -s /usr/lib/kernel-overlays/base/lib/firmware/rtl_bt/rtl8761b_config.bin /lib/firmware/rtlbt/rtl8761b_config
```
But these links disappear after reboot.

I am wondering what is the **correct** way to add these links in system startup, because there are other links that show up after reboot.

And once these links are added in system startup, how to enable  rtkbt-firmware-aml service so that it will start automatically?

**EmuELEC version:**
4.6

**Device (please complete the following information):**
 - SOC: S905L3A(S905X2)
 - Brand: China Mobile
 - Model: M401A

**Additional context**

wolflink2021, 2023-12-30:
add these command lines to the paragraph before the frontend of custom_start.sh runs.

ln -s /lib/firmware/rtl_bt/rtl8761b_fw.bin /lib/firmware/rtlbt/rtl8761b_fw
ln -s /lib/firmware/rtl_bt/rtl8761b_config.bin/ lib/firmware/rtlbt/rtl8761b_config
rtk_hciattach -n -s 115200 ttyS1 rtk_h5 &

you may need to reboot the system once.

IOsetting, 2023-12-31:
This works, thank you very much!

closed 2023-12-31 by IOsetting