EmuELEC/EmuELEC #1227: A module "easyrpg" can't compile

yingw, 2023-09-03:
A module "easyrpg" can't compile. All module of latest dev branch compiled successfully except this one : "easyrpg" , I have to delete this in packages folder to complete the compile, but wonder what's wrong with it. It says:
```
a2x: ERROR: "xmllint" --nonet --noout --valid "/home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/easyrpg-69aaeb2d8bf17eee565ac3e46a98809a9ebe4df0/.aarch64-libreelec-linux-gnu/resources/easyrpg-player.6.xml" returned non-zero exit status 4
```
[image]

I ran this and nothing return and no error :  
```
/usr/local/bin/a2x -a player_version=\"0.7.0\" -f manpage -D /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/easyrpg-69aaeb2d8bf17eee565ac3e46a98809a9ebe4df0/.aarch64-libreelec-linux-gnu/resources /home/yinguowei/EmuELEC/build.EmuELEC-Amlogic-ng.aarch64-4/build/easyrpg-69aaeb2d8bf17eee565ac3e46a98809a9ebe4df0/resources/easyrpg-player.6.adoc
```

and at the start of the progress, I notice a warning:
`-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)` , not sure it's related to this error.

[image]

Could it be something wrong about the toolchain? (For the lack of doxygen?) Can I just rebuild the toolchain?, I deleted the 
build.EmuELEC-Amlogic-ng.aarch64-4/toolchain folder and build with:
`PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC ./scripts/build toolchain` but not working.

shantigilbert, 2023-09-09:
Have you tried cleaning this package and recompiling? 

```
PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC ./scripts/clean easyrpg
```

then delete the easyrpg sources from the `sources` folder and then 

```
PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC ./scripts/build easyrpg
```

Sometimes these weird errors appear because of the multithreading. Maybe this is the case

yingw, 2023-09-22:
> Have you tried cleaning this package and recompiling?
>
> ```
> PROJECT=Amlogic-ce DEVICE=Amlogic-ng ARCH=aarch64 DISTRO=EmuELEC ./scripts/clean easyrpg
> ```
>
> then delete the easyrpg sources from the [...]

Thanks for your reply, I tried several ways, delete sources, delete total build, others modules may be error randomly and can be fixed by doing rebuild, just this easyrpg module cant be fixed by clean and rebuild. It keeps telling me:

"xmllint" .. returned non-zero exit status 4,

According to another warning about "Doxygen" (some doc tool), I guess maybe some xml tool or xsd xml file not in this module download not completely.
[image]

But not big deal, as I said , I gave up and delete the easyrpg module and the image built successfully.

closed 2023-09-22 by yingw