EmuELEC/EmuELEC #1245: 8bitdo M30 2.4G, 8bitdo Ultimate 2.4G pad is not connected

nevmind, 2023-10-30:
**Describe the bug**
8bitdo M30 2.4G, 8bitdo Ultimate 2.4G pad is not connected with ODROID N2 PLUS

**To Reproduce**
1. plug in 2.4G receiver of 8bitdo M30 or Ultimatate 2.4G pad
2. check connection is working or not

**Expected behavior**
Successful connection

**Screenshots**
N/A

**EmuELEC version:**
 - Version 4.6

**Device (please complete the following information):**
 - SOC: [Amlogic S922X]
 - Brand: [Hardkernel]
 - Model [Odroid N2 PLUS]
 - Version [N/A]

**Additional context**
 N/A

worstcase-scenario, 2023-12-10:
Have you updated the controller with the newest firmware?

Langerz82, 2023-12-14:
@nevmind try doing above also connect the devices and in ssh enter the following:
```
cat /proc/bus/input/devices
```
It will help us diagnose your issue.

nevmind, 2023-12-25:
@Langerz82 Check following output
------------
EMUELEC:~ # cat /proc/bus/input/devices
I: Bus=0010 Vendor=0001 Product=0001 Version=0100
N: Name="gpio_keypad"
P: Phys=gpio_keypad/input0
S: Sysfs=/devices/platform/ff800000.gpio_keypad/input/input0
U: Uniq=
H: Handlers=kbd event0
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0010 Vendor=1b8e Product=0cec Version=0001
N: Name="cec_input"
P: Phys=
S: Sysfs=/devices/virtual/input/input1
U: Uniq=
H: Handlers=kbd event1
B: PROP=0
B: EV=3
B: KEY=1 0 0 10000000000000 0

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="meson-ir"
P: Phys=meson-ir/input0
S: Sysfs=/devices/platform/ff808040.meson-ir/rc/rc0/input2
U: Uniq=
H: Handlers=kbd event2
B: PROP=0
B: EV=100013
B: KEY=fff 0 4200108fc32e 237605100000000 0 700158000 419240004801 9e96c000000000 10000ffc
B: MSC=10

EMUELEC:~ #

nevmind, 2023-12-25:
@Langerz82 I can find  Name="8BitDo 8BitDo Receiver" when I use 8bitdo wireless adapter 2(https://www.8bitdo.com/usb-wireless-adapter-2/) & 8bitdo M30 bluetooth pad.
But this is bluetooth connection. and requires to buy another receiver to connect BT pad.
-----------------
EMUELEC:~ # cat /proc/bus/input/devices
I: Bus=0010 Vendor=0001 Product=0001 Version=0100
N: Name="gpio_keypad"
P: Phys=gpio_keypad/input0
S: Sysfs=/devices/platform/ff800000.gpio_keypad/input/input0
U: Uniq=
H: Handlers=kbd event0
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0010 Vendor=1b8e Product=0cec Version=0001
N: Name="cec_input"
P: Phys=
S: Sysfs=/devices/virtual/input/input1
U: Uniq=
H: Handlers=kbd event1
B: PROP=0
B: EV=3
B: KEY=1 0 0 10000000000000 0

I: Bus=0019 Vendor=0000 Product=0000 Version=0000
N: Name="meson-ir"
P: Phys=meson-ir/input0
S: Sysfs=/devices/platform/ff808040.meson-ir/rc/rc0/input2
U: Uniq=
H: Handlers=kbd event2
B: PROP=0
B: EV=100013
B: KEY=fff 0 4200108fc32e 237605100000000 0 700158000 419240004801 9e96c000000000 10000ffc
B: MSC=10

**I: Bus=0003 Vendor=2dc8 Product=3105 Version=0111
N: Name="8BitDo 8BitDo Receiver"**
P: Phys=usb-xhci-hcd.0.auto-1.4/input0
S: Sysfs=/devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.4/1-1.4:1.0/0003:2DC8:3105.0008/input/input7
U: Uniq=E417D8059E79
H: Handlers=event3 js0
B: PROP=0
B: EV=1b
B: KEY=7fff000000000000 0 0 0 0
B: ABS=30627
B: MSC=10

EMUELEC:~ #

Langerz82, 2024-01-15:
@nevmind also in SSH can you try running:
```
sdljoytest -skip_loop
```
and see if SDL is picking up the inputs? Also paste the output here.

nevmind, 2024-02-03:
@Langerz82 Check following outputs. A developer in Hardkernel modified and built kernel.img to solve this problem.
He succeeded. but there was critical 3 side-effect. 

1. Black screen with standalone emulator 
2. 2.4G usb keyboard connection is not working
3. Black screen with zfast shader
and he couldn't solve this so far. Could this be hint for you?

EMUELEC:~ # sdljoytest -skip_loop
Sys_InitInput: Compiled with SDL version 2.24.2
Sys_InitInput: Linked   with SDL version 2.24.2
Sys_InitInput: SDL2 joystick subsystem init
Sys_InitInput: Loading /storage/.config/SDL-GameControllerDB/gamecontrollerdb.txt
Sys_InitInput: SDL_GameControllerAddMappingsFromFile() added 0 controller maps
Sys_InitInput: Joystick subsytem - Found 0 joysticks at startup
Sys_InitInput: 0 haptic devices detected.
SDL2: Joytick hotplug supported
Sys_ShutdownInput: SDL joystick not initialized. Nothing to close.
EMUELEC:~ #

Langerz82, 2024-02-03:
SDL is not detecting the input, and  /proc/bus/input/devices is not listing it either, I dont think there is a very easy fix. @7Ji might know a bit more but it's beyond my knowledge.

shantigilbert, 2024-02-13:
Confirmed not working, my ultimate 2.4ghz only works in bluetooth mode, but not on 2.4ghz even tho the device is recognized and registered:

```
MUELEC:~ # lsusb
Bus 002 Device 003: ID 1f75:0621 Innostor Technology Corporation IS621 SATA Storage Controller
Bus 002 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2357:0604 TP-Link TP-Link UB500 Adapter
Bus 001 Device 028: ID 2dc8:3109 8BitDo Ultimate Wireless Controller
Bus 001 Device 004: ID 046d:c21d Logitech, Inc. F310 Gamepad [XInput Mode]
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dmesg:

```
[ 9120.913683@2]- usb 1-1.4: new full-speed USB device number 27 using xhci-hcd
[ 9121.038985@2]- usb 1-1.4: New USB device found, idVendor=2dc8, idProduct=3106
[ 9121.038990@2]- usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9121.038992@2]- usb 1-1.4: Product: Ultimate Wireless Controller
[ 9121.038993@2]- usb 1-1.4: Manufacturer: 8BitDo
[ 9121.038995@2]- usb 1-1.4: SerialNumber: c353d8d817e4
[ 9121.050399@2]- usb 1-1.4: Unsupported device
[ 9132.186721@2]- usb 1-1.4: USB disconnect, device number 27
[ 9132.945134@2]- usb 1-1.4: new full-speed USB device number 28 using xhci-hcd
[ 9133.070204@2]- usb 1-1.4: New USB device found, idVendor=2dc8, idProduct=3109
[ 9133.070209@2]- usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9133.070211@2]- usb 1-1.4: Product: Ultimate Wireless Controller
[ 9133.070213@2]- usb 1-1.4: Manufacturer: 8BitDo
[ 9133.070214@2]- usb 1-1.4: SerialNumber: c353d8d817e4
[ 9133.083601@2]- hid-generic 0003:2DC8:3109.0009: hiddev0,hidraw0: USB HID v1.11 Device [8BitDo Ultimate Wireless Controller] on usb-xhci-hcd.0.auto-1.4/input0
```

shantigilbert, 2024-02-13:
maybe this can help: 

https://gist.github.com/ammuench/0dcf14faf4e3b000020992612a2711e2

shantigilbert, 2024-02-13:
yes, confirmed working now with:

`/bin/sh -c 'echo 2dc8 3106 > /sys/bus/usb/drivers/xpad/new_id'`

but only as a generic Xbox gamepad

shantigilbert, 2024-02-13:
@nevmind can you provide dmesg from both controllers? right after connecting each one, and paste this line (should be similar):

`[ 9133.070204@2]- usb 1-1.4: New USB device found, idVendor=2dc8, idProduct=3109`

closed 2024-02-14 by shantigilbert

nevmind, 2024-02-18:
@shantigilbert 
I've tried and works very good!. 
GREAT THANKS!!!!

nevmind, 2024-02-25:
@shantigilbert 
I've tried with 8bitdo M30 2.4G pad (White version). 
When I plug in 2.4G receiver of 8bitdo M30 2.4G, emuelec find 8b controller.
and then I tired map controller key. 

But 
D-PAD LEFT is recognized when I press D-PAD LEFT twice, and
D-PAD RIGHT doesn't work at all.

shantigilbert, 2024-02-25:
> @nevmind can you provide dmesg from both controllers? right after connecting each one, and paste this line (should be similar):
> 
> `[ 9133.070204@2]- usb 1-1.4: New USB device found, idVendor=2dc8, idProduct=3109`

@nevmind please provide that information

nevmind, 2024-02-26:
after connecting 8bitdo ultimate 2.4G
[   25.136849@2]- usb 1-1.4: New USB device found, idVendor=2dc8, idProduct=3106
after connecting 8bitdo M30 2.4G
[  281.370491@2]- usb 1-1.4: New USB device found, idVendor=0ca3, idProduct=0024

and so I tried following
1. copy /usr/lib/udev/rules.d/99-8bitdo-bluetooth-controllers.rules to /storage/.config/udev.rules.d
2. modify like this
# 8Bitdo M30
ACTION=="add", ATTRS{idVendor}=="0ca3", ATTRS{idProduct}=="0024", RUN+="/sbin/modprobe xpad", RUN+="/bin/bash -c 'echo 0ca3 0024 > /sys/bus/usb/drivers/xpad/new_id'"
3. Reboot and try mapping again.

But exactly same bug is reproduced.

robtor-de, 2025-05-16:
I also have issues with the Ultimate Bluetooth in 2.4G mode on newer ArchLinux kernels. On an older libreelec setup the wireless dongle works like a charm. I think it may have something to do with changes to the xpad driver that are in the newer kernel. When blacklisting the `xpad` module, the controller stays connected. When the module is loaded, the controller gets disconnected.

commit d0633959 (2024-02-14): Add udev rule for 8bitdo 2.4ghz Ultimate Wireless Controller

Fixes: #1245