EmuELEC/EmuELEC #1254: EmuELEC transfer Empty (or Wrong) Timezone info to RetroArch

yingw, 2023-11-18:
noticed every time open RetroArch, the timezone looks right same as in EmuELEC, but it's Empty (in the list value), and time is set to UTC ,not the correct time set in EmuELEC, tested on 4.5, 4.6 and 4.6-dev, same issue.

If change the timezone in RetroArch service settings to some like "Asia/Seoul", then the time is correct, but next time re-enter RetroArch or reboot device , the time return to UTC, not the same in EmuELEC.

yingw, 2023-11-18:
And system timeinfo looks correct:
```sh
EMUELEC:~ # cat /storage/.cache/timezone 
TIMEZONE=Asia/Seoul
EMUELEC:~ # ls -la /usr/share/zoneinfo/Asia/Seoul 
-rw-r--r--    2 root     root           415 Mar 24  2022 Seoul
EMUELEC:~ # ls /etc/localtime -l
lrwxrwxrwx    1 root     root            18 Mar 24  2022 /etc/localtime -> /var/run/localtime
EMUELEC:~ # ls /var/run/localtime -l
lrwxrwxrwx    1 root     root            30 Nov 18 22:47 /var/run/localtime -> /usr/share/zoneinfo/Asia/Seoul
```

shantigilbert, 2023-11-18:
Can you do this: 
using something like pastebin, paste your current retroarch.cfg , then open retroarch, do the time zone changes and make sure the time is correct, close retroarch via the quit menu (do not force quit it). Now paste retroarch.cfg again.  

This way we can see if there is a setting that is getting cleared on launching retroarch.

yingw, 2023-11-21:
> Can you do this: using something like pastebin, paste your current retroarch.cfg , then open retroarch, do the time zone changes and make sure the time is correct, close retroarch via the quit menu [...]

ok .. I forgot to say, I already did test that, I monitor the retroarch.cfg in shell, and found that: If I just edit the timezone info, then save the default file, the retroarch.cfg didn't even updated, but the time is correct temporally, it will be reset to UTC after restart RetroArch. But if I change other configs, the file updated. And search keyword: Local, Time, Zone, and found nothing. Looks like the RA didn't save that...

And to confirm that, I re-flash a new 4.6 image, and after first start, change nothing, such as language, timezone, just quit to RetroArch, the time is just different with in EmuELEC, it's obvious issue.

First run 4.6, change nothing:
[image]

First boot to RetroArch, the time is wrong:
[image]

The config looks correct (But it's not):
[image]

Edit it, it not locate to Mexico, it's UTC ?
[image]

Change to Mexico:
[image]

Time is correct (But won't save):
[image]

Then save default config file, quit to EmuELEC, then enter RetroArch ag, time wrong again.

And for retroarch.cfg sample, it's here: https://pastebin.com/PY3wLcrU , But I guess it's no use. 

I think it's a obvious issue, may be you can have a try.

shantigilbert, 2023-11-25:
https://github.com/EmuELEC/EmuELEC/commit/dab6df6465de243d148d0c1de0f55567d500da5a

closed 2023-11-25 by shantigilbert

yingw, 2023-11-26:
Tested, and now it's working perfectly. Thank you, @shantigilbert

commit dab6df64 (2023-11-25): Remove new line from timezone. Fixes: #1254