EmuELEC/EmuELEC #1255: Emuelec restarting every time I enter the favorites folder

SpectralVoid31, 2023-11-19:
**Description**

Before, my emuelec worked normally, until it started restarting the system every time I entered the favorites folder, and if there were no favorites, I could enter normally, I would like to know what I should do to fix this bug.

**Reproduce**
Steps to reproduce the behavior:
1. Go to ‘favorites’
2. Click on ‘favorites’
3. See error: System reset

**Expected behavior**
I was hoping that I could get into the Favorites folder and be able to access them without any major problems.

**EmuELEC version:**
   - Version 34

**Device (fill in the following information):**
   - SOC:RK3126C
   - Brand: Generic
   - Model M17
   - Standard version

SpectralVoid31, 2023-11-19:
Bug resolved, it was the number of games that were doing this, the M17 limit, I believe, should be 1500 games

shantigilbert, 2023-11-19:
That is not a device we give support to. https://github.com/EmuELEC/EmuELEC/wiki/Supported-Devices#special-note

closed 2023-11-19 by shantigilbert