EmuELEC/EmuELEC #1261: Can you add openbor fork to support Final Fight LNS Ultimate v0.4?

wolflink2021, 2023-11-25:
This is a great fansgame. a specific version of openbor fork is required to work properly.
It would be great if EmuELEC could support it.

official website
https://fflnsultimate.firebaseapp.com/home

engine source code
https://github.com/gonzalomvp/openbor

shantigilbert, 2023-11-26:
I had to add a patch for it to compile and while the game run it looks like this:

[image]

not sure if this is correct or if I am missing something

shantigilbert, 2023-11-26:
Unzip this file into /emuelec/bin

[openborffnls.zip](https://github.com/EmuELEC/EmuELEC/files/13466125/openborffnls.zip)

And try and load the game, warning that you will most likely need a keyboard to configuration it on first run as the master.cfg we use is not compatible. 

You need to:
Use a keyboard to manually go o into gamepad options and set up your gamepad
Sound options and set volumes to 100
System options and turn of all debugging settings

Not sure how or if this this will be implemented but in the meantime you can play it with this method.

shantigilbert, 2023-11-26:
Still does not look correct tho
[image]

not sure if its the game or the openbor version, or is that how its supposed to look?

shantigilbert, 2023-11-27:
Ok this works, all you need to do is replace the master.cfg in /emuelec/configs/openbor [image] with this one in the zip
[master.zip](https://github.com/EmuELEC/EmuELEC/files/13468661/master.zip)

wolflink2021, 2023-11-30:
Thank you very much.

commit 87b1b35e (2023-12-01): OpenBOR: bump to b8303cc - Add OpenBORFFNLS 

Not the best way to add another version, but it will have to do for now. 

Closes #1261